dorssel/usbipd-win #423: Device error: USB transfer error 4 [LibUsbDataInCallback]

lamaks, 2022-07-15:
I am getting error while running Coral USB device on WSL2 in Docker Ubuntu 20.04 container. The USB driver returns the following error:

`F driver/usb/usb_driver.cc:1148] HandleQueuedBulkIn transfer in failed. Invalid argument: USB transfer error 4 [LibUsbDataInCallback]
Aborted`

The detailed setup is described in the [Coral issue](https://github.com/google-coral/edgetpu/issues/623)

Is this know issue or what is possible root cause of the above error?

dorssel, 2022-07-16:
Have a look at https://github.com/dorssel/usbipd-win/wiki/Troubleshooting. Maybe a USB capture on the Windows side will show the root cause of the error.

lamaks, 2022-07-16:
Here is capture file and service trace output attached. There are several `URB status: Broken pipe (-EPIPE) (-32)` in a capture.
[usbipd-trace.txt](https://github.com/dorssel/usbipd-win/files/9126467/usbipd-trace.txt)
[coral.zip](https://github.com/dorssel/usbipd-win/files/9126468/coral.zip)

dorssel, 2022-07-16:
Hmmm.
The capture looks quite normal:

- 1-32: normal device setup, ending in a successful SET_CONFIGURATION, no errors
- 33-144: device specific control messages to endpoint 0 (some sort of device specific setup). All good.

Then interleaved endpoints 1, 2, and 3:

- From 145: some input endpoint 2, one request at a time. First response in 171, next request shortly after in 179 ... etc. etc. all without error
- 146: interrupt endpoint 3, never a reply (until aborted at the end). Also not unusual for interrupt endpoints.
- From 147: multiple input requests for main endpoint 1. Quite normal for bulk reads. They do get replies later, after which the queue of pending requests gets topped up. Again, all normal.
- From 155: bulk writes to main endpoint 1. All successful.

TL;DR: This looks all good. The EPIPEs are all from the aborts due to:

> Unbind or unplug while attached

At that time, the pending input requests all get aborted, which is why you are seeing EPIPE. So, the EPIPE is a consequence of the abort, which is a consequence of the device suddenly disappearing.

What kind of device is this? Are you flashing new firmware perhaps?

lamaks, 2022-07-17:
No new firmware update. This is [Coral USB device](https://coral.ai/products/accelerator/) I am just running the example from [this doc](https://coral.ai/docs/accelerator/get-started/) and getting the error at the end of the image detection. This is actually inline with EPIPE error at the end of the sequence 251 - 260. The device is still bind and connected after example run ends. The example can be run after that multiple times with the same error. No need to re-bind device. Also device successfully detects image from the example but errors at the end. This behavior is seen using other image detection app.

dorssel, 2022-07-17:
@lamaks 
Interesting ... It could be that `usbipd-win` is reporting the wrong return code upon abort/EPIPE at the end. My guess is that the abort in itself is normal (it is at the end of normal operation), but that the mapping to the right "error return" might be wrong. That subsequently confuses the software, which was expecting "abort value X", but got "abort value Y" instead.

Unfortunately, it is hard to guess the right return value without extensive debugging. Are you willing to do that?

Plan of action:
- get USB capture working on WSL. See https://github.com/dorssel/usbipd-win/issues/300#issuecomment-1066104831
- get a USB capture using WSL when things go wrong
- get a USB capture on a native/bare metal Linux machine, where things go right
- compare the difference

It is necessary to get the captures on the *Linux* side this time, not hard on native, but a bit of work for WSL.

dorssel, 2022-08-23:
@lamaks 
Any news?

poulti, 2022-09-17:
Hello,
I got the same issue with a Coral USB device on WSL2 Ubuntu 22.04 and accessed by a Frigate in a docker container.
Can I help diagnose the issue?

dorssel, 2022-09-19:
@poulti 
Can you do the plan in https://github.com/dorssel/usbipd-win/issues/423#issuecomment-1186507849?

poulti, 2022-09-28:
> @poulti Can you do the plan in [#423 (comment)](https://github.com/dorssel/usbipd-win/issues/423#issuecomment-1186507849)?

Still on my todo-list, lots on at the moment - bear with me :)

LennZA, 2022-09-29:
@dorssel @poulti 
Same issue as others.

I did the USB capture in WSL with the modified kernel a few times over the weekend. Here's the pcap shortest failure time from my attempts
[wsl2usbipd.zip](https://github.com/dorssel/usbipd-win/files/9675564/wsl2usbipd.zip)
Events captured in sequence
- start tshark
- attach coral
- run container using coral until failure
- detach coral
- stop tshark

Will not have time soon to get everything running on native linux, hopefully still somewhat useful.

dorssel, 2022-09-29:
@LennZA 
That's interesting. This looks very similar to the capture in https://github.com/dorssel/usbipd-win/issues/423#issuecomment-1186273480, except this is from the Linux point of view. And there are some extra errors (i.e., ones that come from Linux itself, but from `usbipd`):

At the end of the (successful) transfer, pipe number 3 returns ENOENT. This is *not* generated by `usbipd-win`, so it must come from the Linux kernel. Then another request is immediately put to that pipe again, which is normal. But that one gets an immediate response ESHUTDOWN. Again, this is *not* generated by `usbipd-win`, so it is from the Linux kernel itself.

ESHUTDOWN means:

> The device or host controller has been disabled due to some problem that could not be worked around, such as a physical disconnect.

However ... after these, the other pipes are aborted and each outstanding request returns (correctly) EPIPE, which *does* come from `usbipd-win` (and this is very similar to what can be seen in the Windows capture).

This is the weird part: Linux first determines ESHUTDOWN, but this is *not* due to the detach, which comes much later. The USBIP connection is still up and responding correctly at that time. So, the question is: why does the Linux kernel complete the outstanding pipe 3 request with ESHUTDOWN, even though the device is in fact still attached and responsive?

Aryankids, 2022-10-29:
@dorssel any news on this one, pls?

dorssel, 2022-10-30:
Your trace is identical to the original in https://github.com/dorssel/usbipd-win/issues/423#issuecomment-1186273480.
From the Windows side it all looks normal. We need to figure out what is causing this from the Linux side. See my comment https://github.com/dorssel/usbipd-win/issues/423#issuecomment-1262851628. Until we get some new insights there, there is no progress on this issue.

dorssel, 2023-09-12:
This is probably fixed by #684, which is in the latest 3.2.0 release. Can you confirm?

dorssel, 2023-10-20:
Stale

closed 2023-10-20 by dorssel