dorssel/usbipd-win #425: USB device is not supported

asuffke, 2022-07-22:
Hi,

When attaching the USB stick A to the latest ubuntu 22.04 distribution
running in wsl2 system, it gets suddenly disconnected from my PC and
the usbipd server shows up with an error. If I do the same with USB
stick B then everything works fine.  Could you please help me what is
going wrong here ?

**Error log:**
debug: Usbipd.ConnectedClient[1000]
       Received opcode: OP_REQ_IMPORT
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 292
         at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
         at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 76
         at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 82
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 263
dbug: Usbipd.Server[1000]
      connection closed

**USB Stick A:**
Name: SanDisk Extreme GO USB 3.1
VID:PID: 0781:558b
Storage Size: 64GB

**USB Stick B:**
Name: SanDisk PRO
VID:PID: 0781:5588
Storage Size: 256GB

Regards
Andreas

dorssel, 2022-07-22:
Unfortunately, this is yet another SanDisk device that apparently does not work.
See https://github.com/dorssel/usbipd-win/wiki/Tested-Devices.

You can try it in a USB2 port, instead of a USB3 port. That has helped with other SanDisk devices.

asuffke, 2022-07-25:
I spent half a day to find out how to attach the SanDisk Extreme GO USB stick to the WSL2 environment anyway. The workaround is really easy.

1. Executing the attaching device command like: "usbipd.exe wsl attach -b 4-1"
2. Waiting for the output: "usbipd: info: Using default distribution 'Ubuntu-22.0'."
3. Now manually disconnect and connect the device from the USB port
4. The device has been attached.

dorssel, 2022-08-02:
If you don't need to also use the device on Windows, you could try `usbipd bind --force`. That will probably make step 3 unnecessary (not sure). Anyways ... SanDisk flash drives are very weird devices. Some don't seem to handle a port reset, which is required to detach from Windows and then attack to Linux. The quick "unplug-replug" trick is a nice workaround. Thanks for the report.

dorssel, 2022-09-28:
OP found a workaround. Problems with SanDisk are already listed on the supported device list. Closing.

closed 2022-09-28 by dorssel

RedTail72, 2023-07-16:
@dorssel Not sure if this is worth opening a new issue but wanted to share some info on this.  Sadly, I found this issue while trying to find an existing issue to comment on, as I found I was able to get my SanDisk (Device ID 0781:5581) to work.  I had same errors as mentioned above and was stepping through the comments from #245 in an attempt to get it working.

I didn't have any luck and decided to clone the project and run it locally in debug.  On my first run, I set a breakpoint in [GetVBoxDevice](https://github.com/dorssel/usbipd-win/blob/0726f7f08e5a542ace4d5addcf3805ccf7e11e50/Usbipd/ConfigurationManager.cs#L292) and found `deviceInterfaces` was empty (which explained the error).  I decided to step farther back and set a breakpoint in [Get_Device_Interface_List](https://github.com/dorssel/usbipd-win/blob/0726f7f08e5a542ace4d5addcf3805ccf7e11e50/Usbipd/ConfigurationManager.cs#L48).  When I set it there, I found the list was populated for each call into the method.  It was a little slower, so I'm guessing it has something to do with the device driver being replaced.  

Like I said, this may be better to open under a separate issue, but thought I'd share what I found and get your input first.

dorssel, 2023-07-16:
@RedTail72 
Aha! So it may be timing related...
I want to try and fix this. Would be nice to have these devices work out-of-the-box. However, for the next few weeks I am on a holiday trip. Would you be willing to help out testing in a month or so, when I am back? Alternatively, could you send me such a device. You'll get it back once fixed.

RedTail72, 2023-07-17:
> @RedTail72 Aha! So it may be timing related... I want to try and fix this. Would be nice to have these devices work out-of-the-box. However, for the next few weeks I am on a holiday trip. Would you [...]

I'd be happy to help out with testing.  Just let me know when.  I'll also try sending a sample for you to keep (can't figure out which model I have), do you have a preferred mailing address?

dorssel, 2023-07-17:
@RedTail72 
Great. Please use my company office address in the Netherlands: https://www.nccgroupplc.com/who-we-are/our-office-locations/

Fox-IT
Attention: Frans van Dorsselaer
Olof Palmestraat 6
2616 LM Delft
The Netherlands

RedTail72, 2023-07-20:
@dorssel I went ahead and opened issue #659 to track this timing issue, rather than continuing to hijack this issue.  Also, assuming Amazon is accurate, you should have the SanDisk delivered to your office on August 3rd.