dorssel/usbipd-win #426: Serial device attached but not showing up as ttyS(n)?

jrj2211, 2022-08-02:
I've followed the steps to setup usbipd and attached my USB device but don't see it as mounted as ttyS device. 

In windows I attached it to my default WSL instance:
```
C:\Users\joe.jankowiak>usbipd wsl list
BUSID  VID:PID         DEVICE                                                             STATE
3-10     046d:c337    USB Input Device                                              Not attached
4-1       067b:2303    Prolific USB-to-Serial Comm Port (COM4)       Attached - Ubuntu
4-3       0a12:0001     Generic Bluetooth Radio                                 Not attached
```

In my WSL ubuntu I can see the device:
```
joe@ENGINEERING-03:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Checking the dmesg it shows the info on the USB device but does not say where its mounted such as /dev/ttyS1 or /dev/ttyACM0.

```
joe@ENGINEERING-03:~$ dmesg | tail
[32171.174415] vhci_hcd vhci_hcd.0: devid(262145) speed(2) speed_str(full-speed)
[32171.174425] vhci_hcd vhci_hcd.0: Device attached
[32171.460801] vhci_hcd: vhci_device speed not set
[32171.540811] usb 1-1: new full-speed USB device number 3 using vhci_hcd
[32171.620811] vhci_hcd: vhci_device speed not set
[32171.704748] usb 1-1: SetAddress Request (3) to port 0
[32171.792862] usb 1-1: New USB device found, idVendor=067b, idProduct=2303, bcdDevice= 4.00
[32171.792904] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[32171.792907] usb 1-1: Product: USB-Serial Controller D
[32171.792908] usb 1-1: Manufacturer: Prolific Technology Inc.
```

My WSL ubuntu instance lists ttyS0-ttyS4 and tty0-tty63 but none seem to be my actual USB serial adapter?
```
/dev/ttyS0
/dev/ttyS1
/dev/ttyS2
/dev/ttyS4
```

Pretty generic RS232 USB adapter:
https://www.trendnet.com/products/USB-adapters/TU-S9

dorssel, 2022-08-02:
That device uses the PL 2303X chip. The driver is not enabled in the WSL kernel by default:

```
# CONFIG_USB_SERIAL_PL2303 is not set
```

You need to compile your own kernel with this driver enabled, see https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel.

jrj2211, 2022-08-03:
Ahh I stopped reading after "already include support for common scenarios like USB-to-serial adapters" and assumed mine was included :) Thanks for the info!

I think I'd rather just buy adapters that work out of the box for our other developers than make them all rebuild WSL if possible. I see FTDI_SIO is enabled by default, does this mean any RS232 FTDI adapter would work? Such as:

https://ftdichip.com/products/usb-rs232-we-1800-bt_0-0/

dorssel, 2022-08-03:
As far as I know, all FTDI devices work out of the box.
Then again ... better to test one first instead of flying blind on my assumption ...

jrj2211, 2022-08-03:
Great, thanks for the help!

closed 2022-08-03 by jrj2211