dorssel/usbipd-win #429: Add Package for CONFIG_DEBUG_INFO_BTF to Kernel Building Instructions

DC4JG, 2022-08-11:
I suggest adding `dwarves` to the installation prerequisites.

---

Running Ubuntu 20.04 WSL2 with 5.10.102.1 kernel, I followed [the wiki](https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel) how to build an USB/IP enabled kernel.

At the step running `sudo make -j 8 && sudo make modules_install -j 8 && sudo make install -j 8` I got the following error message.

```sh
BTF: .tmp_vmlinux.btf: pahole (pahole) is not available
Failed to generate BTF for vmlinux
Try to disable CONFIG_DEBUG_INFO_BTF
make: *** [Makefile:1179: vmlinux] Error 1
```

Aunt Google [suggested](https://stackoverflow.com/questions/61657707/btf-tmp-vmlinux-btf-pahole-pahole-is-not-available) to either disable the `CONFIG_DEBUG_INFO_BTF` flag or install `dwarves`, which fixed the error.

PS: thank you for this tool and documentation ❤️

dorssel, 2022-08-11:
Thanks for the report.

Ping @benmcmorran

dreaquil, 2023-02-02:
Can confirm installing dwarves worked for me.

foureight84, 2023-02-09:
Installing `dwarves` fixes my issues too. I was trying to install pahole on debian wsl2 prior and it was complaining about libbpf0 version 0.4.0 not being available.