dorssel/usbipd-win #438: WSL Support Wiki instructions might lead to poor performance

glaucon1984, 2022-08-27:
Hi,

I wonder if it's possible to either remove the option "Device Drivers -> USB Support -> USB/IP -> Debug messages for USB/IP" from the kernel recompilation instructions [https://github.com/dorssel/usbipd-win/wiki/WSL-support], or at least put a note indicating that the option is only needed for debug/troubleshooting and it will lead to very poor performance for regular use.

In my case the performance went, from reading at 56 KB/s with USB/IP debug enabled, to 56 MB/s when recompiled with it disabled.

dorssel, 2022-08-27:
Thanks for the report.

I guess Microsoft really tested only for slow, serial devices. Although the instructions are not mine (they are more-or-less maintained by the WSL team), I took the liberty to change them. Can you review the result (as in: if you would encounter this for the first time, would it be the right wording?) Thx.

glaucon1984, 2022-08-27:
Yeah, definitely agree with the edit, the people able to debug USB issues probably know already what do do about this option, for the rest of us it will be much easier this way.

closed 2022-08-27 by dorssel