dorssel/usbipd-win #439: Cannot set udev rules file for software defined radio USB stick

mwickert, 2022-08-28:
I am trying to use WSLg on Win11 with GNU Radio Companion (GRC) and in due time multiple USB software defined radio (SDR) devices. I am running Ubuntu 20.04. I want to run as a USER (not root) but in spite of placing a rules file I get usb_open error -3. See details below.

**Background**
GRC is installed and the GUI works great. Running this app sudo is not my desire and seems to present issues of its own. The first device I am working with is the RTL-SDR [sample instructables link](https://www.instructables.com/rtl-sdr-on-Ubuntu/#:~:text=Rtl-sdr%20on%20Ubuntu%201%20Step%201%3A%20Install%20Gnu-radio,...%204%20Step%204%3A%20Test%20It%20Out%20)  or [RTLSDR rules site](https://github.com/keenerd/rtl-sdr/blob/master/rtl-sdr.rules).

**Connecting the Device**
usbipd works great here:
*From Powershell -*
PS C:\Users\mwick> usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
1-2    046d:c52b  Logitech USB Input Device, USB Input Device                   Not attached
1-4    8087:0a2b  Intel(R) Wireless Bluetooth(R)                                Not attached
1-9    04f3:24a1  USB Input Device                                              Not attached
1-12   1bcf:2b95  Integrated Webcam                                             Not attached
3-1    0bda:2838  Bulk-In, Interface                                            Not attached

PS C:\Users\mwick> usbipd wsl attach --busid 3-1

*Bash --*
wickert@prec5520:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

*Test with a Sample RTLSDR App, and Receive Errors Similar to Running GRC*
For the RTLSDR install the package: `sudo apt-get install rtl-sdr`. I place a rules file `/lib/udev/rules.d/10-rtl-sdr.rules`.
Now I run a test program as a user and receive usb open error -3:
wickert@prec5520:~$ rtl_test
Found 1 device(s):
  0:  , , SN:

Using device 0: Generic RTL2832U OEM
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
Failed to open rtlsdr device #0.

Now I run the test program as root and the sample program runs as expected:
wickert@prec5520:~$ sudo rtl_test
[sudo] password for wickert:
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
[R82XX] PLL not locked!
Sampling at 2048000 S/s. ...
...

As a cross check, I perform these same steps on my MacBook with an Ubuntu 20.04 VM and placing the same rules file works as expected. I am expecting greater throughput from the USB under WSLg vs the VM, so I am excited to get this issue resolved. I also have more USB SDR devices waiting in the wings, especially the ADALM-Pluto SDR, which also mounts properly, but has different errors when trying to connect using GRC (I am hold back on explaining this until the RTLSDR issues is resolved.)

dorssel, 2022-08-28:
WSL does not start the udev service itself. Also, the (new) udev rules only take effect at the moment the device is first seen by udev. I think the order of things should be:
1) put udev rules in place
2) start udev
3) attach USB device to WSL

mwickert, 2022-08-29:
Thank you very much for this pointer. I figured it should be obvious to someone, but not to me. To implement step 2 I added a file wsl.conf in /etc/:
```
[boot]
command="service udev start"
```
This worked great for me! Let me know if this a reasonable solution.

mwickert, 2022-08-29:
@dorssel I was able to get a second SDR running under GRC by installing it's rule file. Thanks again for your help and rapid response.

dorssel, 2022-08-29:
> ```
> [boot]
> command="service udev start"
> ```
> 
> This worked great for me! Let me know if this a reasonable solution.

I've seen other people do similar auto-startup things. It probably is distro-specific, but for Ubuntu(-like) distros it is the right thing to do.

closed 2022-08-29 by dorssel