dorssel/usbipd-win #445: usbipd-win installation in windows 10 caused loss of user profile

preenu91, 2022-09-06:
After building own USB/IP enabled WSL 2 kernel in Windows 10, I lost access to my windows user profile. Windows messed up my login process and I got signed into a temporary profile.

preenu91, 2022-09-06:
Great. Thanks. Why is the installation messing up my original profile? Could you please provide some insights into this

dorssel, 2022-09-10:
This is unrelated to `usbipd-win`, which doesn't do anything with your profile. And I've also never heard of WSL messing with the user profile either, but you could ask the WSL project at https://github.com/microsoft/WSL.

dorssel, 2022-10-25:
@preenu91 Any news? If not, I will close this issue as stale.

preenu91, 2022-11-03:
@dorssel Yes , please close it.

closed 2022-11-08 by dorssel