dorssel/usbipd-win #447: XBox360 Controller Not Accessible

Diztruxion, 2022-09-09:
Hi,

I'm attempting to get a XBox360 USB controller to pass from Windows 10 into WSL2 using Ubuntu 18.04 LTS to create inputs to a Robot Operating System (ROS) simulation environment.

So far I've been successful in installing all required packages, finding the device in PS with usbipd wsl list. I've successfully attached the device using the busid, and I can see in the Ubuntu terminal that it has registered the device as a controller:

[image]

However, I'm not sure how the device should be accessible now. I can see that the device is created when I call lsusb:
[image]

From my reading, the developer of the simulation environment indicates that these controllers should be accessible using the jstest-gtk tool, and registers in /dev/input/js###

I know this might be a bit outside the scope of the windows side, just wondering if anyone has seen this before.

Thanks.

dorssel, 2022-09-09:
From https://www.kernelconfig.io/config_joystick_xpad it looks like you need a special driver (xpad), which needs to be enabled in the kernel (which it isn't in the default WSL kernel).

Follow https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel and build your own kernel with this driver enabled.

dorssel, 2022-09-09:
And you may also like https://www.kernelconfig.io/CONFIG_JOYSTICK_XPAD_FF

agaertner, 2022-10-19:
I actually did that. Compiled my own linux kernel with xbox drivers enabled and I still get 
``usbip: error: Attach Request for 1-2 failed - Device busy (exported)```
when trying to attach 
``Xbox 360 Controller for Windows``

dorssel, 2022-10-19:
"Device busy" means exactly what it says. Looks like something on the host (service/software/driver) is holding an exclusive lock. Either stop that service/software, or use `usbipd bind --force`.

agaertner, 2022-10-19:
Obviously yes. The service that has exclusive lock is probably one of those xbox services. I didn't bother checking because after badly wanting the ``--force`` argument which I thought didn't exist because I overlooked it I had decided on using the other controller though.😅 Thanks for the hint.

dorssel, 2022-11-02:
@Diztruxion 
Has your problem been resolved?

Diztruxion, 2022-11-02:
@dorssel 
Thank you for the follow-up; I've been a bad user, and should have closed this. We decided to go with another solution that avoided this challenge entirely so I have not followed up on it.

closed 2022-11-02 by dorssel

Doom4535, 2022-12-05:
I'm having this issue as well; I'm using a Logitech F310 USB gamepad (currently showing up as `3-13   046d:c21d  Xbox 360 Controller for Windows` ) and running `usbipd wsl attach --busid <BUSID>` returns:
```
usbip: error: Attach Request for 3-13 failed - Device busy (exported)

usbipd: error: Failed to attach device with busid '3-13'.
```

However, I am able to run `usbipd bind --busid <BUSID>` (after unbinding the controller) without errors; but I am unsure how to then attach it to WSL.