dorssel/usbipd-win #448: Attach USB(COM port) failed and it disappeared from  usbipd list

NINJA-J, 2022-09-16:
Hi,

I'm trying to use wsl2 to read COM port. I followed [microsoft usbipd tutorial](https://docs.microsoft.com/en-us/windows/wsl/connect-usb) to install usbipd and run the following cmd.
```
usbipd wsl list
    BUSID  VID:PID    DEVICE                                                        STATE                                           
    2-3    1bcf:28cc  Integrated Webcam                                             Not attached                            
    2-11   067b:2303  Prolific USB-to-Serial Comm Port (COM3)                       Not attached                            
    2-14   0cf3:e007  Qualcomm QCA61x4A Bluetooth                                   Not attached       
usbipd wsl attach --busid 2-11                                                                  
    usbip: error: Attach Request for 2-11 failed - Device in error state                                                                                                                                                                            
    usbipd: error: Failed to attach device with busid '2-11'.
usbipd wsl list
    BUSID  VID:PID    DEVICE                                                        STATE
    2-3    1bcf:28cc  Integrated Webcam                                             Not attached
    2-14   0cf3:e007  Qualcomm QCA61x4A Bluetooth                                   Not attached
```
Attach COM3 failed and it disappeared from upbipd wsl list.

This COM port works under Microsoft's Serial Debug Assistant.

Do you know what is happening?

dorssel, 2022-09-18:
Some devices and USB hubs don't like the "soft unplug/replug" that must be done to replace the driver. Essentially what usbipd does is during attach is:
- simulate an unplug
- let the vbox monitor driver mock the hardware id (such that it appears to be a different device altogether
- simulate a replug
- now the device has the vbox hardware id and the vbox stub driver handles it (instead of the original driver)

Unfortunately, not all devices (and sometimes the hub that it is connected to) respond too well. In that case, you will have to use `usbipd bind --force`. This will permanently let the vbox stub driver kick in *for the original device id*. This has 2 side effects:
- the vbox monitor driver is no longer involved (that's actually a good thing)
- you cannot use the device under Windows until you use `usbipd unbind`

I am always interested in *why* your device doesn't work the normal way. What does the Windows device manager show after the failure? Is the device really gone? Or does it show a yellow bang with an error code?

Did you get any other warnings during `usbipd wsl attach --busid 2-11`? Was any Windows software using the device when you tried to attach? Is the hub that the device is connected to using the stock Windows drivers?

NINJA-J, 2022-09-21:
The device just disappear from everywhere.

closed 2022-09-21 by NINJA-J

NINJA-J, 2022-09-21:
The device is attached now, not by using `usbipd bind --forse`. I didn't know what happend. There were a bunch of problems about wsl2 lately on my computer. They just recovered after some unknow operations. I'll let you know if it happens again!