dorssel/usbipd-win #450: Invalid number_of_packets in ret_submit

vadimgrn, 2022-09-21:
Hello, according to usbip protocol, number_of_packets for USBIP_CMD_SUBMIT and USBIP_RET_SUBMIT shall be set to 0xffffffff if not ISO transfer.
But de facto linux server never set this field to -1, it uses zero. This is just outdated protocol document. 

Your servers sets number_of_packets to -1 for ret_submit (as protocol says, but linux never does), but doesn't accept this value in cmd_submit (violation of the protocol).

The right way is set number_of_packets to zero for non isoch transfers as Linux client and server do.

dorssel, 2022-09-25:
Thanks for the report.

> but doesn't accept this value in cmd_submit (violation of the protocol)

It does since version 2.3.0. What version did you test?

Linux actually does worse. For non-ISO transfers it doesn't touch the value at all. For freshly allocated URBs, the default initialization value is 0. Most drivers have a pool of URBs, and most drivers are either ISO, or non-ISO. But some drivers use a pool for both, and then the value is whatever it was for the previous URB. This is why I changed `usbipd` to ignore the value on input altogether for non-ISO transfers.

Besides that, I know of at least one client that actually expects -1 on return. That's why `usbipd` returns that; it gets ignored by Linux anyway.

Have a look at PR #454, I think that will accomplish maximum compatibility.

vadimgrn, 2022-09-26:
Thank you. I tested the latest release.

PR #454 (2022-09-28): Fix number_of_packets
Fixes #450
closed 2022-09-28 by dorssel