dorssel/usbipd-win #451: usbipd-win on Windows 10 using WSL2

toncho11, 2022-09-22:
Hi,

Would the usbipd-win library work on Windows 10 with WSL2? I see [this article](https://www.xda-developers.com/wsl-connect-usb-devices-windows-11/) says that it works on Windows 11, but it is not clear about Windows 10. My objective is to use Bluetooth device inside WSL2 on Windows 10. We do not have Windows 11 right now.

dorssel, 2022-09-22:
The specific mention of Windows 11 originates from Microsoft, in their original article https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/. At the time a) they wanted to push Windows 11, and b) the WSL2 kernel for Windows 11 was not backported to Windows 10 yet.

However, `usbipd` works from Windows 8.1/Server 2012 and up. And WSL2 is available from Windows 10 (cannot remember from which update, could be 1903). It is true that newer WSL2 kernels are released for Windows 11 first, but by now the most up-to-date Windows 10 WSL2 kernel supports USBIP just fine.

Do note that in any case you will have to rebuild the kernel to include Bluetooth drivers, as the default build does not include them.

See https://github.com/dorssel/usbipd-win/discussions/310 for a nice write-up.

dorssel, 2022-09-28:
The short answer is: yes, this is possible,

closed 2022-09-28 by dorssel