dorssel/usbipd-win #452: Attaching one among multiple instances of the same device fails

vszakd, 2022-09-23:
Hello,
if I have two identical device connected (same VID/PID) only one of the two succeeds in attaching to WSL2. 
It does not depend on the order of the attach (e.g.: if I have dev 1 and dev 2, the one succeeding is always the same).
If I physically disconnect one of the two and try attaching the remaining one, everything works OK.

Versions: 
```
PS C:\WINDOWS\system32> usbipd --version
2.3.0+42.Branch.master.Sha.3d9f5c5acc4e133ab8147684ad1463cbaec43240
```
```
user@pc:~$ uname -r
5.15.57.1-microsoft-standard-WSL2
```

```
PS C:\WINDOWS\system32> [System.Environment]::OSVersion.Version

Major  Minor  Build  Revision
-----  -----  -----  --------
10     0      22000  0
```

Attached devices:
```
PS C:\WINDOWS\system32> usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
13-2   10c4:ea60  Silicon Labs CP210x USB to UART Bridge (COM7)                 Not attached
13-3   10c4:ea60  Silicon Labs CP210x USB to UART Bridge (COM12)                Not attached
```

In this case, e.g. 13-3 will always work. 13-2 never (or viceversa, I could not find a pattern). 

The returned error is:
```
PS C:\WINDOWS\system32> usbipd wsl attach --busid 13-2
usbipd: info: Using default distribution 'Ubuntu-20.04'.
usbip: error: Attach Request for 13-2 failed - Device in error state

usbipd: error: Failed to attach device with busid '13-2'.
```

Thank you.

dorssel, 2022-09-25:
Thanks for the report. I knew eventually someone would find a problem like this...

The root cause is VBoxUsbMon, the VirtualBox driver that does the filtering. It has no way of filtering on USB hub number, and the filtering on port number is "wrong". The definition of port number has been a problem in the past (see #82).

This means that, essentially, VBoxUsbMon can only filter reliably on VID, PID, and device/interface class. For identical devices, it will likely filter either both or none. Nevertheless, I would like to try and resolve this. VBoxUsbMon *is* capable of filtering on port number, if only I would know for what definition of port number...

Can you test the following:
- run `usbipd` with logging, as explained in https://github.com/dorssel/usbipd-win/wiki/Troubleshooting
- attach 13-3 (which works, as you say) and go to device manager and see how many "USBIP Shared Device"s there are. There should be one. If two, then VBoxUsbMon indeed captured both (for a single attach).
- detach 13-3 again. Are all "USBIP Shared Device"s gone in device manager?
- with device manager open, try to attach 13-2 (which does not work, as you say). Does device manager "update" its view, i.e. is there any indication that one or more devices are reconfigured at all during the attempt?

As a workaround, you can use `usbipd bind --force`. This bypasses VBoxUsbMon altogether and uses Windows DeviceIDs internally. This should work properly for telling the two apart. That should also allow you to attach 13-3, and even both 13-2 and 13-3 at the same time if you want that.

dorssel, 2022-09-28:
I have tested this with 2 ATEN USB-to-serial bridges with `usbipd` version 2.3.0. They are entirely identical.

I cannot reproduce the problem. Both devices can be attached and detached, in any order, to both a normal Linux client as well as WSL.

This must be something specific for your device/hub/installation. I'm happy to try and figure out what is going on, but since I cannot reproduce the problem, you will have to provide the debug info.

vszakd, 2022-10-13:
Hello dorssel, thanks for you reply. I will run the troubleshooting steps you suggested and update the thread.

vszakd, 2022-10-18:
Hello,
I managed to run the tests you suggested. Results follow.

### Test 1

> attach 13-3 (which works, as you say) and go to device manager and see how many "USBIP Shared Device"s there are. There should be one. If two, then VBoxUsbMon indeed captured both (for a single attach).

There is only one "USBIP Shared Device".

This is the log for a successful attach: 

> dbug: Usbipd.Server[1000]
>       new connection from 172.19.21.190
> dbug: Usbipd.Server[1000]
>       connection closed
> dbug: Usbipd.Server[1000]
>       new connection from 172.19.21.190
> dbug: Usbipd.ConnectedClient[1000]
>       Received opcode: OP_REQ_IMPORT
> info: Usbipd.ConnectedClient[1]
>       Client 172.19.21.190 claimed device at 13-1 (USB\VID_10C4&PID_EA60\0001).
> dbug: Usbipd.AttachedClient[1000]
>       Trapped SET_CONFIGURATION: 1

### Test 2

> detach 13-3 again. Are all "USBIP Shared Device"s gone in device manager?

Yes.

### Test 3

> with device manager open, try to attach 13-2 (which does not work, as you say). Does device manager "update" its view, i.e. is there any indication that one or more devices are reconfigured at all during the attempt?

It flashes a couple of times, but nothing seems to be added/removed (at least under "Universal Serial Bus controllers").

Here is the log for a failed attach: 

```
dbug: Usbipd.Server[1000]
      new connection from 172.19.21.190
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.19.21.190
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 292
         at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
         at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 76
         at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 82
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 158
dbug: Usbipd.Server[1000]
      connection closed	  
```

### Workaround

> As a workaround, you can use usbipd bind --force. This bypasses VBoxUsbMon altogether and uses Windows DeviceIDs internally. This should work properly for telling the two apart. That should also allow you to attach 13-3, and even both 13-2 and 13-3 at the same time if you want that.

It does not work. The forced bind works, and the status of `usbipd wsl list` is (13-1 is the one that works):

```
13-1   10c4:ea60  Silicon Labs CP210x USB to UART Bridge (COM12)                Attached
13-2   10c4:ea60  Silicon Labs CP210x USB to UART Bridge (COM12)                Shared (forced)
```

But issuing `usbipd wsl attach --busid 13-2` yields the infamous error. And the device is not present in WSL.

Let me know if there are other tests I can help with.

Thanks.

dorssel, 2022-10-18:
It looks like there may be a problem with detection of the port number of the bus. Can you try connecting 1 of the 2 devices to a different hub (maybe another port on your pc, or else connect an external hub)? Only temporarily, just to narrow down the problem. For this test, you should have the 2 devices connected to the PC somehow with different *bus* numbers *and* different *port* numbers (e.g. 13-1 and 14-2). Does it work then?

dorssel, 2022-10-18:
Just to be sure ... do you, or do you not, have VirtualBox installed? If so, there may be a driver version conflict... `usbipd-win` can, in theory, co-exist with a VirtualBox installation, but that has certainly not been tested extensively. And it has caused problem for others. Just asking...

dorssel, 2022-10-18:
Going back to 13-1 and 13-2, if you physically exchange the devices, does Windows assign the same COM-port number to the same device, or to the same port?

Test: if you first plug device A in 13-1 and it (for example) gets assigned COM12, and subsequently device B in 13-2 and it (for example) gets assigned COM14. If you then unplug both devices and first plug in device B in 13-1 and then device A in 13-2, how are the COM ports assigned?

jamesadevine, 2022-11-29:
@dorssel sorry to crop up again, but I am seeing a similar issue to @vszakd. Two of the same device do not seem to be able to be claimed regardless of their bus IDs. 

However, in my case, connection order seems to be important. The first device connected can be reliably claimed, the second connected device always fails with:  

```
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 289
         at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
         at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 76
         at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 82
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 165
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 266
         at Usbipd.ConnectedClient.RunAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 58
```
Disconnecting the first device allows the second device to be claimed reliably.

This is probably something to do with what you said earlier in this thread:

> This means that, essentially, VBoxUsbMon can only filter reliably on VID, PID, and device/interface class. For identical devices, it will likely filter either both or none. Nevertheless, I would [...]

Happy to open a separate issue if you feel it is a different error modality.

toncho11, 2022-12-14:
I have the same problem. I have 4 usb bluetooth devices.

version usbipd: 2.3.0+42.Branch.master.Sha.3d9f5c5acc4e133ab8147684ad1463cbaec43240
Windows:  Microsoft Windows 10 Professionnel 10.0.19042 N/A version 19042

toncho11, 2022-12-14:
I tried with usbipd bind --force --busid 4-2 (4-2 being my second dongle) and it did not make a difference.
Also each time it fails Windows makes the sound pf connecting/disconnecting which is not the case when it works without problem with the first dongle.

dorssel, 2022-12-14:
@toncho11 Can you try this test: https://github.com/dorssel/usbipd-win/issues/452#issuecomment-1282439961?

toncho11, 2022-12-15:
> @toncho11 Can you try this test: [#452 (comment)](https://github.com/dorssel/usbipd-win/issues/452#issuecomment-1282439961)?

I already tried on different USB ports. I think it is important that the first number in the BUSID is different, right? But even in this case when I have for example 7-2 and 2-14 it still does not work.

dorssel, 2022-12-15:
@toncho11 @vszakd 
The problem is that I have a similar (but obviously not identical) setup, with 2 identical USB-to-serial converters, and it works as intended. And I've run out of ideas to get more information. Is any of you willing to ship me the two devices, so I can try to reproduce and fix the problem? Of course, I'll ship them back when done.

jamesadevine, 2022-12-15:
In my experience the issue seems to be more specific to PC setup than duplicate identical devices. A colleague with a different laptop is able to attach two of the same device just fine. 

My laptop is a Lenovo Yoga X1 for reference.

toncho11, 2022-12-15:
> @toncho11 @vszakd The problem is that I have a similar (but obviously not identical) setup, with 2 identical USB-to-serial converters, and it works as intended. And I've run out of ideas to get more [...]

Yes. @dorssel Can you send me your address? I would prefer to send them today.

dorssel, 2022-12-15:
@toncho11 
Great!
Use my work address, https://www.fox-it.com/nl-en/contact-us/
Attention: Frans van Dorsselaer

dorssel, 2022-12-16:
@jamesadevine 
Interesting. Are you saying you yourself can reproduce this problem on one machine and (with the same two devices) it works on another?

If so, I have some questions about those machines (and mostly their differences):

1) What is the OS (windows 10, 11 ?)
2) What type of USB ports (USB2 or USB3)
3) Does `usbipd` report anything about USB filters?
4) What is the manufacturer of the USB root?
5) Is the hub-driver (the root hub and any hubs in between the root and the device) proprietary or Microsoft?
6) Do you have VirtualBox installed, or was it ever installed?

jamesadevine, 2022-12-16:
@dorssel The devices in question are Azure Sphere devices which have an FTDI 4232H chip to communicate between device and PC. To the PC, there is no difference between the USB interface exposed by the FTDI chip. The PC always sees that same metadata.

When connecting two devices to my Lenovo laptop, I am not able to connect two azure sphere devices to WSL. The behaviour is exactly the same as @toncho11 reports. My colleague using a different laptop, is able to connect two azure sphere devices without issue. Unfortunately, they have taken time off for Christmas so won't be around to supply the requested diagnostic information.

__My information__:

> What is the OS (windows 10, 11 ?)

11

> What type of USB ports (USB2 or USB3)

On the PC, they are all USB3. When testing, one device was connected through a USB 2.0 hub to my display which connects over USBC to my PC.

> Does usbipd report anything about USB filters?

Here is another round of debug:

```

> What is the manufacturer of the USB root?

<img width="252" alt="image" src="https://user-images.githubusercontent.com/4742697/208079899-13d8fd67-ac27-47eb-9017-5adad741eb28.png">

Looks to be quite a generic manufacturer. Is there a more specific descriptor you would like me to look at?

> Is the hub-driver (the root hub and any hubs in between the root and the device) proprietary or Microsoft?

I tested this multiple ways. With both devices connected directly to my PC (no intermediate hubs), the error persists. With one device connected via my display and the other via a root hub, the error persists.

> Do you have VirtualBox installed, or was it ever installed?

I can't quite remember if I installed it or not. I have the installation exe downloaded, but I am not sure I ever needed to install VBox. It is not in my list of installed programs.

dorssel, 2022-12-16:
> Is the hub-driver (the root hub and any hubs in between the root and the device) proprietary or Microsoft?

Can you open the hub(s) that the devices are connected to in Windows device manager and verify that they are using the stock Microsoft hub driver?

Like so (this is Windows 10), for all hubs between the root and your device:

[image]

Can you run the following to confirm there are no third party USB filter driver installed:

```
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
```

jamesadevine, 2022-12-16:
All hubs are using the standard Microsoft Hub driver.

```
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters

ERROR: The system was unable to find the specified registry key or value.
```

dorssel, 2022-12-16:
@jamesadevine 

OK, so all Microsoft drivers, no third party filters.

Now lets rule out (or in!) if it is the VirtualBox filter driver. Can you individually `bind --force` the two devices?
Test: 

```
# cleanup
usbipd list
usbipd unbind -b <everything currently shared> (repeat)

# test device 1
usbipd list
usbipd bind -f -b <device 1>
usbipd list
usbipd unbind -b <device 1>
usbipd list

# test device 2
usbipd list
usbipd bind -f -b <device 2>
usbipd list
usbipd unbind -b <device 1>
usbipd list

# test device 1 + 2
usbipd list
usbipd bind -f -b <device 1>
usbipd list
usbipd bind -f -b <device 2>
usbipd list
usbipd unbind -b <device 1>
usbipd list
usbipd unbind -b <device 2>
usbipd list

# test device 2 + 1
usbipd list
usbipd bind -f -b <device 2>
usbipd list
usbipd bind -f -b <device 1>
usbipd list
usbipd unbind -b <device 2>
usbipd list
usbipd unbind -b <device 1>
usbipd list
```

jamesadevine, 2022-12-16:
@dorssel I turned your test steps into a powershell script if someone else needs to re-run the steps:

```
$dev1 = "4-4"
$dev2 = "2-7"

# test device 1
usbipd list
Write-Output "usbipd bind -f -b $dev1 :"
usbipd bind -f -b $dev1
usbipd list
Write-Output "usbipd unbind -b $dev1 :"
usbipd unbind -b $dev1
usbipd list

# test device 2
usbipd list
Write-Output "usbipd bind -f -b $dev2 :"
usbipd bind -f -b $dev2
usbipd list
Write-Output "usbipd unbind -b $dev2 :"
usbipd unbind -b $dev2
usbipd list

# test device 1 + 2
usbipd list
Write-Output "usbipd bind -f -b $dev1 :"
usbipd bind -f -b $dev1
usbipd list
Write-Output "usbipd bind -f -b $dev2 :"
usbipd bind -f -b $dev2
usbipd list
Write-Output "usbipd unbind -b $dev1 :"
usbipd unbind -b $dev1
usbipd list
Write-Output "usbipd unbind -b $dev2 :"
usbipd unbind -b $dev2
usbipd list

# test device 2 + 1
usbipd list
Write-Output "usbipd bind -f -b $dev2 :"
usbipd bind -f -b $dev2
usbipd list
Write-Output "usbipd bind -f -b $dev1 :"
usbipd bind -f -b $dev1
usbipd list
Write-Output "usbipd unbind -b $dev2 :"
usbipd unbind -b $dev2
usbipd list
Write-Output "usbipd unbind -b $dev1 :"
usbipd unbind -b $dev1
usbipd list
```

I ran the script with the Out-File cmdlet (e.g. `test.ps1 | Out-File "test-output.txt"`), which gave:
```
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-3    06cb:00fc  Synaptics UWP WBDI                                            Not shared
2-4    174f:2454  Integrated Camera, Integrated IR Camera                       Not shared
2-5    045e:07a5  USB Input Device                                              Not shared
2-7    0403:6011  USB Serial Converter A, MSFT MT3620 Std Interface, USB Se...  Not shared
2-10   8087:0026  Intel(R) Wireless Bluetooth(R)                                Not shared
3-2    045e:0840  Microsoft Modern Webcam, USB Input Device                     Not shared
3-5    0451:82ff  USB Input Device                                              Not shared
4-4    0403:6011  USB Serial Converter A, MSFT MT3620 Std Interface, USB Se...  Not shared

Persisted:
GUID                                  DEVICE
[... 246 lines cut ...]

Persisted:
GUID                                  DEVICE
2d7494ec-0704-4301-baf3-696bc32884f5  USB Serial Converter A, MSFT MT3620 Std Interface, USB Se...
65aedc1c-b409-4597-afd2-24b2b34e5075  USB Serial Converter A, MSFT MT3620 Std Interface, USB Se...
ffdead05-cf6d-40d9-a286-4b7f5b84ca82  USB Serial Converter A, MSFT MT3620 Std Interface, USB Se...

usbipd unbind -b 4-4 :
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-3    06cb:00fc  Synaptics UWP WBDI                                            Not shared
2-4    174f:2454  Integrated Camera, Integrated IR Camera                       Not shared
2-5    045e:07a5  USB Input Device                                              Not shared
2-7    0403:6011  USB Serial Converter A, MSFT MT3620 Std Interface, USB Se...  Not shared
2-10   8087:0026  Intel(R) Wireless Bluetooth(R)                                Not shared
3-2    045e:0840  Microsoft Modern Webcam, USB Input Device                     Not shared
3-5    0451:82ff  USB Input Device                                              Not shared
4-4    0403:6011  USB Serial Converter A, MSFT MT3620 Std Interface, USB Se...  Not shared

Persisted:
GUID                                  DEVICE
2d7494ec-0704-4301-baf3-696bc32884f5  USB Serial Converter A, MSFT MT3620 Std Interface, USB Se...
65aedc1c-b409-4597-afd2-24b2b34e5075  USB Serial Converter A, MSFT MT3620 Std Interface, USB Se...
ffdead05-cf6d-40d9-a286-4b7f5b84ca82  USB Serial Converter A, MSFT MT3620 Std Interface, USB Se...
```

toncho11, 2022-12-16:
> @toncho11 Great! Use my work address, https://www.fox-it.com/nl-en/contact-us/ Attention: Frans van Dorsselaer

@dorssel I have sent you the two USB Bluetooth dongles Edimax.

dorssel, 2022-12-17:
@jamesadevine 
That looks good! Can you force-bind both devices and then attempt to attach them (one, the other, both)?

jamesadevine, 2022-12-19:
@dorssel same issue!

```
PS C:\software\usbipd-debugging> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-1    045e:07a5  USB Input Device                                              Not shared
2-3    06cb:00fc  Synaptics UWP WBDI                                            Not shared
2-4    174f:2454  Integrated Camera, Integrated IR Camera                       Not shared
2-7    0403:6011  USB Serial Converter A, MSFT MT3620 Std Interface, USB Se...  Shared (forced)
2-10   8087:0026  Intel(R) Wireless Bluetooth(R)                                Not shared
3-2    045e:0840  Microsoft Modern Webcam, USB Input Device                     Not shared
3-5    0451:82ff  USB Input Device                                              Not shared
4-4    0403:6011  USB Serial Converter A, MSFT MT3620 Std Interface, USB Se...  Attached

Persisted:
GUID                                  DEVICE
2d7494ec-0704-4301-baf3-696bc32884f5  USB Serial Converter A, MSFT MT3620 Std Interface, USB Se...l
65aedc1c-b409-4597-afd2-24b2b34e5075  USB Serial Converter A, MSFT MT3620 Std Interface, USB Se...l
ffdead05-cf6d-40d9-a286-4b7f5b84ca82  USB Serial Converter A, MSFT MT3620 Std Interface, USB Se...l

PS C:\software\usbipd-debugging> usbipd wsl attach -b 2-7
usbipd: info: Using default WSL distribution 'Ubuntu'; specify the '--distribution' option to select a different one.
usbip: error: Attach Request for 2-7 failed - Device in error state

usbipd: error: Failed to attach device with busid '2-7'.
PS C:\software\usbipd-debugging> usbipd wsl attach -b 4-4
usbipd: info: Using default WSL distribution 'Ubuntu'; specify the '--distribution' option to select a different one.
```

toncho11, 2022-12-19:
This is the result of the script on my computer . I am testing on my two "Edimax Wi-Fi N150" devices. I also have a third device which is the built-in BT in laptop: "Intel(R) Wireless Bluetooth"
[result_test_usb_bt.txt](https://github.com/dorssel/usbipd-win/files/10258777/result_test_usb_bt.txt)

toncho11, 2022-12-19:
This is the result of the script of @jamesadevine on my computer . I am testing on my two "Edimax Wi-Fi N150" devices. I also have a third device which is the built-in BT in my laptop: "Intel(R) Wireless Bluetooth"

[result_test_usb_bt3.txt](https://github.com/dorssel/usbipd-win/files/10259660/result_test_usb_bt3.txt)

dorssel, 2022-12-26:
@toncho11 
Thanks for sending the devices. TL;DR: I tracked down the problem, and fixed it!

Here is the elaborate report. Maybe I'll put it in Wiki at some point, or write a blog about it, or so.

The devices are identical. In fact **_too much_** identical. They report the same unique (?) serial number! Here is what happens when you plug one device in: Windows asks for the VID:PID (in your case 7392:a611). Then it reads the string table to find out if the device reports a unique serial number.

a) If there is a serial number, then Windows decides the device is uniquely identifiable and assigns an InstanceID like so: `USB\\VID_7392&PID_A611\\00E04C000001`, where the last bit is the serial number. This means no matter in which port you plug the device, Windows recognizes it as that one device.

b) If there is no serial number , then Windows decides the device is not uniquely identifiable and assigns an InstanceID like so: `USB\\VID_2149&PID_270B\\6&17F32605&0&1` (this is one of my devices), where the last bit represents the hub and port (related to BusId). This means that Windows sees it as a different device in different ports.

What happens with your devices is that the first device you plug in will be uniquely identified by its serial number and it appears immediately. But when you plug in the second device, Windows gets really confused, as the device reports *the same* 'unique' serial number. Windows then refuses to use that serial number for the second device and acts as if it does not have one at all. However, this takes approximately 5 seconds, i.e. the OS only exposes the device after 5 seconds.

When `usbipd` does an attach, it needs to reset the device to have the stub driver take over. This reset takes again 5 seconds for the second device (the first device, like any other device I have seen, takes almost no time at all). As if Windows is rediscovering and handling the duplicate serial number again. And how long does `usbipd` wait for the device to appear? 5 seconds... Every now and then it works, but most of the time it does not. I just increased the waiting time to 10 seconds, and voila!

So, in summary, Edimax messed up. They should either have put truly unique serial numbers on the devices, or none at all. Windows can deal with it, at the cost of 5 seconds delay for the second (duplicate) device. This does not happen at all for identical devices without a serial number, only devices that lie about having a unique serial number.

Here is the output of UsbView for the first device plugged in (note it doesn't matter which specific device, they are truly identical):

```
          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0210
bDeviceClass:                      0xEF  -> This is a Multi-interface Function Code Device
bDeviceSubClass:                   0x02  -> This is the Common Class Sub Class
bDeviceProtocol:                   0x01  -> This is the Interface Association Descriptor protocol
bMaxPacketSize0:                   0x40 = (64) Bytes
idVendor:                        0x7392 = Vendor ID not listed with USB.org
idProduct:                       0xA611
bcdDevice:                       0x0200
iManufacturer:                     0x01
     English (United States)  "Realtek"
iProduct:                          0x02
     English (United States)  "Edimax Wi-Fi N150 Bluetooth4.0 USB Adapter"
iSerialNumber:                     0x03
     English (United States)  "00e04c000001"
bNumConfigurations:                0x01
```

And here for the second one:
```
          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0210
bDeviceClass:                      0xEF  -> This is a Multi-interface Function Code Device
bDeviceSubClass:                   0x02  -> This is the Common Class Sub Class
bDeviceProtocol:                   0x01  -> This is the Interface Association Descriptor protocol
bMaxPacketSize0:                   0x40 = (64) Bytes
idVendor:                        0x7392 = Vendor ID not listed with USB.org
idProduct:                       0xA611
bcdDevice:                       0x0200
iManufacturer:                     0x01
     English (United States)  "Realtek"
iProduct:                          0x02
     English (United States)  "Edimax Wi-Fi N150 Bluetooth4.0 USB Adapter"
iSerialNumber:                     0x03
*!*ERROR:  No String Descriptor for index 3!
bNumConfigurations:                0x01
```

Note that Windows simply gives an error for the second serial number. It simply refuses to return the same serial number twice!

dorssel, 2022-12-26:
@vszakd @jamesadevine 
There is a new master build that fixes this problem for @toncho11. Could you give that a try? Artifact is at https://github.com/dorssel/usbipd-win/actions/runs/3783699084. Note that the installer itself is not signed, but the drivers are.

dorssel, 2022-12-26:
@toncho11 
Return address as on envelope?

dorssel, 2022-12-27:
> I tried with `usbipd bind --force --busid 4-2` (4-2 being my second dongle) and then `usbipd wsl attach --busid 4-2` - it did not make a difference. Also each time it fails Windows makes the sound of connecting/disconnecting which is not the case when it works without problem with the first dongle.

I noticed this too. The reason is that for the second device, Windows is re-enumerating the device after the `usbipd`-initiated reset due to the duplicate serial number. That's why it takes another 5 seconds to resolve the situation.

toncho11, 2022-12-27:
Hi,

Many thanks! That seems like a complicated problem!
I am on vacation so it might take some time for me to test. 
Can you please write to me using the email I have provided you? I think our shipping address is different from the one on the envelope. In any case it is better to shop them after New Year. 

Thanks again!!!

dorssel, 2022-12-27:
@toncho11 
Just wrote the second email. Are you receiving those?

toncho11, 2022-12-27:
Now yes and I have responded.
Will test when I have the time with the artifact.
What about 4 Edimax devices in the same time?

dorssel, 2022-12-30:
@toncho11 
4 devices should work fine. One of them will be quick to respond (the first one plugged in). The other 3 will have a 5 second delay every attach/detach. Note the delay is introduced by the Windows OS (hub driver). I don't think it can be avoided unless you update the firmware and either remove the serial number, or make it truly unique.

vszakd, 2022-12-31:
> @vszakd @jamesadevine There is a new master build that fixes this problem for @toncho11. Could you give that a try? Artifact is at https://github.com/dorssel/usbipd-win/actions/runs/3783699084. Note that the installer itself is not signed, but the drivers are.

Great! Thanks @dorssel, I will try as soon as I have the devices handy. 

> @toncho11 4 devices should work fine. One of them will be quick to respond (the first one plugged in). The other 3 will have a 5 second delay every attach/detach. Note the delay is introduced by the [...]

If the devices are UART/USB bridges, one could solve the problem by flashing the serial number manually via the utilities from the vendors of the bridges. For common manufacturers (which cover most of the market):
- FTDI (FT232, etc.), you need to download FT PROG from https://ftdichip.com/wp-content/uploads/2022/05/FT_Prog_v3.12.29.638-Installer.zip
- Silicon Labs (CP21xx), you can download the CP21xxCustomizationUtility from https://www.silabs.com/documents/public/example-code/AN721SW.zip

jamesadevine, 2023-01-03:
@dorssel thanks so much for the effort!

I am afraid after installing the artifact, usbpid reports that both of my devices are in an error state. Before, it reported that only one of the devices was in an error state. A new behaviour is that I can now no longer attach any of my devices to WSL (regardless of how many devices are plugged in).

dorssel, 2023-01-03:
@jamesadevine 
That is very weird. Also when only one device is plugged in? Maybe a reboot is required after update (this non-release version didn't go through all the QA). I am myself running the same build successfully.

jamesadevine, 2023-01-03:
@dorssel yes - this also happens when only one device is plugged in. I restarted my PC as prompted after the update completed successfully. I'll do another restart later just to ensure consistent behaviour.

toncho11, 2023-01-03:
Hi everyone,

I am happy to report that this new version allows me to attach two devices from the same manufacturer (Edimax).  The devices now appear under WSL when using: `bluetoothctl list` which was not the case before.
I still need to test if the BT communication is OK though.

Thank you @dorssel !

jamesadevine, 2023-01-03:
@dorssel a further restart fixed the errors I was seeing thankfully :smile: great stuff

jamesadevine, 2023-01-03:
To clarify: I can attach two of the same device now!

henrygab, 2023-02-08:
Thrilled to see this traiged.  I was having the same problem attaching two Proxmark3 devices (which also use non-unique S/N ... I opened an issue to fix that in the firmware).

THANK YOU for tracking this difficult bug down!

dorssel, 2023-02-20:
@vszakd 
Since you are the OP for this issue... Has it been resolved?

henrygab, 2023-02-20:
Although not the OP, I can confirm this resolved an identical issue for a different USB device that reported identical Serial Number string descriptor (VID, PID, and all String Descriptors were identical).  I used (am using) the build artifacts you [previously pointed to](https://github.com/dorssel/usbipd-win/issues/452#issuecomment-1365484637).

vszakd, 2023-02-20:
@dorssel I could not (and won't be able for a while) to test with the original devices, but I tried with two identical FT232 and I can say that the fix works!

Thank you very much.

closed 2023-02-20 by vszakd

toncho11, 2023-02-20:
Thanks again!!!