dorssel/usbipd-win #458: Odd issue sharing a USB device between Windows and WSL2

thijsputman, 2022-10-03:
Using `usbipd-win` 2.4.0 to connect a Yubikey into WSL2. 

Next to using the Yubikey in WSL2, I'm running a `gpg-agent` on the Windows-side to be able to use the Yubikey for SSH operations from Windows too. I'm attaching and detaching the Yubikey from WSL2 as needed in order to use it in Windows.

This work like a charm, with one very odd caveat:

If upon booting the system I use the Yubikey in Windows _**first**_ (through `gpg-agent` / `scdaemon`), I'm subsequently unable to attach it to WSL2 until I reboot the system (details below). If instead I *first* attach it into WSL2 I can seemingly keep detaching and attaching it at will (and using it in Windows) without any further issue...

In the first (failing) case, the following is shown on the WSL2-side:

```console
$ sudo usbip attach -r 192.168.xxx.xxx -b 3-2
usbip: error: Attach Request for 3-2 failed - Device in error state
```

The log produced by `usbipd` on the Windows-side (captured in 2.3.0; the problem persists in 2.4.0 though) appears to indicate the device is somehow busy?

```log
dbug: Usbipd.Server[1000]
      new connection from 192.168.xxx.xxx
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 292
         at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
         at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 76
         at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 82
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 158
dbug: Usbipd.Server[1000]
      connection closed
```

What I've tried to resolve the issue:

- Physically unplug/plug the Yubikey
- Unbind/bind the Yubikey using `usbipd` on the Windows-side
- Using another USB port (and binding that one)
- Using another Yubikey of the same make and model (in both of the aforementioned ports)
- Killing `gpg-agent` and `scdaemon` on the Windows-side
- Restarting the `usbipd` service

None of these actions appear to resolve the issue. The only solution (thus far) appears to be to reboot Windows and ensure I first attach the device to WSL2 before attempting to use it from Windows....

dorssel, 2022-10-03:
That is weird indeed.

It *looks like* (nut sure, though), that VBoxUsbMon is not able to replace the hardwareId/driver. So, the device is present (with the original driver), then `usbipd` instructs VBoxUsbMon to grab it, VBoxUsbMon reports success, but then the device (with the replaced driver) cannot be found. Maybe that is because the driver was not changed by VBoxUsbMon after all, or maybe the device itself is indeed in some kind of error state due to the driver swap.

Can you try the following:
- open Windows device manager
- set it to "View devices by connection"
- find the device, and make sure it remains in view for the remainder of this test
- `wsl attach` such that it fails (i.e. reach the error state)
- while doing that, check the state of the device in device manager
- does it show any changes? in device name? in error state ("yellow bang")? is this very brief and then recovers, or does the change persist?
- afterwards, check the Windows EventLog to see if VBoxUsbMon reported any errors

thijsputman, 2022-10-03:
Reproduction of the issue appears less reliable than I initially assumed — just restarted my machine and can't seem to reproduce it for now... 🫤

Anyway, I'm using this setup on a daily basis so it'll probably pop up again in the near future. If it does, I'll go through the above instructions. Thanks for the prompt response!

For now, the only (potentially) relevant observation is that the Yubikey initially gets identified as a `NIST SP 800-73 [PIV]` smart card in Device Manager; after an initial attach/detach to WSL2 it comes back as an `Unknown Smart Card` (which apparently is a valid driver — there's no yellow exclamation mark or anything like that).

1-bit-wonder, 2022-10-05:
@thijsputman 

Would you mind sharing how exactly you get that set-up working? I have been trying to do that exact thing, have been able to attach the Yubikey just fine to WSL, however when I try to do something like generate or import a key it can't seem to find the device. 

ykman finds the card after attaching: 
```
$ ykman list
YubiKey 5 NFC [OTP+FIDO+CCID] Serial: XXXXXXXX
```

and can also get info about the device:
```
$ ykman info
Device type: YubiKey 5 NFC
Serial number: XXXXXXXX
Firmware version: 5.4.3
Form factor: Keychain (USB-A)
Enabled USB interfaces: OTP+FIDO+CCID
NFC interface is enabled.

Applications    USB     NFC
OTP             Enabled Enabled
FIDO U2F        Enabled Enabled
OpenPGP         Enabled Enabled
PIV             Enabled Enabled
OATH            Enabled Enabled
FIDO2           Enabled Enabled
```

but when I try to get or generate a key:
```
$ ssh-add -vvvv -K
Enter PIN for authenticator:
debug3: start_helper: started pid=1826
debug3: ssh_msg_send: type 5
debug3: ssh_msg_recv entering
debug1: start_helper: starting /usr/lib/openssh/ssh-sk-helper
debug1: sshsk_load_resident: provider "internal", have-pin
debug1: ssh_sk_load_resident_keys: fido_dev_info_manifest failed: FIDO_ERR_INTERNAL
Provider "internal" returned failure -1
debug1: ssh-sk-helper:  sshsk_load_resident failed: invalid format
debug1: ssh-sk-helper: reply len 8
debug3: ssh_msg_send: type 5
debug1: client_converse: helper returned error -4
debug3: reap_helper: pid=1826
Unable to load resident keys: invalid format
```

```
$ ssh-keygen -vvvv -K
Enter PIN for authenticator:
debug3: start_helper: started pid=1795
debug3: ssh_msg_send: type 5
debug3: ssh_msg_recv entering
debug1: start_helper: starting /usr/lib/openssh/ssh-sk-helper
debug1: sshsk_load_resident: provider "internal", have-pin
debug1: ssh_sk_load_resident_keys: fido_dev_info_manifest failed: FIDO_ERR_INTERNAL
Provider "internal" returned failure -1
debug1: ssh-sk-helper:  sshsk_load_resident failed: invalid format
debug1: ssh-sk-helper: reply len 8
debug3: ssh_msg_send: type 5
debug1: client_converse: helper returned error -4
debug3: reap_helper: pid=1795
Unable to load resident keys: invalid format
```

```
$ ssh-keygen -vvvv -t ecdsa-sk -O resident
Generating public/private ecdsa-sk key pair.
You may need to touch your authenticator to authorize key generation.
debug3: start_helper: started pid=1756
debug3: ssh_msg_send: type 5
debug3: ssh_msg_recv entering
debug1: start_helper: starting /usr/lib/openssh/ssh-sk-helper
debug1: sshsk_enroll: provider "internal", device "(null)", application "ssh:", userid "(null)", flags 0x21, challenge len 0
debug1: sshsk_enroll: using random challenge
debug1: pick_first_device: fido_dev_info_manifest failed: FIDO_ERR_INTERNAL
debug1: ssh_sk_enroll: pick_first_device failed
debug1: sshsk_enroll: provider "internal" returned failure -4
debug1: ssh-sk-helper: Enrollment failed: device not found
debug1: ssh-sk-helper: reply len 8
debug3: ssh_msg_send: type 5
debug1: client_converse: helper returned error -60
debug3: reap_helper: pid=1756
Key enrollment failed: device not found
```

Another strange thing happens when I try to get any info about fido, and I believe this may be the root of the problem:
```
$ ykman fido list
Traceback (most recent call last):
  File "/usr/bin/ykman", line 11, in <module>
    load_entry_point('yubikey-manager==3.1.1', 'console_scripts', 'ykman')()
  File "/usr/lib/python3/dist-packages/ykman/cli/__main__.py", line 273, in main
    cli(obj={})
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/lib/python3/dist-packages/ykman/cli/fido.py", line 112, in list_creds
    controller = ctx.obj['controller']
  File "/usr/lib/python3/dist-packages/ykman/cli/util.py", line 127, in __getitem__
    self.resolve()
  File "/usr/lib/python3/dist-packages/ykman/cli/util.py", line 124, in resolve
    self._objects[k] = f()
  File "/usr/lib/python3/dist-packages/ykman/cli/__main__.py", line 194, in resolve_device
    dev = _run_cmd_for_single(ctx, subcmd.name, transports, reader)
  File "/usr/lib/python3/dist-packages/ykman/cli/__main__.py", line 132, in _run_cmd_for_single
    return descriptor.open_device(transports)
  File "/usr/lib/python3/dist-packages/ykman/descriptor.py", line 96, in open_device
    for drv in _list_drivers(transports):
  File "/usr/lib/python3/dist-packages/ykman/descriptor.py", line 164, in _list_drivers
    for dev in open_fido():
  File "/usr/lib/python3/dist-packages/ykman/driver_fido.py", line 97, in open_devices
    for dev in CtapHidDevice.list_devices(descriptor_filter):
  File "/usr/lib/python3/dist-packages/fido2/hid.py", line 135, in list_devices
    for d in hidtransport.hid.Enumerate():
  File "/usr/lib/python3/dist-packages/fido2/_pyu2f/linux.py", line 183, in Enumerate
    for hidraw in os.listdir('/sys/class/hidraw'):
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/hidraw'
```

Curious if you had any issues like this?

dorssel, 2022-10-05:
What is the output from `usbipd list` for these different identifiers? (NIST ... vs. Unknown). Is the actual hardwareID (VID:PID) different?

thijsputman, 2022-10-06:
They're identical (`1050:0407`).

Good to note that the actual device listed is a `Microsoft Usbccid Smartcard Reader (WUDF)` (by connection the parent device of the smart card device). It has the VID and PID of the Yubikey so I'm guessing it's the actual Yubikey. The smart card device appears to be some kind of software component (provided by the `ScFilter` driver) — I'm guessing it's unrelated to my problem...

thijsputman, 2022-10-11:
Well... The issue does not appear to be related to `usbipd-win`.

Instead, it seems that Windows occasionally doesn't load the Yubikey smart card driver properly. When encountering the problem described above, the smart card functionalities don't work on the Windows-side either (hadn't actually tested that). The entire `NIST SP 800-73 [PIV]` / `Unknown Smart Card` device never appears in device manager when inserting the Yubikey in that situation...

I'm guessing VBoxUsbMon simply trips over the underlying issue on the Windows-side (whatever that may be; still trying to work that one out).

thijsputman, 2022-10-12:
Okay — spoke too soon, the saga might continue...

I've installed the [Yubikey Smart card Minidriver](https://www.yubico.com/support/download/smart-card-drivers-tools/) in Windows, replacing the generic Microsoft driver for the smart card with a driver provided by Yubico.

Now the problem only occurs when using `usbipd`. On the Windows-side, the Yubikey recovers from the error state after unplugging and plugging it back in (it's fully usable again on the Windows-side after that). Repeated attempts at attaching the device to WSL all result in the same error: `Attach Request for 3-2 failed - Device in error state`.

This device shows up in device manager after `usbip` on WSL generates the error:

[image]

These are the only related errors and warnings that show up in Event Viewer (repeated for every attempt at attaching the device):

[image]

With these messages:

```
The driver \Driver\VBoxUSB failed to load for the device USB\Vid_80EE&Pid_CAFE\6&1a565baf&0&2.
Metadata staging failed, result=0x80072EFD for container '{00000000-0000-0000-FFFF-FFFFFFFFFFFF}'
Metadata staging failed, result=0x80072EFD for container '{F824E50E-D258-5D12-AE08-73B9A1D7F224}'
Metadata staging failed, result=0x80004005 for container '{F824E50E-D258-5D12-AE08-73B9A1D7F224}'
```

Furthermore, one sort of related "information" event shows up:

```
The description for Event ID 26 from source Application Popup cannot be found. Either the component that raises this event is not installed on your local computer or the installation is corrupted. You can install or repair the component on the local computer.

If the event originated on another computer, the display information had to be saved with the event.

The following information was included with the event: 

\SystemRoot\System32\Drivers\VBoxUSB.sys failed to load

The message resource is present but the message was not found in the message table
```

[image]

dorssel, 2022-10-25:
Can you try version 2.4.1?
It fixes some Authenticode issues with the installer signatures. It looks like in your case Windows does not trust the VBoxUSB driver, which seems to be Authenticode related. I thought the differences between 2.4.0 and 2.4.1 would not have any effect once you make it passed the installer, but maybe I was wrong...

thijsputman, 2022-10-30:
Just installed it — will keep you posted should anything change (or not change 😉)...

dorssel, 2022-12-17:
@thijsputman
Has this issue been resolved?

dorssel, 2022-12-22:
Still interesting... Are you perhaps running a third party virus scanner?

thijsputman, 2022-12-23:
Nope.

Only thing I can think of is that I'm running a relatively "old" Windows installation; about 2,5-years since it was freshly installed.

supleed2, 2023-03-11:
I've been having a similar issue for a while now but never discovered this particular issue before. It seems to have the same symptoms. Powershell looks like this:

```powershell
> usbipd wsl attach -b 1-1
usbipd: info: Using default WSL distribution 'Ubuntu'; specify the '--distribution' option to select a different one.
usbip: error: Attach Request for 1-1 failed - Device in error state

usbipd: error: Failed to attach device with busid '1-1'.
```

Looking at Device Manager, the device now shows up as something called `VirtualBox USB` with VID:PID `80EE:CAFE`. This seems to happen regardless of which device I use (USB Drive, An emulated USB Device on a development board) but only on one windows machine. The USB devices work when I try to connect to them from the same wsl distro, but running usbipd on another windows machine on the same LAN network.

<details>
<summary>Device Manager Screenshots</summary>

[image]
[image]
</details>

<details>
<summary>Event viewer for `usbipd-win` shows a similar error as above:</summary>

```log
Category: Usbipd.ConnectedClient
EventId: 3

An exception occurred while communicating with the client:

Exception: 
System.IO.FileNotFoundException: Unable to find the specified file.
   at Usbipd.ConfigurationManager.GetVBoxDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 289
   at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
   at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 76
   at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 82
   at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 272
   at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 272
   at Usbipd.ConnectedClient.RunAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 74
```
</details>

<details>
<summary>I have the same Kernel-PnP warning:</summary>

```log
The driver \Driver\VBoxUSB failed to load for the device USB\Vid_80EE&Pid_CAFE\D9116FBF.
```
</details>

But I don't see the DeviceSetupManager Errors.

Is there anything I can try / get that might help figure this out? I'm able to work by copying files over to Windows first but it's still clunky. When `usbipd-win` worked, it worked great :)

dorssel, 2023-03-12:
@supleed2 
Obviously, this has to do with driver loading. `usbipd-win` installs the same drivers (binary identical) as VirtualBox. There is a "generic driver" install for normal attach (PnP) during installation, and the driver is installed explicitly when using `--force`. If you are not having a full installation of VirtualBox (which is known to clash with `usbipd-win` if the driver versions don't match), first try to repair or reinstall `usbipd-win`. For good measure, reboot after the (re-)install, even if the installer does not think it is necessary.

If that does not work ... maybe your issue is related to https://borncity.com/win/2018/07/06/windows-10-v1803-hcvi-causes-driver-error-code-39/. Try with core isolation off.

supleed2, 2023-03-12:
Thank you for the help! I managed to get it working, now I can get back to being productive ❤️

I had installed VirtualBox to test yesterday and it wasn't able to pass through the usb device either, so I had uninstalled it immediately.

Noting the steps I took here incase it helps in the future:

- Uninstall the device in Device Manager (including the option to remove the driver if available)
- Remove `usbipd-win` via `winget uninstall --purge dorssel.usbipd-win` and then reboot (Previously I had just uninstalled via control panel)
- Install `usbipd-win` via `winget install dorssel.usbipd-win` and then reboot again
- Connect device and use the force option to install the right driver, then bind normally to allow the device to be used in windows until attached (In admin powershell): `usbipd bind -f -b <busid>`, then `usbipd bind -b <busid>`

dorssel, 2023-06-02:
@thijsputman 
Does this issue still happen with 3.0.0?

thijsputman, 2023-06-02:
Haven't seen the issue occur since I installed 3.0.0 — prior to that it'd only been happening occasionally (once or twice per month), so maybe it still will...

Anyway, we can probably conclude it's an esoteric problem specific to my hardware/drivers; feel free to close the issue.

closed 2023-06-02 by dorssel

RT-Tap, 2023-08-14:
Hey I'm having this exact issue.  I tried and successfully passed the yubikey through for the firs time a couple days ago , everyhting worked fine including attaching then detaching multiple times, then a windows update forced a reboot (I cant describe in words how much I hate it and how much time ive spent trying to prevent the reboot but I digress) and ever since the reboot I can't connect again, I just get the same VirtualBox USB error.