dorssel/usbipd-win #464: Device not showing up in Ubuntu

MrBillium, 2022-10-17:
I have a pretty clean fresh install of Ubuntu using  wsl --install on my Win 11 machine and want to see a USB device from Linux so I followed the MS tutorial at  https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/#how-it-works

I can see all my USB device in usbipd wsl list command prompt and have attached one of them which now lists as 'shared'

C:\Program Files\usbipd-win>usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-11   048d:5702  USB Input Device                                              Not shared
2-14   8087:0026  Intel(R) Wireless Bluetooth(R)                                Not shared
3-1    04e8:6860  Bill's Tab A7 Lite, SAMSUNG Mobile USB Modem, SAMSUNG And...  Shared
3-2    046d:c52b  Logitech USB Input Device, USB Input Device, Logitech Uni...  Not shared
4-1    046d:0990  Logitech QuickCam Pro 9000, Pro 9000                          Not shared
4-2    046d:c408  USB Input Device                                              Not shared
4-4    051d:0002  American Power Conversion USB UPS                             Not shared

Persisted:
GUID                                  DEVICE

so fa so good - BUT lsusb in Ubuntu just shows two root hubs - not my Android device

bill@Alder-Fractal:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
bill@Alder-Fractal:~$

I have tried rebooting with no change - any idea what is going on or how to troubleshoot?

dorssel, 2022-10-17:
Seems like you only bound ("shared") the device. That is different from the next step: attaching. What is the output of `usbipd wsl attach`?

MrBillium, 2022-10-19:
I assumed that shared meant it was working and did not realize that they were different states. 
Playing around gave inconsistent results including a firewall error which others had fixed with a reinstall.
So that's what i did and now it seems to be working properly.

closed 2022-10-19 by MrBillium