dorssel/usbipd-win #465: usbipd-win on windows 10 needs sudo prefix for openocd commands

noeldiviney, 2022-10-19:
[image]
I have successfully installed usbipd-win version 2.4.1 in windows 10.
When I use openocd to connect to a stm32f4 target I have to prefix the command with sudo as follows
sudo J-Tagit/openocd/bin/openocd -f interface/J-Tagit/j-tagit/swd.cfg -f target/J-Tagit/j-tagit/stm32f4x.cfg
See the attached screenshot for more details.
Is it possible to remove this requirement

Many thanks for the marvelous work in creating usbipd
Noel Diviney

dorssel, 2022-10-19:
This is entirely a Linux issue, not `usbipd-win`. Are you using WSL? In that case you may have to setup/start the udev service. See also https://github.com/dorssel/usbipd-win/wiki/WSL-support#udev.

noeldiviney, 2022-10-19:
Hi 
Many tks for the link. I will experiment some more tomorrow as it is now late here, and report back.
Cheers Noel.

noeldiviney, 2022-10-20:
Hi Frans,
Ok success finally.
The Udev rules I was using were an earlier version which work ok on Linux.
I updated the openocd rules to the latest version but still not working.
I finally decided to look at the FTDI Rules and also changed the Mode to "06666" from "0660".
That did the trick.
Attachment 1 is a screenshot of Arduino Ide and Putty terminal running in WSL.
Attachment 2 is the the FTDI Rules required
So finally thanks to you good self we cal fullfill the dream of Linux embedded Development on Windows

Cheers Noel.

[image]
[image]

closed 2022-10-20 by noeldiviney