dorssel/usbipd-win #466: Need build instructions to

paulpv, 2022-10-19:
I looked in this repo for build instructions and/or minimum requirements and did not see any.
Maybe they are here and I just missed them.

I cloned the repo and loaded it in Visual Studio Community 2022 and ran in to the following issues:
1. I needed to install the WIX Visual Studio Extension 2022; no biggie, but it would be nice to know ahead of time it is a requirement.
2. I am currently blocked trying to build `UsbIpd.Automation` with the following Dependencies\Analyzers:
[image]

In addition to build steps, manual installation steps such as installing the service [if it is not auto-installed] maybe should be documented.

dorssel, 2022-10-19:
Build instructions are just following https://github.com/dorssel/usbipd-win/blob/master/.github/workflows/build-installer.yml.

1) install dotnet 6 SDK
2) `dotnet build`

That's all. I don't know about Visual Studio Community 2022,

paulpv, 2022-10-19:
Interesting: It builds from the command-line on a fresh checkout, but not inside of Visual Studio on a fresh checkout.

Still investigating...

paulpv, 2022-10-20:
Scratch that: It actually does build inside of VS2022 on a fresh checkout.

It suspect it might have had something to do with not having all of the pre-reqs installed prior to loading the solution the first time and trying to build.

I installed the pre-reqs and it still errored with the same problem even after a `Restore NuGet Packages`:
[image]

I fresh cloned this repo in to a separate folder and built it from the command-line and it built successfully.

I cd'd to my original folder and deleted all files (except for `.git`) and then `git reset --hard`.
I then build it from the command-line and it built successfully.

I don't care to bother to uninstall the recently installed dependencies to try to repo on this machine.
I am attempting a repro on an identical machine...

paulpv, 2022-10-20:
No repro on my 2nd identical [hardware] machine that appears to have already had .NET6 installed on it. :/

So, just chalk it up to weirdness.

I'll close this bug.

closed 2022-10-20 by paulpv