dorssel/usbipd-win #467: Having troubles using a USB TV tuner

flextcg1, 2022-10-20:
I've set up USBIP which seems to work fine in itself, but I'm having issues using my TV tuner card with WSL2. I have to admit, I only started using WSL2 and linux like a week ago, so my knowledge about this operating system is next to nothing, please keep that in mind.

After I attach the device, it shows up in lsusb and also in dmesg, however, it seems like it wouldn't even try to load its firmware or anything. Therefore, doing a scan in w_scan results in an error message saying there are no DVB devices connected. I also tried rebuilding the kernel and turning on most multimedia setting and device drivers, even manually installing V4L using [this](https://www.linuxtv.org/wiki/index.php/How_to_Obtain,_Build_and_Install_V4L-DVB_Device_Drivers) tutorial, but it made no difference.

I assumed that the tuner card needed direct USB connection, and wouldn't work over IP, but I tested that on Hyper-V with a full fledged Ubuntu, and it worked just fine, so that shouldn't be the limitation here. I did a dmesg on that VM as well to see what the differences are compared to the one I did on WSL, and it looks like the module (or driver/service?) "dvb_usb_v2" is not showing up after listing the device. There is also no /dev/dvb folder that exists. 

I attached the two dmesg logs below, hopefully that helps in some way. 

Thanks for your help in advance!

[dmesg_WSL.txt](https://github.com/dorssel/usbipd-win/files/9827414/dmesg_WSL.txt)
[dmesg_hyperV_ubuntu.txt](https://github.com/dorssel/usbipd-win/files/9827418/dmesg_hyperV_ubuntu.txt)

flextcg1, 2022-10-21:
I've gotten a little further with this. With a fresh install, I rebuilt the kernel multiple times, trying to see if messing with the settings a little more would make any difference and it actually did. It was a trial and error thing, but instead of modules, I set everything to "bulit-in" under Multimedia, and now the capture card appears in TVH and it looks like, it's searching for channels, however, it couldn't find any. It could be a firmware issue, but I have no idea how to fix that, since putting the working firmware into the lib/firmware directory makes no difference. I couldn't get it to work properly yet.

dorssel, 2022-10-21:
Other people have had problems getting the firmware to load. See here for the reason why it is so difficult, and for user stories on how they succeeded: https://github.com/dorssel/usbipd-win/issues/390. Unfortunately, I do not know anything more about it. You may have to resort to asking on https://github.com/microsoft/WSL/issues.

flextcg1, 2022-10-22:
Thanks, I've tried their solution too, but it's weird, the adapter and frontends show up at first, dmesg shows no errors either, then I do a test scan with the app "scan", and all the sudden, the next dmesg says "firmware download failed -11". Might be a limitation or bug of WSL, which I can't get around. Will ask in the WSL github as well.