dorssel/usbipd-win #469: USB Serial delays issues

ExtremeGTX, 2022-10-24:
Hello usbipd-win team,
Thanks for providing such tool to unlock the possibility to work with USB from WSL2!

I have a corner case here that I faced before also with VMWare:
So I have a DevKit for ESP32 ([Example kit](https://shop.m5stack.com/products/basic-core-iot-development-kit?variant=16804801937498)) and to force the board in Firmware flashing mode the flashing tool use RTS/DTR lines to reset and enter flashing mode.

The Boot mode switching process is explained [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html?highlight=boot#automatic-bootloader)  which make use of the circuit shown below.

[image]

The problem is this is not working when the usb device is shared over usbip.
When i had this issue with VMWare before i had to use VMWare Serial port instead of Detach the USB Device from Host and attach it to the guest.
[image]

Thanks in Advance 🙂

dorssel, 2022-10-25:
> The problem is this is not working when the usb device is shared over usbip.

What is the error? Any syslog messages? Do you have more logging, as explained here: https://github.com/dorssel/usbipd-win/wiki/Troubleshooting.

ExtremeGTX, 2022-10-30:
Here are the logs and also a screencast.
```
❯ usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = ''
dbug: Usbipd.Server[1000]
      2.4.1+1.Branch.master.Sha.90acf9456020ca8c6310ca62a71ee23cb6ca34ad
dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveInterval = 500 ms
dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveTime = 10000 ms
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\ExtremeGTX
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: Usbipd.Server[1000]
      new connection from 172.21.6.53
dbug: Usbipd.Server[1000]
      connection close: Attempted to read past the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.21.6.53
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
info: Usbipd.ConnectedClient[1]
      Client 172.21.6.53 claimed device at 9-2 (USB\VID_10C4&PID_EA60\014487EC).
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      Trapped SET_CONFIGURATION: 1
trce: Usbipd.AttachedClient[1001]
      Unlinking 78600, pending = True, pending count = 2
trce: Usbipd.AttachedClient[1001]
      Aborting endpoint 130
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
```

https://user-images.githubusercontent.com/6371120/198883139-8c2b8ff2-477f-46f9-95b4-23ddb32cb058.mp4

Please notice that while the flashing tool is in `connecting ..........` phase there are no logs on USBIP side, dunno if the feature of controlling DTR/RTS is supported.

dorssel, 2022-10-30:
Logging of actual USB communication is done using pcap, not to console. See https://github.com/dorssel/usbipd-win/wiki/Troubleshooting#usb-capture. Can you turn that on too? That will show much more useful data. The EPIPEs that you are seeing now are quite normal, at the end of each communication session. The first ones are after device discovery (normal), and the last one at cleanup *after* the connection failure (also normal). Hopefully the pcap trace will show what is the actual error that is causing your software to fail.

ExtremeGTX, 2022-11-03:
Here is the pcap trace for the scenario shown in the screencast:
with a minor change which is `bus-id` on Windows is `1-2` instead of `9-2`
[usbip_dtr_rts_issue.zip](https://github.com/dorssel/usbipd-win/files/9932661/usbip_dtr_rts_issue.zip)

dorssel, 2022-11-03:
The message received from the device is:

```
Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6596
load:0x40078000,len:15216
ho 0 tail 12 room 4
load:0x40080400,len:3992
entry 0x40080634
(28) boot: ESP-IDF 8c7ee781e 2nd stage bootloader
(28) boot: compile time 00:41:17
(28) boot: chip revision: 1
(31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
(38) boot.esp232: SPI Speed      : 40MHz
(43) boot.esp32: SPI Mode       : DIO
(48) boot.esp32: SPI Flash Size : 4MB
(52) boot: Enabling RNG early entropy source...
(58) boot: Partition Table:
(61) boot: ## Label            Usage          Type ST Offset   Length
(68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
(76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
(83) boot:  2 factory          factory app      00 00 00010000 00100000
(91) boot: End of partition table
(95) boot_comm: chip revision: 1, min. application chip revision: 0
(102) esp_image: segment 0: paddr=00010020 vaddr=00000020 size=0001ch (    28)
(110) esp_image: segment 1: paddr=00010044 vaddr=3ffb0000 size=00af0h (  2800) load
(120) esp_image: segment 2: paddr=00010b3c vaddr=3ffb0af0 size=001e8h (   488) load
(128) esp_image: segment 3: paddr=00010d2c vaddr=40080000 size=03f3ch ( 16188) load
(143) esp_image: segment 4: paddr=00014c70 vaddr=00000000 size=0b3c8h ( 46024)
(161) esp_image: segment 5: paddr=00020040 vaddr=3f400040 size=07b04h ( 31492) map
(172) esp_image: segment 6: paddr=00027b4c vaddr=00000000 size=084cch ( 33996)
(185) esp_image: segment 7: paddr=00030020 vaddr=400d0020 size=282f0h (164592) map
(247) boot: Loaded app from partition at offset 0x10000
(247) boot: Disabling RNG early entropy source...
*** Booting Zephyr OS build zephyr-v3.2.0-843-gd832b04e9627  ***
```
 
This happens over and over, with only a few tenths of a second in between. So, the sequence is:

- control message (presumably to trigger a reset)
- message is received as above (in chunks of approx 1-3 lines per request)
- approx. 0.2 to 0.9 seconds of nothing (no errors either! looks like waiting for boot to complete)
- repeat

After a number of such cycles the connection is aborted. To me it looks like the timeout for boot to complete is too short. Remember that USBIP does introduce some latency;  not much, 0.1 second or so, but same order of magnitude.

ExtremeGTX, 2022-11-04:
> Remember that USBIP does introduce some latency; not much, 0.1 second or so, but same order of magnitude.

yes, I understand, and unfortunately that breaks any use case which involves some kind of timing constraints, i had another tool also failed because of the delays introduced over USBIP.
Same problem is in VMWare and VirtualBox when USB Device is connected to the guest machine, but not when a "Serial Port" is configured and linked to the device serial port on the Host OS as explained [Here](https://techtooltip.wordpress.com/2008/09/12/using-host-serial-port-from-guest-in-virtual-box/)

[image]

dorssel, 2022-11-04:
You could try:
- open Windows task manager
- go to details
- find `usbipd.exe`
- right click: select "Process Priority -> Realtime"

That will reduce latency (a bit).

ExtremeGTX, 2022-11-05:
Unfortunately, that didn't help :( .
if had time and effort, i would try to investigate/compare the way "Serial port" is implemented vs sharing USB device. and i think part of improving the delays, is skipping USB stack on the guest side. so `ttyUSB0` will end up be `ttyS0`.

Jayspek, 2024-08-19:
@ExtremeGTX I know this is an old issue, but did you make any progress with this? Would love to know if you came up with a solution.