dorssel/usbipd-win #488: Unable to attach 同

Incipe-win, 2022-11-28:


Incipe-win, 2022-11-28:
Hello usbipd-win team,
I have installed usbip according to the [Microsoft documentation](https://learn.microsoft.com/en-us/windows/wsl/connect-usb), and can view the device information through usbipd. However, when I execute `usbipd wsl attach`, it does not take effect. Here is my screenshot.
[image]
My device is connected to the windows usb through the usb adapter, is this the cause?
Thanks in Advance 🙂

dorssel, 2022-11-28:
Like the warning says, you will need to use `usbipd bind --force` first...

Incipe-win, 2022-11-29:
Thank you very much for your reply. 
This warning is because I have installed USBPcap. If I uninstall USBpcap, there will be no Warning. 
But I still can't attach to usb device.
[image]

dorssel, 2022-11-29:
1) Can you post the results of `usbipd list` (instead of `usbipd wsl list`)?
2) Can you run `usbipd` with full logging? See https://github.com/dorssel/usbipd-win/wiki/Troubleshooting

dorssel, 2023-01-05:
Stale

closed 2023-01-05 by dorssel