dorssel/usbipd-win #489: Can't Attach NVIDIA Jetson To WSL2 Instance

thegreatco, 2022-11-28:
I am trying to attach an NVIDIA Jetson to my WSL2 instance to update it using the NVIDIA `sdkmanager`. I am able to do this when the jetson is in "recovery" or "bootloader" mode, but when I attempt to connect it while the Jetson is booted normally, I am unable to do so.

Version: `2.4.1+1.Branch.master.Sha.90acf9456020ca8c6310ca62a71ee23cb6ca34ad`
Error: 
```
usbipd wsl attach -b 2-4 -d Ubuntu-18.04
usbip: error: recv op_common
usbip: error: query
usbipd: error: Failed to attach device with busid '2-4'.
```

here is what the device listing looks like:
```
BUSID  VID:PID    DEVICE                                                        STATE
2-4    0955:7020  Remote NDIS Compatible Device #2, USB Serial Device (COM5...  Not attached
2-9    27c6:609c  Framework Fingerprint Reader                                  Not attached
2-10   8087:0032  Intel(R) Wireless Bluetooth(R)                                Not attached
3-1    05e3:0626  Generic SuperSpeed USB Hub                                    Not attached
6-5    0bda:5450  BillBoard Device                                              Not attached
```

Note how the Jetson shows up as several devices.

thegreatco, 2022-11-28:
I installed `linux-tools-virtual-hwe-18.04`:
```
$ sudo apt install linux-tools-virtual-hwe-18.04
Reading package lists... Done
Building dependency tree
Reading state information... Done
linux-tools-virtual-hwe-18.04 is already the newest version (5.4.0.132.148~18.04.109).
0 upgraded, 0 newly installed, 0 to remove and 13 not upgraded.
```

Next I ran:
```
$ sudo update-alternatives --install /usr/local/bin/usbip usbip /usr/lib/linux-tools/5*/usbip 30
update-alternatives: using /usr/lib/linux-tools/5.4.0-132-generic/usbip to provide /usr/local/bin/usbip (usbip) in auto mode
$ update-alternatives --query usbip
Name: usbip
Link: /usr/local/bin/usbip
Status: auto
Best: /usr/lib/linux-tools/5.4.0-132-generic/usbip
Value: /usr/lib/linux-tools/5.4.0-132-generic/usbip

Alternative: /usr/lib/linux-tools/4.15.0-197-generic/usbip
Priority: 20

Alternative: /usr/lib/linux-tools/5.4.0-132-generic/usbip
Priority: 30
```

Finally I ran this on windows:
```
> usbipd wsl attach -b 2-4 -d Ubuntu-18.04
usbip: error: Attach Request for 2-4 failed - Device busy (exported)

usbipd: error: Failed to attach device with busid '2-4'.
```

Thinking this may fall under "needs to be run as admin the first time", I got the same error there:
```
C:\Windows\System32>usbipd wsl attach -b 2-4 -d Ubuntu-18.04                                                            usbip: error: Attach Request for 2-4 failed - Device busy (exported)                                                                                                                                                                            usbipd: error: Failed to attach device with busid '2-4'. 
```

It still shows as not attached:
```
> usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
2-4    0955:7020  Remote NDIS Compatible Device #2, USB Serial Device (COM5...  Not attached
2-9    27c6:609c  Framework Fingerprint Reader                                  Not attached
2-10   8087:0032  Intel(R) Wireless Bluetooth(R)                                Not attached
3-1    05e3:0626  Generic SuperSpeed USB Hub                                    Not attached
6-5    0bda:5450  BillBoard Device                                              Not attached
```

dorssel, 2022-11-29:
Now you're running into https://github.com/dorssel/usbipd-win/issues/304. See solution there.

thegreatco, 2022-11-29:
It just so happens that when I tried again this morning, I didn't need to force attach. Thanks for the super quick help!

Quick summary for anybody finding this later:
1. Ensure latest hardware tools are installed, in my case:
   ```
   sudo apt install linux-tools-virtual-hwe-18.04
   ```
   Replace `18.04` with your OS version
2. Make sure the right tools are selected
   ```
   $ update-alternatives --query usbip
   Name: usbip
   Link: /usr/local/bin/usbip
   Status: auto
   Best: /usr/lib/linux-tools/5.4.0-132-generic/usbip
   Value: /usr/lib/linux-tools/5.4.0-132-generic/usbip

   Alternative: /usr/lib/linux-tools/4.15.0-197-generic/usbip
   Priority: 20

   Alternative: /usr/lib/linux-tools/5.4.0-132-generic/usbip
   Priority: 30
   ```
   If the higher version is not selected, you will need to update it with:
   ```
   sudo update-alternatives --install /usr/local/bin/usbip usbip /usr/lib/linux-tools/5*/usbip 30
   ```
3. Attempt to attach again, if that fails, you may need to force bind to prevent something in the host from locking the device:
   ```
   usbipd bind --force --busid X-X
   ```
   Replace `X-X` with the proper BusID. Note that this may require a reboot of the Host OS and the device will be permanently claimed by `usbipd`. You will need to use `usbipd unbind` to free it for use by the Host OS.

closed 2022-11-29 by thegreatco