dorssel/usbipd-win #492: Attach request failed - device in error state

jpicklyk, 2022-12-02:
Software was working perfectly fine 2 days ago.  I'm using usbipd-win on Windows 11 21H2 with WSL 2 and an Ubuntu 22.04 image that I built a custom kernel for.  Today I went to do some testing for a project and found that I can no longer attach any usb device to the Ubuntu WSL 2 instance.

Win 11 21H2 + all updates
WSL2 (Ubuntu 22.04 + custom kernel)
usbipd-win v 2.4.1

PS C:\Users\Jeff> usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
4-9    067b:2303  Prolific USB-to-Serial Comm Port (COM5)                       Not attached
4-11   0b05:1a52  Realtek USB Audio, USB Input Device                           Not attached
4-12   0b05:19af  AURA LED Controller, USB Input Device                         Not attached
4-13   1b1c:1c1e  USB Input Device                                              Not attached
5-7    046d:c539  USB Input Device                                              Not attached
5-11   046d:c52b  Logitech USB Input Device, USB Input Device                   Not attached
5-14   0489:e0e2  MediaTek Bluetooth Adapter                                    Not attached
6-4    0cf2:a102  LianLi-SL-infinity-v0.8, USB Input Device                     Not attached
7-4    0cf2:a100  LianLi-UNI FAN-SL-v1.8, USB Input Device                      Not attached

PS C:\Users\Jeff> usbipd wsl attach -b 4-9
usbip: error: Attach Request for 4-9 failed - Device in error state

Windows event viewer shows the following error for usbipd-win when the failure happens:

Category: Usbipd.ConnectedClient
EventId: 3

An exception occurred while communicating with the client:

Exception: 
System.IO.FileNotFoundException: Unable to find the specified file.
   at Usbipd.ConfigurationManager.GetVBoxDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 289
   at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
   at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 76
   at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 82
   at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 165
   at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 266
   at Usbipd.ConnectedClient.RunAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 58

Windows device manager shows that the device fails with the following:
[image]

I've tried rebooting, updating all drivers, windows updates, repairing the windows install of usbipd-win to no avail.  Any help would be appreciated.

Thank you,
Jeff

jpicklyk, 2022-12-02:
Here are trace logs, looks like the server is reading past the end of the stream causing IOException if I am reading this correctly.

PS C:\Users\Jeff> usbipd server
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\Jeff
info: Microsoft.Hosting.Lifetime[0]
      Application is shutting down...
PS C:\Users\Jeff> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = ''
dbug: Usbipd.Server[1000]
      2.4.1+1.Branch.master.Sha.90acf9456020ca8c6310ca62a71ee23cb6ca34ad
dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveInterval = 500 ms
dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveTime = 10000 ms
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\Jeff
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: Usbipd.Server[1000]
      new connection from 172.20.196.209
dbug: Usbipd.Server[1000]
      connection close: Attempted to read past the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.20.196.209
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 289
         at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
         at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 76
         at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 82
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 165
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 266
         at Usbipd.ConnectedClient.RunAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 58
dbug: Usbipd.Server[1000]
      connection close: Unable to find the specified file.
dbug: Usbipd.Server[1000]
      connection closed

lanru, 2022-12-03:
I'm in the same situation

dorssel, 2022-12-03:
Do you (also) have VirtualBox installed? Can you try `usbipd bind --force`?

The problem is really with the driver. The errors/trace from the service are consistent with the driver failing to load.

jpicklyk, 2022-12-03:
The drivers initially show up fine in Device Manager, there are no warnings or errors on the devices being used.  usbipd bind --force didn't initially help.  In fact I cannot seem to share any USB device, not just the one I actually need.

I do not have VirtualBox (Oracle's separate software) installed.

Weirdly enough, I just happened to turn my computer on today and test again to provide further data and it magically seems to be working.  All I did was turn the computer off last night and turn it on today to investigate this issue further.  I did originally try rebooting the machine several time, detaching the WSL linux machine and reattach it and none of it helped.  I also went as far as uninstalling all of USBIPD-WIN from both Linux and Windows and reinstall everything only to face the same issue.

I guess I'll close this issue as it appears to be a Windows problem, but what that issue is I have no idea.

closed 2022-12-03 by jpicklyk

bdog720, 2023-01-06:
I for anyone else looking into this. I did have VirtualBox installed and after uninstalling it everything worked (no restarts)

888zxc, 2024-04-03:
After using ```usbipd bind --force```, I was able to attach a usb device to wsl2

CommodoreOS, 2025-01-08:
usbipd 4.0.0 does not know the command `usbipd bind --force`

MichaelFeistETC, 2025-02-17:
@CommodoreOS 
[image]

You're just using the command wrong.

[image]