dorssel/usbipd-win #497: Cannot attach Class_FF USB device, which doesn't show up with "usbipd wsl list"

niekvse, 2022-12-07:
I have a Xilinx Platform Cable USB, which I would like to use in Xilinx Web ISE on Ubuntu through WSL2. However, sadly `usbipd wsl list` doesn't list this device, and so I have no way to attach it. After some Googling, I realized that apparently this is caused by the fact that this device is a `Class_FF` USB device. It would be great if this limitation could be resolved. Xilinx Web ISE is not supported on Windows 11, so probably quite a few of its users will be interested in getting the Linux version to work through WSL2.

The same issue is described by another user, here: https://superuser.com/questions/1736505/attach-an-unknown-device-to-wsl2-with-usbipd

See in particular this answer written below it: 

> "This is a super interesting problem. Your USB device is probably identifying itself as class FFh (vendor specific who-ha) and isn't getting passed through. MOST devices identify is one of the know USB class types making it easy to pass through. The known types don't actually use drivers beyond the default BT stack. I have to wonder if this is a shortcoming of WSL2. You might first identify if you are indeed using a class FFh device.. then look into how to connect THOSE to WSL of any brand to get closer.. just a thought. – 
> [Señor CMasMas](https://superuser.com/users/1000631/se%c3%b1or-cmasmas)"

dorssel, 2022-12-09:
The class has nothing to do with it, FF will work just fine.
The problem is that Windows device manager lists it as 'error' (in your linked article).
Can you post what the device looks like in device manager? And its property pages?

niekvse, 2022-12-09:
Good to hear that FF has nothing to do with it: I mentioned that mostly because that was the conclusion from the linked page, but I'm by no means an expert on this topic. I made some screenshots of the device & various properties in the device manager, hopefully that can help you shed some light on this issue! You can find them in the zip:

[Screenshot_20221209_155004.zip](https://github.com/dorssel/usbipd-win/files/10195775/Screenshot_20221209_155004.zip)

Let me know if you need any additional properties, or other information. I'm happy to help!

dorssel, 2022-12-10:
Ah, I see. No driver whatsoever, not even a generic or a NUL one... That *should* still work, but you are saying that the device doesn't even show up in `usbipd list`, right? That would be a bug, indeed, there is no reason to not list it. Unfortunately I don't have a device with such behavior (so I cannot debug this myself), but I will try to do some blind coding and send you test version(s) that may fix it. I'm quite busy, so it may take a week or so.

niekvse, 2022-12-10:
Indeed, no driver whatsoever - this is probably why it's giving an error. But because it's not showing up with `usbipd wsl list`, it's not possible to attach it. 

I would love to try your potential fix! 

I did try yesterday to install the driver from my Windows 10 machine on my Windows 11 machine, and in that case it did list it, but it's not really an ideal solution for many users (also because that driver made me disable the "core isolation" windows security setting, and mainly because this driver isn't easily downloadable anywhere) so it would be better if a real fix were available. I deleted that again so I'm back to the original state, ready to test any fixes you may come up with!

dorssel, 2022-12-11:
I managed to get my device in the same state by removing its drivers from my system. Same problem code 38, etc.
However, `usbipd` lists the device normally, and I can bind/attach it.

There is a difference, however. My device does not show as "Unknown device" but has a proper name. Probably because the string table in the device configuration contains one. Maybe yours does not (it is not mandatory). Could you try `usbview` from the SDK and get the raw configuration? See https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview.

My guess is that somewhere while getting the device details (e.g., the name), my device without a driver does return one, but yours does not and that's why `usbipd` is skipping it. The code to list all devices simply moves on to the next device if the basic information cannot be obtained. Problem is, I don't know which information is missing and which function call fails.

dorssel, 2022-12-11:
I think I found it. Can you check with `regedit.exe` in 

'''
Computer\HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Enum\USB\VID_03FD&PID_000F\5&354B5809&0&1
'''

Is there an entry `DeviceDesc`? I guess not...
If not, create a new string entry `DeviceDesc` with some value of your choice.
Does it work now?

niekvse, 2022-12-11:
Ah, this is great, that seems to have done the trick indeed! See these screenshots for some evidence :)

Before adding the `DeviceDesc` string:
[image]

After adding it, and running `usbipd wsl list`
[image]

Success! Thanks a lot for your quick help! Presumably you can alter the tool a bit to show "Unknown device" or whatever for these cases, but this help was good enough already, at least for me 👍

niekvse, 2022-12-11:
Hmm.. perhaps I got happy a bit too quickly. It does show up, which is great progress, but it doesn't yet want to attach itself. When I try, I get this:

[image]

Which does make some sense of course (given that it has an "error" state), but ideally it would still work. Any thoughts?

dorssel, 2022-12-11:
Baby steps ... one at a time.
(Sure, next version will allow for devices without any description.)

The 'error state' reported by Linux-side `usbip` is totally unrelated to the Windows driver state. Please run with full logging on the Windows side as here: https://github.com/dorssel/usbipd-win/wiki/Troubleshooting

dorssel, 2022-12-11:
I reproduced the problem by removing DeviceDesc explicitly. All things are equal to your case (BTW: my device is also class FF).

The linked PR solves the problem. The device now lists, binds, and attaches fine. When detached, it returns to the original Windows error state (code 38: no driver). Once the PR is merged, it will close this issue automatically.

Feel free to add more on your new (unrelated) problem, or open a new issue.

closed 2022-12-11 by dorssel

Programmer86, 2023-12-25:
Hello, I probably have a similar case. I have a webcam that is connected to a Hyper-V server, and it has no drivers. The device shows as "USB Composite Device" and 2 devices with driver error. When I run "usbipd list" I don't see this device. Researched these issues and tried adding DeviceDesc for these two devices with driver error and it worked. This is how it was before the register change:
[Get-PnpDevice.txt](https://github.com/dorssel/usbipd-win/files/13767182/Get-PnpDevice.txt)

PR #502 (2022-12-11): Add support for devices without any description
Fixes #497