dorssel/usbipd-win #498: Can't properly attach USB-Ethernet network adapter

HufnagelD, 2022-12-08:
Hello usbipd-win team,
Thanks for providing such tool to unlock the possibility to work with USB from WSL2!

I'm currently trying to use a USB-Ethernet network adapter to connect a Java GUI in the WSL to a external microcontroller. The setup is the following:

- External microcontroller, whose Ethernet interface has the following ip-config:

> IP-addr: 192.168.1.233
> netmask: 255.255.255.0
> gateway: 192.168.1.1

-On windows i give the USB-adapter the following IPv4 config:

> IP-addr: 192.168.1.1
> netmask: 255.255.255.0
> gateway: 192.168.1.233

The Java GUI connects to the 192.168.1.233 address. On windows this works excellent and i can connect the GUI to my external microcontroller. I tried this same setup on native Ubuntu as well (so not in a WSL2) and this works as well. 
The issue arrises, when i try to move this setup into the WSL. 
The installation of the Usbipd project works as expected. I can attach the USB-adapter to the WSL and inside the WSL it is visible. The driver for this adapter should be natively installed in a Ubuntu distribution. This imo is the case, since its properly recognized.
[image]
However, contrary to the native Ubuntu installation, it does not show up in the `ifconfig `list.
[image]
I.e. for the native Ubuntu installation, i have an additional entry, whose IP-address i can configure. Since the WSL doesn't seem to recognize this adapter as some sort of Ethernet interface, i can't configure it.

My question is first, if the principle of my setup is actually possible? I.e. Use an USB-ethernet adapter to connect to an external IP-address.
Secondly, if this is in general possible, if you have any idea, why this doesn't work, or why the adapter does not show up if i run `ifconfig`.

dorssel, 2022-12-09:
That will work fine. But the default WSL kernel does not have drivers for USB network devices.
See https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel

HufnagelD, 2022-12-19:
Thank you. I tried to follow this guide but I'm stuck at the following step.
`wsl --import wsl2-usbip <install-path> <temporary-path>\wsl2-usbip.tar`

What is meant with "install-path"? The installation path of the WSL in windows?

dorssel, 2022-12-19:
Yes, any path to which you (as a normal user) have write access. This is where the distro-specific files will be stored.

HufnagelD, 2022-12-19:
Oh, so that "install-path" is the new location, where the new distro-specific files will be stored and not a path to the already existing WSL install? 
So i.e. i just pick a random folder, for which i have write access. And here all my new distro files will be stored?

dorssel, 2022-12-19:
Yes

HufnagelD, 2022-12-19:
Thank you very much for the help. I'll close the issue.

HufnagelD, 2023-04-24:
Whilst this issue was addressed, another one arose. I did follow the guideline (https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel) and every step could be accomplished including the creation of the .wslconfig. However i ran into 2 problems.

1. All of these settings were already built-in in the newest kernel. I didn't have to manually add a single one of them.
[image]
Does that mean, the standard kernel has these configs already installed too and this process is not necessary anymore?
 2. Nonetheless, it still doesnt' work. The ethernet-adapter is still not showing up if i call "ifconfig".