dorssel/usbipd-win #499: it won't work

AndreV84, 2022-12-08:
can not map anything using wsl2 20.04 ubuntu image
first I tried to install ubuntu from wsl2 then from store. it won't work. tried multiple variations of implemented usbip thing
it would return that no usbip is properly installed
although instruction steps which got kinfd of obsolete were not applicable exactly I somehow got them applied.
Any ideas?
[image]

dorssel, 2022-12-09:
You're using the wrong commands, see https://github.com/dorssel/usbipd-win/wiki/WSL-support#usbip-client-tools.

AndreV84, 2022-12-09:
Hi @dorssel 
Could you be more specific, please?
What exactly command is wrong?
Which exactly command to use in windows power shell?
Which exactly to run from ubuntu terminal?
Or only one of the rwo environments needs commands executed?
The url you pointed out to says
```
These instructions have changed.
Updating to usbipd-win version 2.0.0 or higher will require these new instructions, even if the old instructions were followed before.
```

AndreV84, 2022-12-09:
is this particular command errorneous for execution within wsl2 20.04 ubuntu?
```
sudo apt install linux-tools-virtual hwdata
sudo update-alternatives --install /usr/local/bin/usbip usbip `ls /usr/lib/linux-tools/*/usbip | tail -n1` 20
```
it seems to result in erorneous outputs, however:
```
![err](https://user-images.githubusercontent.com/16374873/206655158-e482f133-c8a9-4fda-bb32-7d2b55d09979.png)

AndreV84, 2022-12-09:
will it also work if to map to ubuntu docker container?
thanks

dorssel, 2022-12-10:
I don't know, I don't use docker. But there are many user stories in the discussion list. Just search on 'docker'. I believe some got it to work. As the original issue is closed and this new question isn't really about `usbipd`, I will close this.