dorssel/usbipd-win #500: System.IO.FileNotFoundException during usbip attach

awm129, 2022-12-09:
I'm having some trouble trying to use usbipd-win with an Android device. I'm able to share the device through the windows side, and I can view the shared device if I issue a `usbip list` from the linux side, however, when I try to issue a `sudo usbip attach`, I'm seeing the following stack trace in the server logging:

```
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = ''
dbug: Usbipd.Server[1000]
      2.4.1+1.Branch.master.Sha.90acf9456020ca8c6310ca62a71ee23cb6ca34ad
dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveInterval = 500 ms
dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveTime = 10000 ms
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\WINDOWS\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: Usbipd.Server[1000]
      new connection from 192.168.1.101
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_DEVLIST
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 192.168.1.101
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 289
         at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
         at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 76
         at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 82
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 165
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 266
         at Usbipd.ConnectedClient.RunAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 58
dbug: Usbipd.Server[1000]
      connection close: Unable to find the specified file.
dbug: Usbipd.Server[1000]
      connection closed                    
```

This results in a `usbip: error: Attach Request for 1-17 failed - Device in error state` on the linux side. Any ideas?

dorssel, 2022-12-10:
Some devices require `usbipd bind --force`, as they don't like switching drivers "on the fly". This may also require a reboot. Once done, the device is permanently grabbed by the USBIP driver (until you `usbipd unbind`) and attaching shouldn't be a problem anymore.

awm129, 2022-12-10:
Thanks!

I had used `usbipd bind --force` to share the device. But I did not reboot. After rebooting and running the server in a windows powershell I'm still getting the same exception when trying to attach the device to my linux container. Perhaps my device (A Google Pixel 3) simply isn't supported by usbipd.

dorssel, 2022-12-11:
I think it should work. What is the output of `usbipd list`? After trying to attach, can you find the device in Windows device manager? Does it show an error state? Because with `usbipd bind --force`, the device driver is simply replaced with the VirtualBox driver that `usbipd` uses. And that should (in principle) be device agnostic... (i.e., all USB devices are equal in that sense).

awm129, 2022-12-11:
Here is what I'm seeing:

```powershell
PS U:\Users\User> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-4    0572:180a  HP Dock Audio, USB Input Device                               Not shared
1-7    04f2:b51c  HP HD Camera                                                  Not shared
1-12   8087:0a2b  Intel(R) Wireless Bluetooth(R)                                Not shared
1-17   18d1:4ee7  Android Composite ADB Interface                               Shared
2-4    174c:55aa  USB Mass Storage Device                                       Not shared
4-1    046d:c083  USB Input Device                                              Not shared
4-2    046d:c33c  USB Input Device                                              Not shared

Persisted:
GUID                                  DEVICE
0ed92537-a033-43d4-8a98-9e53acaf7cdb  Pixel 3

usbipd: warning: Unknown USB filter 'ASMTFilter' may be incompatible with this software; 'bind --force' may be required.
```

Using `usbip list` from linux:

```bash
[~]$ usbip list --remote=192.168.1.101
Exportable USB devices
======================
 - 192.168.1.101
   1-17: Google Inc. : unknown product (18d1:4ee7)
        : USB\VID_18D1&PID_4EE7\8BEX193Z9
        : (Defined at Interface level) (00/00/00)
        :  0 - Vendor Specific Class / unknown subclass / unknown protocol (ff/42/01)

[~]$ 
```

This is after a `usbipd bind --force` and a reboot.

This is what I'm seeing in linux when issuing an `attach`:

```bash
[~]$ usbip attach --remote=192.168.1.101 --busid 1-17
usbip: error: Attach Request for 1-17 failed - Device in error state 
```

And I see the stack trace posted above in the server logging.

dorssel, 2022-12-13:
If you use `--force`, then the list should have shown that. For example:

```
BUSID  VID:PID    DEVICE                                                        STATE
3-4    1bcf:0005  USB-invoerapparaat                                            Not shared
4-2    050d:0012  BCM92045DG Non-UHE                                            Shared (forced)
4-4    0557:2008  USB-Serial Controller D                                       Shared
5-1    2149:270b  USB-invoerapparaat                                            Not shared
5-2    046d:c534  USB-invoerapparaat                                            Not shared
5-4    0cf3:3004  Qualcomm Atheros AR3012 Bluetooth 4.0                         Not shared
6-8    1bcf:28a6  Integrated Webcam                                             Not shared
```

Try binding again with

```
usbipd bind --force --busid 1-17
```

And I am a bit worried about ASMTFilter. Looks like it is from a proprietary USB 3.0 driver from ASMedia. You could also try this device on another (maybe USB 2) port. In fact, `bind --force` was created to deal with such uncooperative filters, like the warning says.

awm129, 2022-12-14:
Thanks for the suggestions!

I've tried the following:

```console
PS C:\WINDOWS\system32>
PS C:\WINDOWS\system32>
PS C:\WINDOWS\system32> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-4    0572:180a  HP Dock Audio, USB Input Device                               Not shared
1-7    04f2:b51c  HP HD Camera                                                  Not shared
1-12   8087:0a2b  Intel(R) Wireless Bluetooth(R)                                Not shared
1-17   18d1:4ee7  Android Composite ADB Interface                               Shared
2-4    174c:55aa  USB Mass Storage Device                                       Not shared
4-1    046d:c083  USB Input Device                                              Not shared
4-2    046d:c33c  USB Input Device                                              Not shared

Persisted:
GUID                                  DEVICE
0ed92537-a033-43d4-8a98-9e53acaf7cdb  Pixel 3

usbipd: warning: Unknown USB filter 'ASMTFilter' may be incompatible with this software; 'bind --force' may be required.
PS C:\WINDOWS\system32> usbipd bind --force --busid 1-17
PS C:\WINDOWS\system32> echo $?
True
PS C:\WINDOWS\system32> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-4    0572:180a  HP Dock Audio, USB Input Device                               Not shared
1-7    04f2:b51c  HP HD Camera                                                  Not shared
1-12   8087:0a2b  Intel(R) Wireless Bluetooth(R)                                Not shared
1-17   18d1:4ee7  Android Composite ADB Interface                               Shared
2-4    174c:55aa  USB Mass Storage Device                                       Not shared
4-1    046d:c083  USB Input Device                                              Not shared
4-2    046d:c33c  USB Input Device                                              Not shared

Persisted:
GUID                                  DEVICE
0ed92537-a033-43d4-8a98-9e53acaf7cdb  Pixel 3

usbipd: warning: Unknown USB filter 'ASMTFilter' may be incompatible with this software; 'bind --force' may be required.
PS C:\WINDOWS\system32>
```
But I'm still seeing the same in the `usbipd list` output. I tried a couple different ports with the same results. Perhaps this ASMTFilter is the issue... I'm not even sure what that is.

dorssel, 2022-12-14:
Yes, seems like `--force` isn't having any effect. That's weird... (in fact, that's behavior I've not seen before)

ASMTFilter is a USB filter (i.e., it intercepts/interferes) with all USB devices/requests. Some (rather stupid, I must say) USB chip/hub vendors think they "value add" with this. You can try disabling it.

Run the following to show the filters:
```
reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}" /v UpperFilters
```

With regedit you could try to (temporarily) rename UpperFilters (to UpperFilter.Orig or so). Then reboot.

dorssel, 2023-05-15:
Stale