dorssel/usbipd-win #505: failure to attach to WSL Kali instance

TheProgrammingGamer, 2022-12-13:
After having looked through some issues I have found what I believe to be the correct packages for my distribution. And after having followed the provided steps I have encountered an error during the attaching process. The error message says that a firewall may be blocking the connection then says there was a TCP connection error. I'm not sure whats going on, so any help would be appreciated. Thanks!

[image]

dorssel, 2022-12-15:
Your *Windows host* needs to be accessible on port 3240. See the main README.

dorssel, 2023-02-27:
Stale

closed 2023-02-27 by dorssel