dorssel/usbipd-win #506: Unable to attach a device from the same model two times

toncho11, 2022-12-13:
I am trying to attach a 3rd Bluetooth dongle of type: "generic Bluetooth Adapter, Edimax Wi-Fi N150 Bluetooth4.0". In my project i need 4 Bluetooth devices. 

It fails with: 
```
usbipd: info: Using default distribution 'Ubuntu-20.04'.
usbip: error: Attach Request for 2-13 failed - Device in error state
```

This is the output of `usbipd wsl list`:

```
BUSID  VID:PID    DEVICE                                                                                            STATE
2-3    0bda:5549  Integrated Webcam, Camera DFU Device                                      Not attached
2-11   7392:a611  Generic Bluetooth Adapter, Edimax Wi-Fi N150 Bluetooth4.0...   Attached - Ubuntu-20.04
2-13   7392:a611  Generic Bluetooth Adapter, Edimax Wi-Fi N150 Bluetooth4.0...   Not attached
2-14   8087:0026  Intel(R) Wireless Bluetooth(R)                                                       Attached - Ubuntu-20.04
```
So I am able to attach the internal Bluetooth (Intel(R) Wireless Bluetooth) and one of the Edimax Wi-Fi N150, but I fail on the second - the 2-13 Edimax Wi-Fi N150. 

Any hints? I think even if Windows has some issue recognizing the second USB (Edimax Wi-Fi N150 Bluetooth) as a Bluetooth - this should not prevent usbipd from exporting it to WSL2. We need to recognize it as Bluetooth on WSL2. And it works for two devices, but not for three.

dorssel, 2022-12-13:
Duplicate of #452.  Please try to get some detailed information as described there. We need more debug info to tackle this one. As in: I cannot reproduce it, so I really need others to get me the info to fix this.

toncho11, 2022-12-14:
I commented in #452. Please tell me what else do you need?

dorssel, 2022-12-14:
I'll get back on it on that thread (once time perits). Closing this as duplicate.

closed 2022-12-14 by dorssel