dorssel/usbipd-win #511: Hyper-V guest doesn't see host-shared USB device

Pomax, 2022-12-16:
I have a Windows 10 pro host with a Hyper-V Windows 10 pro VM client, and on the host machine I've run `usbipd bind --busid=3-4` (a midi keyboard), but running `usbipd list` in the client shows no devices:

```
C:\>usbipd list
Connected:
BUSID  VID:PID    DEVICE                            STATE

Persisted:
GUID                                 DEVICE
```

The networking is set to bridged mode, and both host and client can ping each other on their respective 192.168.1.* addresses.

dorssel, 2022-12-17:
`usbipd-win` is only a server. You will need some other software on the Windows *client* side.

Pomax, 2022-12-17:
any you'd recommend? Because I ended up here after going "sorry, USB Network Gate wants $1200?", and VirtualHere not working for hyper-v

dorssel, 2022-12-18:
I don't use a Window client, so I wouldn't know. There are some entries in the discussions from people that successfully got it working. In any case, this is not an issue with `usbipd-win`, so I'll close this.

closed 2022-12-18 by dorssel

Pomax, 2022-12-19:
fair, although as a windows usbipd, it would help folks a _lot_ to also mention which windows usbip clients folks can use in the readme, because I've been spending four days trying to find one that actually works with Hyper-V, and apparently none of them do. Which people who need a windows usbipd probably care about a lot.