dorssel/usbipd-win #517: Issue with multiple devices sharing busid

myp-andrew-kirwin, 2022-12-29:
We seem to be having an issue with attaching a device that shares a BUSID with another device. For example, we are using a Dell Latitude with a USB-C dock connected. With the Yubikey in the other USB-C port on the laptop, the USB NIC and the Yubikey appear to share the same BUSID (as below).

`BUSID  VID:PID    DEVICE                                                        STATE
2-2    0bda:8153  Realtek USB GbE Family Controller #3                          Not attached
2-2    1050:0407  USB Input Device, Microsoft Usbccid Smartcard Reader (WUDF)   Not attached`

When connecting the Yubikey via the USB-C Dock, they have differing BUSIDs.

`BUSID  VID:PID    DEVICE                                                        STATE
2-2    0bda:8153  Realtek USB GbE Family Controller #3                          Not attached
3-2    1050:0407  USB Input Device, Microsoft Usbccid Smartcard Reader (WUDF)   Not attached`

When trying to attach the Yubikey when it's sharing the BUSID of the USB NIC, we get the below.

`Unhandled exception. System.AggregateException: One or more errors occurred. (Sequence contains more than one element)
   at System.Linq.ThrowHelper.ThrowMoreThanOneElementException()
   at System.Linq.Enumerable.TryGetSingle[TSource](IEnumerable`1 , Boolean& )
   at System.Linq.Enumerable.SingleOrDefault[TSource](IEnumerable`1 )
   at Usbipd.CommandHandlers.Usbipd.ICommandHandlers.WslAttach(BusId , Boolean , String , IConsole , CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\CommandHandlers.cs:line 459
   at Usbipd.Program.<>c__DisplayClass11_5.<<Run>b__36>d.MoveNext() in D:\a\usbipd-win\usbipd-win\Usbipd\Program.cs:line 479
--- End of stack trace from previous location ---
   at System.CommandLine.Invocation.AnonymousCommandHandler.InvokeAsync(InvocationContext )
   at System.CommandLine.Invocation.InvocationPipeline.<>c__DisplayClass4_0.<<BuildInvocationChain>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass17_0.<<UseParseErrorReporting>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass22_0.<<UseVersionOption>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass19_0.<<UseTypoCorrections>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<UseSuggestDirective>b__18_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass16_0.<<UseParseDirective>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<RegisterWithDotnetSuggest>b__5_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<CancelOnProcessTermination>b__1_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Invocation.InvocationPipeline.<InvokeAsync>g__FullInvocationChainAsync|2_0(InvocationContext )
   at System.CommandLine.Parsing.ParseResultExtensions.InvokeAsync(ParseResult , IConsole )
   at System.CommandLine.Parsing.ParserExtensions.InvokeAsync(Parser , String[] , IConsole )
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean )
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean )
   at System.Threading.Tasks.Task`1.get_Result()
   at Usbipd.Program.Run(IConsole , ICommandHandlers , String[] ) in D:\a\usbipd-win\usbipd-win\Usbipd\Program.cs:line 642`

Using the --hardware-id flag to attach (instead of the --busid) does not appear to make any difference. We suspect this is because it is simply resolving the hardware-id to the --busid and using this, which returns 2 results - the NIC and the Yubikey.

Could the --hardware-id flag be made to match the hardware-id and use this to attach the device?

dorssel, 2022-12-30:
Maybe, as a workaround, but the real problem is that your hub is assigning duplicate BusIds to begin with. Lets try to fix that first. My guess is that the hub is using either a filter driver, or a proprietary hub driver. I've seen both, and usually those proprietary "value added" stuff is crap. Can you find the hub in Windows device manager? And then check if it is using the stock Microsoft hub driver (which it should)? If not, what is the Manufacturer of the hub driver? And does `usbipd list` report anything about filter drivers?

myp-andrew-kirwin, 2022-12-30:
Right you are! I had the Asmedia extensible drivers from Dell installed for the TB16 dock. After uninstalling these completely and letting Windows install the hub driver it wanted I now have the USB NIC in the dock on BUSID 4-2 and the Yubikey gets 2-2 or 6-2, when connected to the laptop/dock respectively. Thank you.

closed 2022-12-30 by dorssel