dorssel/usbipd-win #524: USB Device Missing From Exportable List from WSL

mriksman, 2023-01-16:
Hi,

I used to perform my attach/bind from PowerShell using the command `

However, after I bridged my Internet and vSwitch together (using a Hyper-V virtual switch), I get the following error
```
PS C:\WINDOWS\system32> usbipd wsl attach --busid 4-2
usbipd: info: Using default WSL distribution 'Ubuntu'; specify the '--distribution' option to select a different one.
usbipd: error: The local IP address for the WSL virtual switch could not be found.
```

No problem, I thought, I can attach/bind it from `wsl` side. At least I thought...
[image]

Except, `4-2` is missing. Why is that?

I need help to either;
* Attach/bind from Windows side now that the IP/network is no longer NAT, or
* Find out why I can't attach a serial COM device from the `wsl` side.

mriksman, 2023-01-16:
I renamed my Virtual Switch to `WSL` and rebooted. I was then able to attach the device from the Windows side.

So `usbip` is hardcoded to look for `WSL` virtual switch? Maybe it should read the `.wsconfig` file to see what switch `wsl` has been configured for?

dorssel, 2023-02-27:
The current master does that. It will be in the next release.

closed 2023-02-27 by dorssel