dorssel/usbipd-win #532: Arduino Leonardo (Microcontroller) Keeps Disconnecting from Arduino IDE - WL2 usbipd-win

acruz7439, 2023-02-03:
Good afternoon,

I've been trying to configure a connection between the Arduino Leonardo to Ubuntu 20.04 WSL2 server. In this depository, I would run a virtual Linux desktop in order to start up an Arduino IDE (1.8.19).

When running usbipd-win, the Arduino IDE in Linux is able to detect the microcontroller. 

NOTE: The left side of the image below showcases the Arduino IDE connected to the Leonardo with usbipd-win. On the right hand side, you can see the connection occuring through the optional GUI for the usbipd-win.

[image]

However, whenever I would try to upload any sort of sketch into the device, it would automatically disconnect from the WSL2.

[image]

Whenever I would try to make the connection between the microcontroller and WSL through administrator or normal terminal or through the GUI (with or without auto-attach) the error STILL occurs.

What can I do to fix the issue?

dorssel, 2023-02-04:
Often, devices "reset" themselves after firmware flashing. To the host, this looks like an unplug/replug event. Have a look at the ```--auto-attach``` option. That will continuously try to re-attach the device if it gets detached (e.g., unplugged).

dorssel, 2023-09-12:
Stale.

It is either fixed by #684, which is in 3.2.0 release, or `--auto-attach` should be used.

closed 2023-09-12 by dorssel