dorssel/usbipd-win #538: Connection drop usbipd v2.4.1 / WSL2

kaplan2539, 2023-02-10:
We are observing connection drops during data transfer via FDTI Serial Converter (VID: 0403 PID: 6001).
The device is passed-through using `usbipd-win version 2.4.1+1.Branch.master.Sha.90acf9456020ca8c6310ca62a71ee23cb6ca34ad` on a Windows 10 Professional system into a WSL2 Ubuntu 20.04 distro.

Logs from `usbipd server`:
```
      Flushing
trce: Usbipd.AttachedClient[1001]
      Unlinking 232631, pending = True, pending count = 2
trce: Usbipd.AttachedClient[1001]
      Aborting endpoint 129
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
trce: Usbipd.AttachedClient[1001]
      Unlinking 233577, pending = True, pending count = 2
trce: Usbipd.AttachedClient[1001]
      Aborting endpoint 129
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.PcapNg[1000]
      Flushing
trce: Usbipd.AttachedClient[1001]
      Unlinking 233589, pending = True, pending count = 2
trce: Usbipd.AttachedClient[1001]
      Aborting endpoint 129
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
trce: Usbipd.AttachedClient[1001]
      Unlinking 233590, pending = False, pending count = 0
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
info: Usbipd.ConnectedClient[2]
      Client 172.19.242.161 released device at 1-6 (USB\VID_0403&PID_6001\A7040OCA).
dbug: Usbipd.ConnectedClient[1000]
      Unbind or unplug while attached
dbug: Usbipd.Server[1000]
      connection close: Attempted to read past the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.PcapNg[1000]
      Flushing
```

Pcapng file is attached
[usbipd.pcapng.zip](https://github.com/dorssel/usbipd-win/files/10705097/usbipd.pcapng.zip)

kaplan2539, 2023-02-10:
We're using version `5.10.102.1-microsoft-standard-WSL2` of the MS Linux kernel

kaplan2539, 2023-02-10:
More logs from another trial:

dmesg:
```
[99847.437516] ftdi_sio ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[99847.437527] vhci_hcd: unlink->seqnum 342135
[99847.437528] vhci_hcd: the urb (seqnum 342135) was already given back
[99847.467224] ftdi_sio ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[99847.467231] vhci_hcd: unlink->seqnum 342150
[99847.467232] vhci_hcd: the urb (seqnum 342150) was already given back
[99851.599562] ftdi_sio ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[99851.599573] vhci_hcd: unlink->seqnum 342881
[99851.599574] vhci_hcd: the urb (seqnum 342881) was already given back
[99886.627387] vhci_hcd: unlink->seqnum 342896
[99886.627389] vhci_hcd: unlink->seqnum 342897
[99886.627391] vhci_hcd: urb->status 0
[99886.627414] vhci_hcd: cannot find a urb of seqnum 342887 max seqnum 342897
[99886.627482] vhci_hcd: stop threads
[99886.627484] vhci_hcd: release socket
[99886.627485] vhci_hcd: unlink cleanup rx 342886
[99886.627486] vhci_hcd: the urb (seqnum 342886) was already given back
[99886.627489] vhci_hcd: disconnect device
[99886.627534] usb 1-1: USB disconnect, device number 11
[99886.627697] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[99886.627703] ftdi_sio 1-1:1.0: device disconnected
[99887.737318] WSL2: Performing memory compaction.
```
usbipd server:
```
      Unlinking 338058, pending = True, pending count = 2
trce: Usbipd.AttachedClient[1001]
      Aborting endpoint 129
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
trce: Usbipd.AttachedClient[1001]
      Unlinking 338059, pending = False, pending count = 0
trce: Usbipd.AttachedClient[1001]
      Unlinking 338951, pending = True, pending count = 2
trce: Usbipd.AttachedClient[1001]
      Aborting endpoint 129
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
trce: Usbipd.AttachedClient[1001]
      Unlinking 338969, pending = True, pending count = 2
trce: Usbipd.AttachedClient[1001]
      Aborting endpoint 129
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
trce: Usbipd.AttachedClient[1001]
      Unlinking 339785, pending = True, pending count = 2
trce: Usbipd.AttachedClient[1001]
      Aborting endpoint 129
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
trce: Usbipd.AttachedClient[1001]
      Unlinking 339796, pending = True, pending count = 2
trce: Usbipd.AttachedClient[1001]
      Aborting endpoint 129
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
[... 61 lines cut ...]
      Flushing
trce: Usbipd.AttachedClient[1001]
      Unlinking 342886, pending = True, pending count = 2
trce: Usbipd.AttachedClient[1001]
      Aborting endpoint 129
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: Usbipd.AttachedClient[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
trce: Usbipd.AttachedClient[1001]
      Unlinking 342887, pending = False, pending count = 0
info: Usbipd.ConnectedClient[2]
      Client 172.19.242.161 released device at 1-6 (USB\VID_0403&PID_6001\A7040OCA).
dbug: Usbipd.ConnectedClient[1000]
      Unbind or unplug while attached
dbug: Usbipd.Server[1000]
      connection close: Attempted to read past the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.PcapNg[1000]
      Flushing
```

usbipd_try2.pcapng is attached
[usbipd_try2.pcapng.zip](https://github.com/dorssel/usbipd-win/files/10711842/usbipd_try2.pcapng.zip)

kaplan2539, 2023-02-10:
The dmesg line `[99886.627414] vhci_hcd: cannot find a urb of seqnum 342887 max seqnum 342897` reminds me of #300

TakeshiSato, 2023-04-18:
I had the same symptoms.
It will be solved if you stop doing open-close every time you communicate.

kaplan2539, 2023-07-17:
Thanks @TakeshiSato !

@dorssel: I still see this v3.0.0 and believe it is a bug in `usbipd-win`. I've written a Python program to trigger this problem with a USB-serial converter based on the FTDI chip (available e.g. on [Amazon](https://www.amazon.nl/AZDelivery-Adapter-compatibel-FT232RL-Inclusief/dp/B0753H4SQS)). The Python program, Linux dmesg, Pcap, and summary of Pcap as .txt are attached as
[break_ftdi.zip](https://github.com/dorssel/usbipd-win/files/12073883/break_ftdi.zip)

My program is not a realistic use case - but it triggers the problem within seconds by opening and closing the serial port (/dev/ttyUSB0) constantly. Usually it takes way longer to observe the connection drop.

kaplan2539, 2023-07-17:
I've just tested another USB-serial adapter with a different chip (CP2101) and I also see the same problem (I takes longer to trigger though) - see pcap and dmesg attached in 
[break_cp2101.zip](https://github.com/dorssel/usbipd-win/files/12074040/break_cp2101.zip)

kaplan2539, 2023-07-17:
In the Linux `dmesg` I get something like
```
[ 5360.442368] vhci_hcd: unlink->seqnum 162590
[ 5360.442370] vhci_hcd: the urb (seqnum 162590) was already given back
```
everytime I close the serial port in the Python program - for both FTDI and CP2101.
Maybe that is a hint?

BTW:
- I also see this happening when I use Matlab reading out the serial port in a real world application that does also read data in between the open and close.
- Adding CPU and IO load via the `stress` program helps to trigger the problem in case of the CP2101 CHIP
- Adding network traffic also helps to trigger the problem

kaplan2539, 2023-08-17:
I am still seeing this in v3.1.0

kaplan2539, 2023-08-17:
However, when connecting through an `ssh` tunnel, it takes longer to drop the connection:
 ~ 180 sec via ssh
 ~ 5 sec via direct connection

kaplan2539, 2023-08-17:
I believe this is a bug in usbipd-win, because I cannot reproduce connection drops with VirtualHere using the same WSL2 distro on the same computer with the same device...
AFAIK VirtualHere behind the scenes is doing almost the same as usbipd-win (VirtualBox USB driver, USB/IP)

dorssel, 2023-08-18:
It looks like the device itself is resetting:

> Unbind or unplug while attached

That really only happens if you indeed unbind or unplug the device. You are not unbinding or unplugging physically, so the only reason this can happen is if the device momentarily "disappears" on the host side. Usually due to the device not being stable and resetting itself. To Windows, this looks like an unplug/replug event. The unplug causes `usbipd-win` to close the connection to the guest. After that, you will have to re-attach, or use the `--auto-attach` option.

> [99851.599574] vhci_hcd: the urb (seqnum 342881) was already given back

This is quite normal for a pipe abort. Linux cancels every URB individually, but Windows can only cancel every URB on a pipe in one go. So, the first cancel succeeds, but the seconds URB has already been "given back" due to the first. Linux still tries to cancel it individually, but it has already happened. This is benign, and a normal resolution to this race condition. This is not an error.

Personally, I think the device is unstable and resets every now and then.

kaplan2539, 2023-08-18:
I don't the believe the device is unstable - I've tested several different USB-serial converters even with different chips (FTDI, CP210x).  In addition, I don't see any connection drops on native Linux or when using the VirtualHere for USB pass-through to WSL2.

dorssel, 2023-08-18:
@kaplan2539
OK, that certainly is a smell. I have one serial converter here (ATEN), but I never use it. What is the reproduction pattern? Repeated open/close only? Or large transfers? It would be great for debugging if I could reproduce this.

dorssel, 2023-08-20:
@kaplan2539 
Good news (sort of). Your `break.py` also breaks my PL2303-based USB-serial converter. I can reproduce the problem!

dorssel, 2023-08-21:
I think I fixed it. Can you confirm? The installer of the PR is at https://github.com/dorssel/usbipd-win/actions/runs/5931974353?pr=684.

TL;DR for my own administration:

There was a possible (and quite rare) race condition where `usbipd` could return both a RET_SUBMIT *and* a positive RET_UNLINK for the same URB. Now it returns either RET_SUBMIT and an "already completed" RET_UNLINK, or no RET_SUBMIT and a positive RET_UNLINK. See https://docs.kernel.org/usb/usbip_protocol.html.

Unfortunately, none of the logging/pcap showed this. So, I had to backtrack (code inspection) in a debug session, while the problem hardly occurred for debug builds at all...

kaplan2539, 2023-08-22:
I wilk try ASAP. Thank you!!

kaplan2539, 2023-08-22:
Good news indeed! I can no longer trigger a connection drop with my code attached above on my CP2102 devices!!! That's great!!!!

However, I really need stability with serial-usb-converters using the FTDI chip... and unfortunately I can easily still reproduce connection drops with the FTDI usb-serial-converter even with the installer 3.1.1 from your PR.

If you don't mind, I can send you a FTDI usb-serial-convert device?

kaplan2539, 2023-08-22:
Company address listed in https://github.com/dorssel/usbipd-win/issues/425 still valid?

dorssel, 2023-08-22:
> Company address listed in #425 still valid?

Yes. Don't forget to include a return address.

kaplan2539, 2023-08-22:
BTW: I have tested two FTDI converters of different origins. With both I triggered a connection drop within seconds.

With the cp2101 (only have one at hand) I was able to reproduce the problem with v3.0.0 (happened to be installed) but after upgrading to the v3.1.1 `break.py` ran stable for 10+ minutes.

I suspect the FTDI devices / the Linux FTDI driver to behave slightly different - possibly leading to another race condition?

dorssel, 2023-08-22:
Before you ship me anything... I made a stupid mistake in the PR. Forgot to replace one TryGetValue with TryRemove. That can very well explain it.

Please try again with the build from https://github.com/dorssel/usbipd-win/actions/runs/5945143101?pr=684. You will have to uninstall/install. Dev versions all share the same version number and therefore you cannot update from dev to dev.

dorssel, 2023-08-26:
Good news (multiple):

1) I can consistently reproduce the problem on the devices you sent me
2) I found a method of logging that still reproduces the problem (ordinary logging/debug makes the problem not appear, or at least very rare --> Heisenbug)
3) I understand the problem now; it has to do with a race condition between writing back the SUBMIT and UNLINK replies to the client. Hard to explain: the SUBMIT wins (URB completes before UNLINK processed), so the the SUBMIT must send a normal reply and UNLINK must reply that the URB already completed. Yet, the UNLINK reply gets sent *before* the SUBMIT reply (even though SUBMIT won). Has to do with .NET async task scheduling. Anyway: Linux does not accept an UNLINK reply first telling it that the URB already completed and then afterwards receive the actual completion reply. The order must be SUMBIT first, UNLINK later (i.e., the order in which the requests were received).
4) it is solvable, and I know how to

Bad news: this is not a small fix. It requires rewriting the reply-logic. This also implies: lots of regression testing.

kaplan2539, 2023-08-26:
Good news and a lot to do... anything I can help with?

dorssel, 2023-08-28:
Could you test the latest version of the PR? It is at https://github.com/dorssel/usbipd-win/actions/runs/5994179196?pr=684.

I believe this should work, but it is not the "Right Way(TM)" to solve it. I just moved some async stuff to run synchronous, but it should do the job. The problem no longer reproduces on the FTDI devices you sent me.

If this works, then my understanding of the problem is correct. I will make a better version later.

dorssel, 2023-09-03:
@kaplan2539 
I refactored the entire UNLINK logic. Please try the latest build of the PR (https://github.com/dorssel/usbipd-win/actions/runs/6061183362?pr=684). This should do the trick.

kaplan2539, 2023-09-03:
Great! Thank you so much. I will test ASAP.

kaplan2539, 2023-09-04:
Looks good! I've tested with the two different FTDI devices I have and the CP2101 one. I don't see any connection drops when running my stress test (each >500s ). 🥳 Thank you!!!!

dorssel, 2023-09-04:
I am merging the PR; the code is much cleaner than before. But it requires some more regression testing before release. However, if anything is wrong with it I would rather fix the new stuff than go back to the old UNLINK mess.

dorssel, 2023-09-04:
@kaplan2539 
Do you have a return address for the devices? Or do you want me to keep them for future regression testing?

closed 2023-09-04 by dorssel

kaplan2539, 2023-09-04:
Please keep them - or donate to the closest hacker space / whoever can make use of them. Not worth shipping them around.

PR #684 (2023-09-04): Fix unlink race condition
Fixes #538