dorssel/usbipd-win #539: after installation on W11, laptop lost external monitor connection (even if W11 still reports it as connected)

mbonesi, 2023-02-10:
Trying to get the USB devices to work under WSL2 by following Microsoft guidelines as listed here: [walk through the steps necessary to connect a USB device to a Linux distribution running on WSL 2 using the USB/IP open-source project](https://learn.microsoft.com/en-us/windows/wsl/connect-usb)

- installed on W11 latest usbipd-win via `usbipd-win_2.4.1.msi`
- updated WSL2 kernel to `5.15.79.1-microsoft-standard-WSL2`
- used a more recent version of linux tools `sudo apt install linux-tools-5.15.0-58-generic hwdata`
- tested usbipd wsl list from within PowerShell `usbipd wsl list` and got the list of USB peripherals, but the service reported itself as not being started and suggested for a system reboot
- did so and came back with a black screen on the secondary monitor (which I use as main)
- all checks under W11 Display Settings reported the external monitor as connected but "extend these displays" or other options always failed to tun it on, and rolled back to "show only on 1"

**the laptop is connected via USB-C to a DELL P2422HE monitor which acts as the docking station (provides ethernet, power delivery, and video support)**

Uninstalling usbipd-win got my monitor functionality back after reboot (but no "full" USB under WSL2)

dorssel, 2023-02-10:
This happened even before binding and/or attaching any USB device, right?

The only thing I can think of is the VBoxUsbMon service, which permanently runs. ```usbipd``` itself does nothing with the hardware/drivers until ```attach```. VBoxUsbMon is a VirtualBox driver that hooks into the USB stack. However, it sometimes interferes (i.e., it is incompatible) with other filter driver. Did ```usbipd``` report anything about detected filter drivers?

Can you try to repeat the situation and then as an administrator run:
```
sc stop usbipd
sc stop VBoxUsbMon
```
That stops the VirtualBox driver. Does that make the monitor work again? (not that this will make ```usbipd``` work, just to confirm my guess)

birch-code, 2023-02-13:
Same issue. Windows 11. Desktop.

Symptoms:
1. Machine would be running ok then external monitor would glitch to mostly black then completely dark.
2. Could not RDP into machine.

Uninstalling from both Windows and Linux did not seem to fix. Disabling WSL did seem to fix.

dorssel, 2023-08-18:
Stale

closed 2023-08-18 by dorssel