dorssel/usbipd-win #542: Usbip is not found for kernel

pravic, 2023-02-20:
Following [the instructions](https://github.com/dorssel/usbipd-win/wiki/WSL-support#usbip-client-tools) I get this:

```sh
~$ usbip
WARNING: usbip not found for kernel 5.15.79.1-microsoft

~$ `which usbip`
usage: usbip [--debug] [--log] [--tcp-port PORT] [version]
             [help] <command> <args>

~$ which -a usbip
/usr/local/bin/usbip
/usr/bin/usbip
/bin/usbip
```

How is that possible?

<details>
  <summary>Full shell log</summary>
  
```sh
~$ sudo apt install linux-tools-virtual hwdata
The following additional packages will be installed:
  linux-tools-5.15.0-60 linux-tools-5.15.0-60-generic linux-tools-common
The following NEW packages will be installed:
  hwdata linux-tools-5.15.0-60 linux-tools-5.15.0-60-generic linux-tools-common linux-tools-virtual
0 upgraded, 5 newly installed, 0 to remove and 2 not upgraded.

~$ which usbip
/usr/bin/usbip
~$ which usbipd
/usr/bin/usbipd
~$ usbip
WARNING: usbip not found for kernel 5.15.79.1-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-5.15.79.1-microsoft-standard-WSL2
    linux-cloud-tools-5.15.79.1-microsoft-standard-WSL2

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard-WSL2
    linux-cloud-tools-standard-WSL2
~$ usbipd
WARNING: usbipd not found for kernel 5.15.79.1-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-5.15.79.1-microsoft-standard-WSL2
    linux-cloud-tools-5.15.79.1-microsoft-standard-WSL2

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard-WSL2
    linux-cloud-tools-standard-WSL2
```

```sh
~$ sudo update-alternatives --install /usr/local/bin/usbip usbip `ls /usr/lib/linux-tools/*/usbip | tail -n1` 20
update-alternatives: using /usr/lib/linux-tools/5.15.0-60-generic/usbip to provide /usr/local/bin/usbip (usbip) in auto mode

~$ which -a usbip
/usr/local/bin/usbip
/usr/bin/usbip
/bin/usbip

~$ usbip
WARNING: usbip not found for kernel 5.15.79.1-microsoft

~$ `which usbip`
usage: usbip [--debug] [--log] [--tcp-port PORT] [version]
             [help] <command> <args>

```
</details>

pravic, 2023-02-20:
I see. Judging by `/bin/usbip` it's a stub that compares `uname -r` with `ls /usr/lib/linux-tools/`. 

I just wonder how come `usbip` executes `/bin/usbip` even though `which usbip` executes the correct binary.

dorssel, 2023-02-20:
What's your distro? (I'm guessing Ubuntu...)

What is the output of
```
ls -l /usr/lib/linux-tools/*/usbip
update-alternatives --list usbip
```

pravic, 2023-02-20:
@dorssel Ubuntu 20.04 and 22.04 - the behavior is the same.

```sh
~$ uname -r
5.15.79.1-microsoft-standard-WSL2
~$ ls -l /usr/lib/linux-tools/*/usbip
lrwxrwxrwx 1 root root 33 Jan 20 16:42 /usr/lib/linux-tools/5.15.0-60-generic/usbip -> ../../linux-tools-5.15.0-60/usbip
~$ update-alternatives --list usbip
/usr/lib/linux-tools/5.15.0-60-generic/usbip

~$ update-alternatives --query usbip
Name: usbip
Link: /usr/local/bin/usbip
Status: auto
Best: /usr/lib/linux-tools/5.15.0-60-generic/usbip
Value: /usr/lib/linux-tools/5.15.0-60-generic/usbip

Alternative: /usr/lib/linux-tools/5.15.0-60-generic/usbip
Priority: 20
```

dorssel, 2023-02-20:
That all looks perfect. What is the output of just plain `usbip` now?

pravic, 2023-02-20:
Huh. Now it works:

```sh
~$ usbip
usage: usbip [--debug] [--log] [--tcp-port PORT] [version]
             [help] <command> <args>
```

Does WSL and/or Windows need to be restarted? Because previously `usbip` showed a warning and only `` `which usbip` `` worked.

dorssel, 2023-02-20:
I don't know. Maybe you just needed to re-login to WSL. Since you changed the resolve of `usbip`, and maybe bash cashed the old location. Anyway, glad it works now...

pravic, 2023-02-20:
Closing then, thanks!

closed 2023-02-20 by pravic