dorssel/usbipd-win #557: Serial device details unavailable when attached

PrisionMike, 2023-03-02:
Issue: When using the libapi (Python HID) I don't see the serial number, manufacturer_name, etc. fields being populated. Product ID, vendor ID, are being picked.

Device: Coinkite ColdCard wallet.

Steps to reproduce: Install hid package in python in windows or any native system. Run the following code with the wallet attached:
```
import hid
hidlist = hid.enumerate()
for hit in hidlist:
    if hit['manufacturer_string'] == 'Coinkite':
        print(hit)
```
You'll see all fields populated.

Now attach the device to the WSL and run the same python program in WSL (Ubuntu 22.04 in my case) and the aforementioned fields will have just an empty string.

dorssel, 2023-03-02:
Is the HID driver loaded in WSL?

PrisionMike, 2023-03-02:
Yes. That is how most of it works. It just seems to work incorrectly.

dorssel, 2023-03-02:
`usbipd-win` does not parse or communicate any details of the device, other than its VID and PID. The client driver or software requests that information directly from the device. There is no way that `usbipd-win` filters or otherwise interferes with device information. This really is a client-side problem. My guess is that WSL is not having the exact same kernel/driver configuration compared to your reference native system.

PrisionMike, 2023-03-05:
It turns out it was because of lack of priviledge that the serial number was not being read. Closing this issue here.

closed 2023-03-05 by PrisionMike