dorssel/usbipd-win #558: Kinova GEN2 robot not supported

focussash, 2023-03-03:
I've been testing with a Kinova MICO robot, and sending it to WSL Ubuntun with usbipd. 
Usbipd version: 2.4.1+1.Branch.master.Sha.90acf9456020ca8c6310ca62a71ee23cb6ca34ad
Ubuntu version: 22.04.2 LTS kernel 5.15.90.1-microsoft-standard-WSL2

Attaching most devices to Ubuntu works via `usbipd wsl --attach <bus-ID>`

Trying to attach the robot gives the error of usbip: error: Attach Request for <bus-ID> failed - Device in error state.

`usbipd list` lists the devices properly without giving any warning message, and `usbipd wsl list` similarly gives proper status (not attahced). Running `usbipd bind --force <bus-ID>` works, and subsequently `usbipd list` shows the device as being shared (forced), but none of these fixes the problem. I'm pretty sure this is an issue on the Kinova side, so just adding the issue here so potentially this could be added in the wiki as a device unsupported (or, I screwed something up and in that case hopefully someone points me out!)

focussash, 2023-03-03:
I want to add that pulling syslog on WSL Ubuntu shows nothing related to the USB - it doesn't get the device attachment attempt at all.

dorssel, 2023-03-04:
Can you run with full logging? See https://github.com/dorssel/usbipd-win/wiki/Troubleshooting.

focussash, 2023-03-04:
> Can you run with full logging? See https://github.com/dorssel/usbipd-win/wiki/Troubleshooting.

I tried to do this as I saw it proposed in another issue filed here, but iirc when shutting down the service nothing happens, and when starting a server for it I get an error saying an instance is already running  (I need to double check). I will get back to this on Monday hopefully with a full log! (Or, with the error I get when running the logging =.=)

focussash, 2023-03-06:
Following up: when trying to connect the robotic arm I get the following log: (Starting from the hosting)
`info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\WINDOWS\system32
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: Usbipd.Server[1000]
      new connection from 172.29.120.132
dbug: Usbipd.Server[1000]
      connection close: Attempted to read past the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.29.120.132
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 289
         at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
         at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 76
         at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 82
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 165
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 266
         at Usbipd.ConnectedClient.RunAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 58
dbug: Usbipd.Server[1000]
      connection close: Unable to find the specified file.
dbug: Usbipd.Server[1000]
      connection closed`

As a comparison, sharing some other USB devices gives the following:
`dbug: Usbipd.Server[1000]
      new connection from 172.29.120.132
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
info: Usbipd.ConnectedClient[1]
      Client 172.29.120.132 claimed device at 1-11 (USB\VID_5986&PID_0683\200901010001).
dbug: Usbipd.AttachedClient[1000]
      Trapped SET_CONFIGURATION: 1`

Side note: somehow for me `sc stop usbipd` (and `sc start usbipd`, for that purpose) do nothing. I had to kill usbipd with kill-process before being able to host a local server without getting "an instance exists" error.

dorssel, 2023-03-07:
Can you try the installer from the latest master build at https://github.com/dorssel/usbipd-win/actions/runs/4278344554?

Note that the installer is unsigned, but the drivers inside them are.

To revert to the release you will have to uninstall/reinstall (downgrading is not possible).

dorssel, 2023-03-07:
> Side note: somehow for me `sc stop usbipd` (and `sc start usbipd`, for that purpose) do nothing.

Maybe something else on your path is called `sc`. Try explicit `C:\windows\system32\sc.exe stop usbipd`.

focussash, 2023-03-10:
Thanks, I will try it out and get back early next week!

focussash, 2023-03-13:
So I tried the new build (unintalled the old one) and got essentially the same error: Log reads:
```
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.

info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production

info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Program Files\usbipd-win\

fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 308
         at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
         at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
         at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 167
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 270
         at Usbipd.ConnectedClient.RunAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 58
```
Something I can think of is the drivers for this particular Kinova model are unsigned (and we have to toggle BIOS to enable the drivers for USB), so maybe that is related?

BTW, explicitly calling sc.exe works!

dorssel, 2023-09-12:
This may be fixed by #677, which is in the latest 3.2.0 release. Can you confirm?

dorssel, 2023-10-20:
Stale

closed 2023-10-20 by dorssel