dorssel/usbipd-win #561: Unable to get NixOS WSL2 VM IP via `wsl -- cat /proc/net/fib_trie`

zanecodes, 2023-03-13:
For reasons that seem to be specific to the way NixOS wraps systemd under NixOS-WSL, I'm unable to run any commands within my NixOS WSL distribution from an elevated PowerShell session; I just get the error `nsenter: reassociate to namespace 'ns/mnt' failed: Invalid argument`.

This means that [the call to `wsl -- cat /proc/net/fib_trie`](https://github.com/dorssel/usbipd-win/blob/master/Usbipd/WslDistributions.cs#L147) always fails in an elevated session (and of course, running `usbipd wsl attach` in a non-elevated session always results in `usbip: error: import device`).

While this definitely indicates something that needs to be fixed in NixOS, I think it could be worked around here by using the [HostNetworkingService API](https://github.com/microsoft/dotnet-computevirtualization/blob/master/src/Microsoft.Windows.ComputeVirtualization/Hns.cs#L20), something like this:

```
[DllImport("vmcompute.dll", PreserveSig = false, ExactSpelling = true)]
private static extern void HNSCall(string method, string path, string request, [MarshalAs(UnmanagedType.LPWStr)] out string response);

...
string result;
HNSCall("GET", "/endpoints/", "", out result);
```

`result` will contain [a JSON object](https://github.com/microsoft/dotnet-computevirtualization/blob/master/src/Microsoft.Windows.ComputeVirtualization/Schema/Networking.cs#L95-L173) with `VirtualNetworkName` and `IPAddress` fields which can be used to find the IP of the WSL VM. You can see this in action by running `Get-HNSEndpoint` in an elevated PowerShell session while a WSL2 distribution is running:

```
PS C:\Users\Zane> Get-HNSEndpoint

ID                 : 5fbd1f7c-9c9b-4364-a4ee-6bd7acb391ee
Name               : Ethernet
Version            : 47244640267
AdditionalParams   : @{SwitchId=8D98ADC5-CDB4-4E03-997B-4739AA98ACA7;
                     SwitchPortId=23313E92-B6C2-469A-9677-DBE47F0A86E0}
Resources          : @{AdditionalParams=; AllocationOrder=2; Allocators=System.Object[]; CompartmentOperationTime=0;
                     Flags=0; Health=; ID=A394D20D-BAB4-4CE8-8DF8-B0562F2FFBB6; PortOperationTime=0; State=1;
                     SwitchOperationTime=0; VfpOperationTime=0; parentId=8E385CFC-058E-418C-9AF3-7EBE7244EE53}
State              : 2
VirtualNetwork     : b95d0c5e-57d4-412b-b571-18a81a16e005
VirtualNetworkName : WSL
MacAddress         : 00-15-5D-8A-7A-0C
IPAddress          : 172.27.234.179
PrefixLength       : 20
GatewayAddress     : 172.27.224.1
IPSubnetId         : 25b279f7-6ac5-41af-894f-ec2bffd1266b
DNSServerList      : 172.27.224.1
SharedContainers   : {}
VirtualMachine     : D5A13EE3-99F5-4542-AE82-98717569CCCD
```

I'd be happy to make the changes myself and open a PR in a few days if I can get my local development environment up and running.

### Relevant software versions
usbipd version: 2.4.1+1.Branch.master.Sha.90acf9456020ca8c6310ca62a71ee23cb6ca34ad
WSL version: 1.1.3.0
Kernel version: 5.15.90.1
WSLg version: 1.0.49
Windows version: 10.0.19045.2364
NixOS-WSL version: https://github.com/nix-community/NixOS-WSL/tree/c6f663e8583912059605b55c07583b6e96b5e690
nixpkgs version: 22.11

dorssel, 2023-03-14:
Thanks for your report.

As I understand, there is a generic problem running remote WSL commands from the Windows host with this distribution. I like the idea of using official Windows API; however, this would only solve part of the problem. It will allow `wsl list` to work (so I will definately look into it), but on the other hand `wsl attach` will never work (it *requires* running a command within Linux).

Since a workaround is available (i.e., running `usbip attach` from within Linux), I'll mark this as "documentation".

zanecodes, 2023-03-14:
That's very true, I can't believe I overlooked that rather important detail; and the possibility remains that I've simply bungled up my distribution's configuration so it may end up being entirely a me problem. Thanks for taking a look though, much appreciated!

dorssel, 2023-12-29:
@zanecodes 
See https://github.com/nix-community/NixOS-WSL/issues/229#issuecomment-1624778092
Is this issue resolved now?

dorssel, 2024-02-04:
stale

closed 2024-02-04 by dorssel