dorssel/usbipd-win #562: mt76-based wifi stick causes WSL2 to kernel panic

voltagex, 2023-03-17:
```
[   75.732415] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[   75.732990] vhci_hcd vhci_hcd.0: devid(131078) speed(5) speed_str(super-speed)
[   75.733478] vhci_hcd vhci_hcd.0: Device attached
[   76.088234] usb 2-1: SetAddress Request (2) to port 0
[   76.088721] usb 2-1: new SuperSpeed USB device number 2 using vhci_hcd
[   76.136410] usb 2-1: New USB device found, idVendor=0e8d, idProduct=7612, bcdDevice= 1.00
[   76.136958] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   76.137320] usb 2-1: Product: 802.11ac WLAN
[   76.137543] usb 2-1: Manufacturer: MediaTek Inc.
[   76.137844] usb 2-1: SerialNumber: 000000000
[  110.848222] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[  110.850234] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[  110.850647] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[  110.851298] cfg80211: failed to load regulatory.db
[  111.068239] usb 2-1: SetAddress Request (2) to port 0
[  111.068836] usb 2-1: reset SuperSpeed USB device number 2 using vhci_hcd
[  111.103712] mt76x2u 2-1:1.0: ASIC revision: 76120044
[  111.191171] mt76x2u 2-1:1.0: Direct firmware load for mt7662_rom_patch.bin failed with error -2
[  111.913185] mt76x2u 2-1:1.0: rx urb failed: -32
[  111.913619] ------------[ cut here ]------------
[  111.913990] rx urb mismatch
[  111.914007] WARNING: CPU: 0 PID: 320 at drivers/net/wireless/mediatek/mt76/usb.c:626 mt76u_complete_rx+0x16f/0x180 [mt76_usb]
[  111.914951] Modules linked in: mt76x2u(+) mt76x2_common mt76x02_usb mt76_usb mt76x02_lib mt76 mac80211 cfg80211
[  111.915847] CPU: 0 PID: 320 Comm: vhci_rx Not tainted 5.15.90.1-custom #6
[  111.916240] RIP: 0010:mt76u_complete_rx+0x16f/0x180 [mt76_usb]
[  111.916645] Code: 24 e9 56 ff ff ff 80 3d ee 40 00 00 00 0f 85 49 ff ff ff 48 c7 c7 d2 24 25 c0 48 89 34 24 c6 05 d6 40 00 00 01 e8 72 7b c5 c1 <0f> 0b 48 8b 34 24 e9 27 ff ff ff 66 0f 1f 44 00 00 0f 1f 44 00 00
[  111.917768] RSP: 0018:ffffc9000227fe28 EFLAGS: 00010082
[  111.918115] RAX: 0000000000000000 RBX: ffff888104bd3a00 RCX: 0000000000000003
[  111.918630] RDX: 0000000000000000 RSI: 00000000ffffefff RDI: ffffffff8284e2e0
[  111.919130] RBP: ffff888106d53910 R08: 0000000000000000 R09: ffffc9000227fc58
[  111.919553] R10: 0000000000000001 R11: 0000000000000001 R12: ffff888106d51d40
[  111.920056] R13: ffff888106d53918 R14: 0000000000000000 R15: ffff888100c63920
[  111.920511] FS:  0000000000000000(0000) GS:ffff8881ecc00000(0000) knlGS:0000000000000000
[  111.920950] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  111.921292] CR2: 00007fae65a2b658 CR3: 000000000280c000 CR4: 0000000000350eb0
[  111.921695] Call Trace:
[  111.921829]  <TASK>
[  111.921957]  __usb_hcd_giveback_urb+0x67/0xe0
[  111.922227]  vhci_rx_loop+0x198/0x2f0
[  111.922424]  ? pickup_urb_and_free_priv+0xb0/0xb0
[  111.922699]  kthread+0x127/0x150
[  111.922895]  ? set_kthread_struct+0x50/0x50
[  111.923099]  ret_from_fork+0x22/0x30
[  111.923297]  </TASK>
[  111.923450] ---[ end trace e78d27ccb7fb7ada ]---
[  111.923788] mt76x2u 2-1:1.0: rx urb failed: -32
[  111.924089] mt76x2u 2-1:1.0: rx urb failed: -32
[  111.924412] mt76x2u 2-1:1.0: rx urb failed: -32
[  111.924659] mt76x2u 2-1:1.0: rx urb failed: -32
[  111.924922] mt76x2u 2-1:1.0: rx urb failed: -32
[  111.925137] mt76x2u 2-1:1.0: rx urb failed: -32
[  111.925393] mt76x2u 2-1:1.0: rx urb failed: -32
[  111.925636] mt76x2u 2-1:1.0: rx urb failed: -32
[  111.925912] mt76x2u 2-1:1.0: rx urb failed: -32
[  111.926369] vhci_hcd: unlink->seqnum 417
[  111.926600] vhci_hcd: the urb (seqnum 417) was already given back
[  111.926724] mt76x2u: probe of 2-1:1.0 failed with error -2
[  111.926995] vhci_hcd: unlink->seqnum 418
[  111.927285] usbcore: registered new interface driver mt76x2u
[  111.927474] vhci_hcd: the urb (seqnum 418) was already given back
[  111.927475] vhci_hcd: unlink->seqnum 419
[  111.928375] vhci_hcd: the urb (seqnum 419) was already given back
[  111.928636] vhci_hcd: unlink->seqnum 420
[  111.928811] vhci_hcd: the urb (seqnum 420) was already given back
[  111.929084] vhci_hcd: unlink->seqnum 421
[  111.929258] vhci_hcd: the urb (seqnum 421) was already given back
[  111.929628] vhci_hcd: unlink->seqnum 422
[  111.929823] vhci_hcd: the urb (seqnum 422) was already given back
[  111.930104] vhci_hcd: unlink->seqnum 423
[  111.930319] vhci_hcd: the urb (seqnum 423) was already given back
[  111.930614] vhci_hcd: unlink->seqnum 424
[  111.930787] vhci_hcd: the urb (seqnum 424) was already given back
[  111.931104] vhci_hcd: unlink->seqnum 425
[  111.931319] vhci_hcd: the urb (seqnum 425) was already given back
[  111.931651] vhci_hcd: unlink->seqnum 426
[  111.931858] vhci_hcd: the urb (seqnum 426) was already given back
[  111.932231] vhci_hcd: unlink->seqnum 427
[  111.932409] vhci_hcd: the urb (seqnum 427) was already given back
[  111.932737] vhci_hcd: unlink->seqnum 428
[  111.932937] vhci_hcd: the urb (seqnum 428) was already given back
[  111.933300] vhci_hcd: unlink->seqnum 429
[  111.933484] vhci_hcd: the urb (seqnum 429) was already given back
[  139.658470] vhci_hcd: connection closed
[  139.658564] vhci_hcd: stop threads
[  139.659105] vhci_hcd: release socket
[  139.659258] vhci_hcd: disconnect device
[  139.659469] usb 2-1: USB disconnect, device number 2
```

(the disconnect is caused by me stopping the server)

PCAP: [usbipd-wifi1.zip](https://github.com/dorssel/usbipd-win/files/10998330/usbipd-wifi1.zip)

dorssel, 2023-09-12:
This may be fixed by #684, which is in latest 3.2.0 release. Can you confirm?

dorssel, 2023-10-20:
Stale

closed 2023-10-20 by dorssel

voltagex, 2023-10-20:
Sorry for not getting back to you - I don't think I have this stick anymore

Closed #562 as completed.