dorssel/usbipd-win #568: usbip err: usbip_network.c: 121 (usbip_recv_op_common) recv op_common, -1

dac2020, 2023-03-18:
I'm trying this on Windows 7 vm. I've been at this for days, looking all over the net, finding many articles saying it works and many more saying it never works.

The server is an openwrt device solely running the usbip as a server. That part is fine but usbip.exe can never find it.

c:\winapp\usbip>usbip -a 192.168.1.128 2-1
usbip err: usbip_network.c: 121 (usbip_recv_op_common) recv op_common, -1
usbip err: usbip_windows.c: 756 (query_interface0) recv op_common
usbip err: usbip_windows.c: 829 (attach_device) cannot find device

c:\winapp\usbip>usbip -l 192.168.1.128
usbip err: usbip_network.c: 121 (usbip_recv_op_common) recv op_common, -1
usbip err:       usbip.c: 216 (query_exported_devices) recv op_common
usbip err:       usbip.c: 288 (show_exported_devices) query

Not enough information anywhere to know how to use this but would love to.

dorssel, 2023-03-19:
I fail to see how this is related to `usbipd-win`, which is server-side and Windows only. You say your server is openwrt. Are you posting to the correct project?

dac2020, 2023-03-19:
No idea, it's a new project to me and like all projects, it takes a while to learn about it.
I'm actually looking for a Windows client, any Windows version but have yet to find one that actually works.

closed 2023-03-19 by dorssel