dorssel/usbipd-win #571: Unable to use USB device after mounted

Zibrandtsen, 2023-03-24:
Hello:)
I am trying to flash code to a microcontroller with USBasp using WSL2. 

It does not seem that I have any issues attaching the USB device, but as soon i try to use it, it does not work. I get the following error:
``` Bash
avrdude: error: could not find USB device with vid=0x16c0 pid=0x5dc vendor='www.fischl.de' product='USBasp'
```

After running the following command
``` Bash
avrdude -c usbasp -p m644p -P usb:16c0:05dc -b 19200 -U flash:w:main.hex
```

The command works fine in a virtual machine or bare metal with debian.  I have also tried to configure udev rules for the USBasb as described in this [LINK](https://andreasrohner.at/posts/Electronics/How-to-fix-device-permissions-for-the-USBasp-programmer/)

So, the USBasp do work fine on other machines with linux natively. So as far as i can tell I cannot get access to the USBasb device even though it says that it is attached to WSL2 as seen below

[image]

What can I do to fix the issue?

dorssel, 2023-03-25:
I think the WSL kernel does not have the driver for the device. WSL comes with a few (but not all!) serial USB drivers. Maybe your device needs one that isn't included by default. Try to figure out which USB driver is being used on your bare metal machine. Then recompile the WSL kernel with that driver enabled. See https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel.

Zibrandtsen, 2023-04-04:
I found out i tried to do something else than my task required. I may look into this again at a later date. Thanks for the help!

closed 2023-04-04 by Zibrandtsen