dorssel/usbipd-win #572: Unexpected behaviour in rtl-sdr when a USB transfer is cancelled

mluckham54, 2023-04-01:
This is for the usbipd-win experts.  I don't know whether it is related to usbipd-win or the way rtl-sdr [https://github.com/osmocom/rtl-sdr](url) is coded.  When run on an rPi, rtl-sdr returns an expected code of 0, but in the wsl/Ubuntu scenario with usbipb-win the return code is 5.

The issue is with rtl-sdr which calls libusb_cancel_transfer() in Ubuntu.  The first call to libusb_cancel_transfer succeeds but additional calls return 5 (LIBUSB_ERROR_NOT_FOUND).

There is a full description of the troubleshooting at [https://github.com/projecthorus/radiosonde_auto_rx/issues/765](url)

dorssel, 2023-04-03:
Can you run `usbipd` with full logging and PcapNg captures? See https://github.com/dorssel/usbipd-win/wiki/Troubleshooting.

mluckham54, 2023-04-05:
Yes, I'll do that

Thanks
Mike

mluckham54, 2023-04-09:
Here are the logs

Mike

mluckham54, 2023-04-09:
Here are the logs
[usbipd-rtlsdr.log](https://github.com/dorssel/usbipd-win/files/11185604/usbipd-rtlsdr.log)
[rtlsdr_pcapng.zip](https://github.com/dorssel/usbipd-win/files/11185606/rtlsdr_pcapng.zip)

dorssel, 2023-04-10:
The software calls read 16 times (this is normal). After the first read completes (with about 256kB of data), the remaining outstanding 15 reads are canceled (also normal). However, Window is unable to cancel individual read (this is a limitation of the VirtualBox driver). So, when the first of the 15 reads is canceled, `usbipd` actually cancels all 15 remaining on the same pipe. This, by the way, is allowed. So it returns EPIPE for all 15. This means the first cancel gets an OK. The other 14 are completed with EPIPE as well. The software nevertheless calls cancel on the remaining 14, which already have been canceled due to the first call, and so URB_NOT_FOUND is returned for those cancel requests.

I think the software is wrong. When canceling outstanding URBs, race conditions like these can always happen. Either you are the one canceling it first, and you get an OK. Or something else caused it to cancel, but you haven't been notified of the completion yet (which is the case here). In that case you get a EEXIST (translated by your software as "not found"). Any cancel routine must be able to deal with "not found", which means the URB has completed already, but the completion is still "under way".

dorssel, 2023-09-06:
This may have been fixed with https://github.com/dorssel/usbipd-win/pull/684, which by now has been merged into master. Could you give the latest master build a try? It's not released yet, the installer is at https://github.com/dorssel/usbipd-win/actions/runs/6078357827.

mluckham54, 2023-09-08:
> Thank you, I'll check it out

It seems to be working.  I updated radiosonde_auto_rx to the latest version
so my changes weren't in it, and installed your msi - everything worked
fine when rtl_sdr was invoked.

Thank you,
Mike

dorssel, 2023-09-08:
Thanks for testing. I will close this issue now. Feel free to reopen if it appears not to be solved after all...

closed 2023-09-08 by dorssel