dorssel/usbipd-win #579: Unexpected Response when attaching iDevice to WSL in usbipd 3.0.0

ASliceOfBread, 2023-04-13:
Hi,

As the title suggests, I am experiencing this issue when trying to attach an iPhone to a WSL2 instance using "usbip attach" on the WSL side. This exact setup works fine in version 2.4.1 but not in 3.0.0. Find logs attached.

[attachErrorLog.txt](https://github.com/dorssel/usbipd-win/files/11223664/attachErrorLog.txt)

The specific error is

usbip: error: Attach Request for 1-2 failed - Unexpected response

The iPhone is binded & forced

Running on Win10 version 20H2 x64

dorssel, 2023-04-15:
Thanks for the report. I can reproduce a similar problem myself with the current master branch. Notice that 3.0.0 has not been released and certainly isn't stable yet. Maybe a different approach compared to other projects, but `usbipd` does not have separate master/dev/stable branches. Feel free to try out the latest source, but YMMV. For stability, please stick to tagged released versions.

dorssel, 2023-04-15:
I've reported the issue upstream (see https://github.com/microsoft/win32metadata/issues/1541).
And for the time being I reverted the change. Could you try the artifact from PR #583? It is at https://github.com/dorssel/usbipd-win/actions/runs/4710286788?pr=583.

ASliceOfBread, 2023-04-18:
Thanks for this, can confirm that the issue no longer occurs using #583

closed 2023-04-18 by dorssel