dorssel/usbipd-win #582: File is being used by another process

jehoshua7, 2023-04-15:
**In Ubuntu 22.04**

`lsusb`

Bus 002 Device 002: ID 0781:5567 SanDisk Corp. Cruzer Blade
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

and the `mount` seems to be connected to the usb (EXT4 formatted)

/dev/sdc on / type ext4 (rw,relatime,discard,errors=remount-ro,data=ordered)

**Windows Power shell**

`wsl --version`

WSL version: 1.2.0.0
Kernel version: 5.15.90.1
WSLg version: 1.0.51
MSRDC version: 1.2.3770
Direct3D version: 1.608.2-61064218
DXCore version: 10.0.25131.1002-220531-1700.rs-onecore-base2-hyp
Windows version: 10.0.19045.2846

 `usbipd wsl list`

BUSID  VID:PID    DEVICE                                                        STATE
1-2    046d:c077  USB Input Device                                              Not attached
1-6    1199:907b  Sierra Wireless EM7430 Qualcomm Snapdragon X7 LTE-A           Not attached
1-7    8087:0a2b  Intel(R) Wireless Bluetooth(R)                                Not attached
1-8    04f2:b5ab  Integrated Camera                                             Not attached
1-15   0bda:0316  Realtek USB 3.0 Card Reader                                   Not attached
1-16   0781:5567  USB Mass Storage Device                                       Attached - Ubuntu-22.04

`GET-CimInstance -query "SELECT * from Win32_DiskDrive"`

DeviceID           Caption                    Partitions Size         Model
--------           -------                    ---------- ----         -----
\\.\PHYSICALDRIVE0 SAMSUNG MZVLW256HEHP-000L7 3          256052966400 SAMSUNG MZVLW256HEHP-000L7

`wsl --mount \\.\PHYSICALDRIVE0`

The process cannot access the file because it is being used by another process.
Error code: Wsl/Service/AttachDisk/0x80070020

dorssel, 2023-04-15:
You're mixing up two totally different ways of sharing a storage device with WSL. *Either* you use `wsl --mount` *or* you use `usbipd` and use `mount` from within Linux. In the first case, the device is managed by Windows, in the second the control of the device is completely transferred to Linux. You cannot use both at the same time.

jehoshua7, 2023-04-15:
> You're mixing up two totally different ways of sharing a storage device with WSL. _Either_ you use `wsl --mount` _or_ you use `usbipd` and use `mount` from within Linux. In the first case, the [...]

Prior to installing `usbipd` I was getting those messages when attempting a `mount` from `wsl`, indicating it is not some sort of sharing conflict between 'wsl' and 'usbipd'. From the wiki at https://github.com/dorssel/usbipd-win/wiki/WSL-support , it appears to me, that wsl/usbipd is not an "either/or" condition ..

> Now we can run list again and see the device is shared with WSL.

That aside, installing `usbipd' has in fact progressed my attempt to read the EXT4 usb though.   :)

> use usbipd and use mount from within Linux. 

I couldn't see an example of doing this in the docs/wiki. I did attempt a (linux) mount, however it kept failing.

dorssel, 2023-04-15:
Can you be more precise about what your issue with `usbipd` is? Your original report only shows an error for `wsl --mount \\.\PHYSICALDRIVE0`, which has nothing to do with `usbipd`.

jehoshua7, 2023-04-15:
Yes, sure. My apologies, I need to better document what happens.

jehoshua7, 2023-04-16:
1. Inserted the usb, then windows10 displayed 2 error messages - [a formatting message ](https://www.imgpaste.net/image/KHYcia) and [the directory name is invalid](https://www.imgpaste.net/image/KHYmOx)
Selected 'cancel' on the format msg and 'ok' on the other message.

2. Started Ubuntu 22.04
 
`uname -a`

> Linux DESKTOP-RBORDL9 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

3. Windows power shell as admin

`usbipd wsl list`

BUSID  VID:PID    DEVICE                                                        STATE
1-2    046d:c077  USB Input Device                                              Not attached
1-6    1199:907b  Sierra Wireless EM7430 Qualcomm Snapdragon X7 LTE-A           Not attached
1-7    8087:0a2b  Intel(R) Wireless Bluetooth(R)                                Not attached
1-8    04f2:b5ab  Integrated Camera                                             Not attached
1-15   0bda:0316  Realtek USB 3.0 Card Reader                                   Not attached
1-16   0781:5567  USB Mass Storage Device                                       Not attached

`usbipd wsl attach --busid 1-16`

`usbipd wsl list`

BUSID  VID:PID    DEVICE                                                        STATE
1-2    046d:c077  USB Input Device                                              Not attached
1-6    1199:907b  Sierra Wireless EM7430 Qualcomm Snapdragon X7 LTE-A           Not attached
1-7    8087:0a2b  Intel(R) Wireless Bluetooth(R)                                Not attached
1-8    04f2:b5ab  Integrated Camera                                             Not attached
1-15   0bda:0316  Realtek USB 3.0 Card Reader                                   Not attached
1-16   0781:5567  USB Mass Storage Device                                       Attached - Ubuntu-22.04

So, now the usb is attached. And Ubuntu shows it as

`lsusb`

Bus 002 Device 002: ID 0781:5567 SanDisk Corp. Cruzer Blade
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

whereas previously it was not displayed.

4.  The wiki, which I had been following (https://github.com/dorssel/usbipd-win/wiki/WSL-support ) states after the attach ..

> You should see the device you just attached and be able to interact with it using normal Linux tools.

But I ran a GUI filemanager under Ubuntu called "Files" and couldn't see it at all. A `mount' does seem to indicate it is mounted though, from a Ubuntu perspective.

/dev/sdc on / type ext4 (rw,relatime,discard,errors=remount-ro,data=ordered)
none on /mnt/wslg type tmpfs (rw,relatime)
/dev/sdc on /mnt/wslg/distro type ext4 (ro,relatime,discard,errors=remount-ro,data=ordered)

Please advise.  :)

jehoshua7, 2023-04-16:
Some additional info. Note the error on the last line

`dmesg | tail`

[  149.303767] vhci_hcd vhci_hcd.0: devid(65552) speed(5) speed_str(super-speed)
[  149.306694] vhci_hcd vhci_hcd.0: Device attached
[  149.651102] usb 2-1: SetAddress Request (2) to port 0
[  149.652990] usb 2-1: new SuperSpeed USB device number 2 using vhci_hcd
[  149.702944] usb 2-1: New USB device found, idVendor=0781, idProduct=5567, bcdDevice= 1.00
[  149.705308] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  149.707942] usb 2-1: Product:  SanDisk 3.2Gen1
[  149.709512] usb 2-1: Manufacturer:  USB
[  149.710672] usb 2-1: SerialNumber: 0401e815cfac1d1586162be1d7a0a8d77cb71837c6c72da357fa21feefa9fe597fe400000000000000000000f438ba450019561867558107c72c4d90
[ 2804.580791] WSL (9) ERROR: InitCreateProcessUtilityVm:1753: write failed 32

dorssel, 2023-04-17:
> A `mount' does seem to indicate it is mounted

/dev/sdc is your normal distro disk, not the USB device.

So far, all you did is attach the USB device to WSL. That was successful. What are you trying to do with the device? Mount it? Then you need to specifically do so manually, using `mount` within WSL. If you're using the stock WSL kernel, then mounting it is not going to work. The default kernel does not have USB storage drivers enabled. You will have to compile the kernel yourself in order to do that. Look through the other issues/discussions. It's been discussed many times.

jehoshua7, 2023-04-17:
> /dev/sdc is your normal distro disk, not the USB device.

Okay, thanks.
 
> So far, all you did is attach the USB device to WSL. That was successful. What are you trying to do with the device? Mount it?  

Yes, just mount it in Ubuntu and read it.

> Then you need to specifically do so manually, using `mount` within WSL.

Hmm, yes that is what I was attempting to do a few days ago. Yet I was advised ..

> **_Either_** you use wsl --mount **_or_** you use usbipd and use mount from within Linux. 

Seems to be conflicting information ??

> If you're using the stock WSL kernel, then mounting it is not going to work. The default kernel does not have USB storage drivers enabled. You will have to compile the kernel yourself in order to do that. Look through the other issues/discussions. It's been discussed many times.

Okay, from the comment at https://github.com/dorssel/usbipd-win/discussions/404#discussioncomment-3478849 , it seems I only need to follow the wiki instructions, so I will start from https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel

jehoshua7, 2023-04-17:
Have been working through compiling the kernel and the `git clone' is taking an incredibly long time. And resources it sees, see https://github.com/microsoft/WSL/issues/6982#issuecomment-1510676065