dorssel/usbipd-win #588: The selected WSL distribution cannot be reached via the WSL virtual switch.

j999yp, 2023-04-25:
About 2 weeks ago I managed to make my pl2303 USB to serial adapter work on wsl with usbipd. Today I want to try another device but sadly to find out it doesn't work anymore. I tried another distribution (`Debian` actually), but have no luck.
Here's some information that I think might be helpful:
- wsl
```bash
╭─j999yp@j999yp2 ~
╰─$ uname -a
Linux j999yp2 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
╭─j999yp@j999yp2 ~
╰─$ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 172.26.191.21  netmask 255.255.240.0  broadcast 172.26.191.255
        inet6 fe80::215:5dff:feed:a200  prefixlen 64  scopeid 0x20<link>
        ether 00:15:5d:ed:a2:00  txqueuelen 1000  (Ethernet)
        RX packets 799  bytes 720514 (720.5 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 581  bytes 91312 (91.3 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 4893  bytes 15343782 (15.3 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 4893  bytes 15343782 (15.3 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
- windows
```bash
Ethernet adapter vEthernet (WSLCore):

   Connection-specific DNS Suffix  . :
   Link-local IPv6 Address . . . . . : fe80::c9b0:ad2f:41d7:babd%57
   IPv4 Address. . . . . . . . . . . : 172.22.48.1
   Subnet Mask . . . . . . . . . . . : 255.255.240.0
   Default Gateway . . . . . . . . . :

Ethernet adapter vEthernet (WSL):

   Connection-specific DNS Suffix  . :
   Link-local IPv6 Address . . . . . : fe80::dcc2:3bdd:829b:3255%69
   IPv4 Address. . . . . . . . . . . : 172.26.176.1
   Subnet Mask . . . . . . . . . . . : 255.255.240.0
   Default Gateway . . . . . . . . . :
```
Both ip addr and subnet mask looks fine, and I can ping my wsl virtual machine from the host. I also checked firewall rules, but everything looks normal. Another issue mentioned the `-d` flag, which is definitely not my case.

By the way, may I request a `verbose` mode similar to ssh so that I can know what is actually going on?

nannerpusser, 2023-05-21:
`usbipd.exe wsl attach --busid 1-6`
`usbipd: error: The selected WSL distribution cannot be reached via the WSL virtual switch; try restarting the WSL distribution.`

Just ran into this earlier after a lengthy period since last using usbip(-win) alongside a Kali WSL2 install. Worked flawlessly a month or two ago, but I am blaming myself here since in the meantime I have installed and setup the Windows Subsystem for Android (WSA) preview beta and as far as I can tell, it uses the same virtual networking method as WSL does, including creating a duplicate vSwitch for another vEthernet adapter. WSA uses the names (WSL Core) on the virtual devices and WSL is still using (WSL) named adapters. I can run both simultaneously and can even throw a virtualbox layer on it and get crazy with NAT and four virtual boxes + kali + Android all running and pinging each other. I am confident my local network settings are in order, but I am clueless about the way `usbipd-win` does its thing and if a complicated scenario like WSA + WSL and all requisite Hyper-V hardware SxS would cause this.

j999yp, 2023-05-21:
> `usbipd.exe wsl attach --busid 1-6` `usbipd: error: The selected WSL distribution cannot be reached via the WSL virtual switch; try restarting the WSL distribution.`
>
> Just ran into this earlier [...]

WSA could be the reason cause this issue showed up right after I installed WSA. Sadly I cant test it because of some important data.

nannerpusser, 2023-05-21:
> ,,,

> WSA could be the reason cause this issue showed up right after I installed WSA. Sadly I cant test it because of some important data.

That is helpful for me then because I suspected it was the way Hyper-V is handling the vHardware setup on first run. The first time I installed WSA it also worked fine, but it had renamed and reassigned my original vEthernet (WSL) adapter that Windows alone has control over and won't retain any settings after restart. It's a problem because we can't do UDP over the WSL adapter and is related to why we have to re-roll a new local on a random subnet each time we restart WSL. It creates a whole new adapter each time. WSA is presumably doing the exact same thing and right now, in such early beta, as long as they can exist with each other there isn't gonna be any focus on outlier niche things like sharing USB devices over TCP/IP with WSL2 lol.

yodaluca23, 2023-08-12:
This is happening to me too, I have latest WSL and WSA (saw that maybe that's the problem..?), first I tried reinstalling my distribution, then all of WSL, and usbpid-win, still not working, it was working a few weeks ago... but I did mess around with WSA recently aswell... How can I get it working again?

nannerpusser, 2023-08-12:
> This is happening to me too, I have latest WSL and WSA (saw that maybe that's the problem..?), first I tried reinstalling my distribution, then all of WSL, and usbpid-win, still not working, it was [...]

Sadly, I tried manually correcting this by way of Terminal, netsh and regedit and could not entirely fix the issue. Ultimately I was able to return network settings to a pre-WSA state, but my registry hacks caused my ControlPanelUI (settings.exe displayed via taskbar and tray) to lose my Ethernet Adapter entirely along with real-time traffic stats. It worked fine, Win 11 UI just forgot it was there and it was a blank space with no sub-menus. I gave up and did a system restore  from an older clone.

WSA seems to be the issue when setup alongside a pre-existing WSL2 install. Unfortunately the virtual adapter and switch are protected and locked down to the system account keeping everyone but Windows from making and viewing and controlling any changes.

n3rd4i, 2023-09-09:
I've had this problem as well. I did the following to resolve it
1. reinstall winusb: https://learn.microsoft.com/en-us/windows/wsl/connect-usb
`winget install --interactive --exact dorssel.usbipd-win`
2. restart windows machine
3. reinstall winusb support within WSL: https://github.com/dorssel/usbipd-win/wiki/WSL-support:
```
sudo apt install linux-tools-virtual hwdata
sudo update-alternatives --install /usr/local/bin/usbip usbip `ls /usr/lib/linux-tools/*/usbip | tail -n1` 20
```

After these, the USB support withing WSL started working again. (I hope this helps)

13xforever, 2023-10-17:
Reinstalling the driver doesn't help me.
When trying to attach from linux side, I can list the device fine, but when trying to attach, it fails with device error
```
root@ls:~# usbip list -r 172.28.32.1
Exportable USB devices
======================
 - 172.28.32.1
        6-2: NVIDIA Corp. : Switch [Tegra Erista] recovery mode (0955:7321)
           : USB\VID_0955&PID_7321\7&5BA284F&0&2
           : (Defined at Interface level) (00/00/00)
           :  0 - Vendor Specific Class / Vendor Specific Subclass / Vendor Specific Protocol (ff/ff/ff)

root@ls:~# usbip --debug attach -r 172.28.32.1 --busid=6-2
usbip: debug: usbip.c:129:[run_command] running command: `attach'
usbip: debug: usbip_network.c:199:[usbip_net_recv_op_common] request failed at peer: 3
usbip: error: Attach Request for 6-2 failed - Device in error state
```

on windows side I have this error in system log:
```
Category: Usbipd.ConnectedClient
EventId: 3

An exception occurred while communicating with the client:

Exception: 
System.IO.FileNotFoundException: Unable to find the specified file.
   at Usbipd.ConfigurationManager.GetVBoxDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 309
   at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
   at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
   at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
   at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 167
   at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 271
   at Usbipd.ConnectedClient.RunAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 58
```

nannerpusser, 2023-10-18:
> Usbipd.VBoxUsb.ClaimDevice(BusId)

Are you certain the device you are trying to attach is supported? Also, have you tried detaching it all and then using the `wsl` flag in the `usbip` command? I know it's just technically a shortcut for ease of use, but doing it this way has corrected a permission error for me (that was my fault) in the past. I wish I knew more and I'm just a random guy who likes this project a lot, but here is a StackExchange answer that runs down the three ways to run and initialize `usbip`:

https://unix.stackexchange.com/a/700149

13xforever, 2023-10-18:
What do you mean by "device is supported"? It is shared with usbipd service on host side if you meant that. And reinstalling the driver removes the sharing, so you have to share it again.

USB pass-through worked fine few months ago just fine, but now all `usbip wsl attach` attempts result in the title error message. And doing it from the guest side results in service erroring out on the host side.

nannerpusser, 2023-10-18:
> What do you mean by "device is supported"? It is shared with usbipd service on host side if you meant that. And reinstalling the driver removes the sharing, so you have to share it again.
> 
> USB pass-through worked fine few months ago, but now all `usbip wsl attach` attempts result in the title error message. And doing it from the guest side results in service erroring out on the host side.

https://github.com/dorssel/usbipd-win/wiki/Tested-Devices 

That's what I meant by supported. Not every device works with the Windows daemon. Passthrough and this are very different things. Also make sure that the distro and kernel that had it working with passthrough are the same WSL2 distro and kernel.

dorssel, 2023-10-19:
This is now a known issue. The "auto-switch-detection" simply looks for the first adapter containing the word "WSL". Apparently, the WSA "WSL Core" switch comes first, and that is (of course) not reachable from WSL.

This code will be refactored, also to include the upcoming "mirrored" mode (see https://github.com/dorssel/usbipd-win/issues/714).

Until then: just do not use the "WSL convenience command `usbipd wsl attach`", but instead use `usbip attach` from within WSL directly.there).

yodaluca23, 2023-10-20:
> Until then: just do not use the "WSL convenience command `usbipd wsl attach`", but instead use `usbip attach` from within WSL directly).

I'm having trouble running this in WSL I get the following
```
yodaluca23@Lucas-Laptop:~$ usbip attach
WARNING: usbip not found for kernel 5.15.90.1-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-5.15.90.1-microsoft-standard-WSL2
    linux-cloud-tools-5.15.90.1-microsoft-standard-WSL2

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard-WSL2
    linux-cloud-tools-standard-WSL2
```
I had linux-tools-standard-WSL2 installed, switched to linux-cloud-tools-standard-WSL2 both had the same issue...

dorssel, 2023-10-20:
@yodaluca23 

> I had linux-tools-standard-WSL2 installed, switched to linux-cloud-tools-standard-WSL2 both had the same issue...

See instructions in https://github.com/dorssel/usbipd-win/wiki/WSL-support#usbip-client-tools

yodaluca23, 2023-10-28:
> See instructions in [dorssel/usbipd-win/wiki/WSL-support#usbip-client-tools](https://github.com/dorssel/usbipd-win/wiki/WSL-support#usbip-client-tools)

Again, really sorry for wasting your time, but now I'm getting this error:

```
yodaluca23@Lucas-Laptop:~$ usbip attach --busid=1-1
usage: usbip attach <args>
    -r, --remote=<host>      The machine with exported USB devices
    -b, --busid=<busid>    Busid of the device on <host>
    -d, --device=<devid>    Id of the virtual UDC on <host>
yodaluca23@Lucas-Laptop:~$ usbip attach --busid=1-1 --remote=*Redacted*
usbip: error: Attach Request for 1-1 failed - Device not found

yodaluca23@Lucas-Laptop:~$ usbip attach --busid=1-1 --remote=*Redacted* --device=*Redacted*:*Redacted*
usbip: error: Attach Request for *Redacted*:*Redacted* failed - Device not found
```

dorssel, 2023-10-29:
Did you actually bind device 1-1? What is the output of `usbipd list`?

yodaluca23, 2023-10-29:
> Did you actually bind device 1-1? What is the output of `usbipd list`?

In PS or WSL?

WSL:

```
yodaluca23@Lucas-Laptop:~$ usbipd list
WARNING: usbipd not found for kernel 5.15.90.1-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-5.15.90.1-microsoft-standard-WSL2
    linux-cloud-tools-5.15.90.1-microsoft-standard-WSL2

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard-WSL2
    linux-cloud-tools-standard-WSL2
yodaluca23@Lucas-Laptop:~$ sudo apt upgrade
[sudo] password for yodaluca23:
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
Get more security updates through Ubuntu Pro with 'esm-apps' enabled:
  libvlc5 vlc-data libvlccore9 libopenexr25 libpostproc55 libavcodec58
  libavutil56 libswscale5 libswresample3 vlc-plugin-video-output libavformat58
  libvlc-bin vlc-plugin-base libavfilter7
Learn more about Ubuntu Pro at https://ubuntu.com/pro
#
# Canonical released microcode updates for both Intel (CVE-2022-40982) and AMD
# (CVE-2023-20593). ‘Unattended upgrades’ provide security updates by default.
# Ensure it remains enabled to always get all updates as they become available.
#
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
yodaluca23@Lucas-Laptop:~$ sudo update-alternatives --install /usr/local/bin/usbip usbip `ls /usr/lib/linux-tools/*/usbip | tail -n1` 20
yodaluca23@Lucas-Laptop:~$ usbipd list
WARNING: usbipd not found for kernel 5.15.90.1-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-5.15.90.1-microsoft-standard-WSL2
    linux-cloud-tools-5.15.90.1-microsoft-standard-WSL2

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard-WSL2
    linux-cloud-tools-standard-WSL2
```

PowerShell:

```
PS C:\Users\yodal> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-1    *Redacted*:*Redacted* Realtek *Redacted* Wireless LAN *Redacted* USB NIC                  Not shared
1-2    *Redacted*:*Redacted*  USB Input Device, *Redacted*                     Not shared
1-7    *Redacted*:*Redacted*  USB Camera                                                    Not shared
1-8    *Redacted*:*Redacted*  USB Input Device, *Redacted*                                 Not shared
1-14   *Redacted*:*Redacted*  Intel(R) Wireless Bluetooth(R)                                Not shared

Persisted:
GUID                                  DEVICE
```

dorssel, 2023-10-29:
> 1-1    *Redacted*:*Redacted* Realtek *Redacted* Wireless LAN *Redacted* USB NIC                  Not shared

"Not shared". You need to bind the device first, as per the main README.

yodaluca23, 2023-10-29:
> "Not shared". You need to bind the device first, as per the main README.

Ok, I did that, and now it shows as shared, but when I try to attach from within WSL, I get this error.

```
yodaluca23@Lucas-Laptop:~$ usbip attach --busid=1-1 --remote=*Redacted*
usbip: error: import device
```

dorssel, 2023-10-29:
@yodaluca23 
Troubleshooting is explained in https://github.com/dorssel/usbipd-win/wiki/Troubleshooting

13xforever, 2023-11-04:
It looks like the issue in my case was some broken system state, after in-place upgrade that refreshed system components, everything works again, even the wsl remote commands from windows host on win11 23h2 and wsa installed

yodaluca23, 2023-11-17:
> This is now a known issue. The "auto-switch-detection" simply looks for the first adapter containing the word "WSL". Apparently, the WSA "WSL Core" switch comes first, and that is (of course) not [...]

Is there any ETA for this update?

darrenhearn, 2023-12-01:
This issue/comments have been very helpful - thanks to all for the details.

Adding a note because I believe this impacts more than just WSL customers who also have WSA (which was how I interpreted the conclusions from the comments here and related issue #136). My Windows 11 + WSL environment (never installed WSA) includes the following interfaces according to `ipconfig`:
```
Ethernet adapter vEthernet (WSL)
Ethernet adapter vEthernet (WSL (Hyper-V firewall))
```
I was running into the same errors about the WSL virtual switch - which makes sense because the root risk is apparently re-use of the string `WSL`. The "bind from Windows, attach from WSL" approach outlined by @dorssel resolved the issue for me.

yuanheci, 2023-12-01:
I also have the problem on win11 23h2 when i set the “networkingMode=mirrored".

yodaluca23, 2023-12-01:
> mirrored mode is tracked in https://github.com/dorssel/usbipd-win/issues/714. Will be resolved in 4.0.0, see https://github.com/dorssel/usbipd-win/discussions/774

And that will fix this issue?

dorssel, 2023-12-06:
4.0.0 has been released, which should resolve this issue. Could you confirm?

darrenhearn, 2023-12-06:
Assume you're looking for a confirmation from the original reporter, but can confirm that 4.0.0 and [its README](https://github.com/dorssel/usbipd-win/tree/v4.0.0) addressed the "selected WSL distribution cannot be reached via the WSL virtual switch" error for me (Win11 + WSL 2). Really appreciate the fix/doc updates and simplifications they deliver, @dorssel !

Note: I uninstalled 3.2.0 and installed 4.0.0 using MSI file from [releases](https://github.com/dorssel/usbipd-win/releases) (I guess `winget` install isn't yet aware of the new release because it complained).

dorssel, 2023-12-06:
@darrenhearn 
Thanks for the report.
BTW, winget is always a little delayed, it requires manual approval: https://github.com/microsoft/winget-pkgs/pull/129445

yodaluca23, 2023-12-07:
> 4.0.0 has been released, which should resolve this issue. Could you confirm?

Can confirm it's working for me, no errors!
[image]
It does show that warning, but it still works and shows up in WSL!
[image]

yodaluca23, 2023-12-07:
> @yodaluca23 The warning seems to be spurious. I would like to know why. Could you run in wsl the following command:
> 
> ```
> time bash -c 'echo < /dev/tcp/<your-host-IPv4-address-here>/3240'
> ```

I tried it again and didn't get the warning, I ran the command anyway...
[image]

OutPut:
```
real    0m0.002s
user    0m0.000s
sys     0m0.001s
```
[image]

dorssel, 2023-12-07:
These last results are what is expected. It does a quick check if the address can be reached, with a timeout of 1 second. For some reason, you got the earlier warning because it didn't respond within that 1 second. It is indeed spurious.

closed 2023-12-07 by dorssel

geosk, 2024-10-09:
winget install usbipd solved my problem