dorssel/usbipd-win #592: Attach by device name, VID/PID, or other?

JPHutchins, 2023-05-01:
Hi again!  I'm finally using this in my professional work flow and everything went great, very exciting!

Because I attach to different hubs, docks, etc., I wonder if I could look into adding a flag for connecting by a parameter other than BUSID? 

Potentially curious about connecting automatically as well.  That is, when a device power cycles and reconnects, it automatically attaches to WSL.

Cheers,
J.P.

dorssel, 2023-05-02:
You can connect by VID:PID already, use `--hardware-id` instead of `--busid` (see `usbipd bind --help`)

You can auto-connect to WSL upon unplug/reset already using `--auto-attach` (see `usbipd wsl attach --help`)

Any other automation can be scripted with the PowerShell interface, see https://github.com/dorssel/usbipd-win/wiki/Automation.

Attaching by name is not going to happen (other than scripting it yourself using the PowerShell interface), see https://github.com/dorssel/usbipd-win/issues/343

JPHutchins, 2023-05-02:
Excellent, I must have an older version installed.  I will update and take advantage of these features!

closed 2023-05-02 by dorssel