dorssel/usbipd-win #593: attach error: Device in error state (Unable to find the specified file)

irelevant25, 2023-05-09:
Hi, I was googling for a some time but could not find nothing what would works.
I have two same USB WiFi devices (tp-link tl-wn722n v4.0) what I am trying to attach to VM kali linux in hyper-v.

**Host system:**
Windows 11 Pro N
22H2
22621.1555
Windows Feature Experience Pack 1000.22640.1000.0

**USBIPD version:**
2.4.1+1.Branch.master.Sha.90acf9456020ca8c6310ca62a71ee23cb6ca34ad

**VM Kali linux system:**
Kali GNU/Linux Rolling
2023.1
Linux 6.1.0-kali7-amd64
x86-64

**USBIP version:**
usbip (usbip-utils 2.0)

**This is the output of command 'usbipd list' on host:**
```
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-5    0c45:636d  PC-LM1E                                                       Not shared
1-8    0a12:0001  Generic Bluetooth Radio                                       Not shared
1-10   0951:16df  HyperX Quadcast, USB Input Device                             Not shared
1-11   2357:010c  TP-Link Wireless USB Adapter #2                               Shared (forced)
1-14   2357:010c  TP-Link Wireless USB Adapter #2                               Attached
2-1    05ac:024f  USB Input Device                                              Not shared
3-2    0b05:1a68  USB Input Device                                              Not shared
```

**This is the output of command 'sudo usbip port' on client (I already attached one device):**
```
Imported USB devices
====================
Port 00: <Port in Use> at High Speed(480Mbps)
       TP-Link : TL-WN722N v2/v3 [Realtek RTL8188EUS] (2357:010c)
       1-1 -> usbip://192.168.4.167:3240/1-14
           -> remote bus/dev 001/014
```
           
**This is the output of command 'sudo usbip attach -r 192.168.4.167 -b 1-11' on client:**
`usbip: error: Attach Request for 1-11 failed - Device in error state`

**Error from event viewer:**
```
Category: Usbipd.ConnectedClient
EventId: 3

An exception occurred while communicating with the client:

Exception: 
System.IO.FileNotFoundException: Unable to find the specified file.
   at Usbipd.ConfigurationManager.GetVBoxDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 289
   at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
   at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 76
   at Usbipd.VBoxUsb.ClaimDevice(BusId ) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 82
   at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 272
   at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 272
   at Usbipd.ConnectedClient.RunAsync(CancellationToken ) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 74
```
   
Couple of additional information. If I unbind all the device and connect in different order. Every time only first one is working. When do that again in different order, again, only first one is working. So the issue I think is not in the device itself. Also, both devices have exactly the same **VID:PID**. But I am not able to change it via any software or manually from device configuration. I guess it is read only in firmware. Also, when I tried to attach it on client, I heard the sound of unplug device from host. And after a couple of seconds, I get the error. I Tried different ports and also disable power saver for devices. The command I used to bind is `usbipd bind -f -b 1-11`.

dorssel, 2023-05-09:
Good news. This is a duplicate of https://github.com/dorssel/usbipd-win/issues/452. It has been fixed in the latest master build and will be release as 3.0.0 soon. If you cannot wait, just download the installer artifact from the latest master pipeline. Note that the installer will not be signed (it is a dev build), but the drivers are, so it should work just fine.

irelevant25, 2023-05-09:
> Good news. This is a duplicate of #452. It has been fixed in the latest master build and will be release as 3.0.0 soon. If you cannot wait, just download the installer artifact from the latest [...]

I am not sure if it is weird or not, but now I would like to give you a hug. I wish you a perfect day your entire life. Thank you again <3 I really appreciate your time

closed 2023-05-09 by irelevant25