dorssel/usbipd-win #596: Unknown Issue with Arduino Due and Shield Device Attaching to WSL.

DrR0bot, 2023-05-12:
# Having an issue trying to attach my Arduino Due, that I haven't been able to solve.

I have used bind as required as well as using elevated privileges in PowerShell. Although the device looks as shared when I get the
usbipd list

However when trying to attach the device to wsl shows:
PS C:\windows\system32> usbipd wsl attach -b 1-1
usbip: error: Attach Request for 1-1 failed - Device in error state

usbipd: error: Failed to attach device with busid '1-1'.

I am attaching the video showing the process...

Thanks for the help!

https://github.com/dorssel/usbipd-win/assets/11063852/6c68e1ba-e170-4604-9428-177bf338e0de

dorssel, 2023-05-13:
First a comment:

After your device has been shared (either explicitly with `usbipd --bind` or after the first `usbipd wsl --attach`) you no longer need admin rights. Subsequent `usbipd wsl attach` can be run with normal user rights.

Some additional questions:
- did you reboot after installing `usbipd`?
- can you try with `usbipd --bind 1-1 --force`?
- can you run the `usbipd` service with full logging while trying to attach? (see https://github.com/dorssel/usbipd-win/wiki/Troubleshooting)

dorssel, 2023-09-12:
This may be solved by #677, which is in latest 3.2.0 release. Can you confirm?

dorssel, 2023-10-20:
Stale

closed 2023-10-20 by dorssel