dorssel/usbipd-win #605: Incompatible with PlatformIO

dakotahorstman, 2023-05-26:
Attempting to flash a device via PlatformIO, which attempts to force reset the device, fails.

I am specifically using a device based on the Arduino Zero (same MCU), but I cannot flash it from WSL. PlatformIO cannot force reset the device which means the device never enters "flashing" mode.

I've already attempted to use `--auto-attach` with my usbipd command, but to no avail. I also tried to "time" it where PlatformIO would attempt to force reset the device and I would mimic that by pressing the reset button on the device (obviously didn't work).

I can, most of the time, connect to the device and communicate via Serial though.

dorssel, 2023-05-26:
Often, devices that enter such "flash mode" report themselves as a "different device" (different VID:PID). You need to `usbipd bind` both.

- what is output of `usbipd list` normally?
- what is output of `usbipd list` right after the device is put into "flash mode" (right after the reset attempt)?

dakotahorstman, 2023-05-28:
**usbipd list, normally**
[image]

**usbipd list, after "Upload and Monitor"**
[image]

I tried to reattach the device at `2-2`, but `usbipd` throws an error claiming that the device is already attached to the client.

dorssel, 2023-05-29:
Just as I thought:

- first, the device is reporting as 2341:804d
- after flashing, it reports as 2341:004d

To Windows, this means it is a different device! You can see the "old" device in the persisted device list.

The solution is simple: you *also* need to `usbipd bind --busid 2-2` the "second" device. After that, the auto-attach script will automagically pick that up.

dakotahorstman, 2023-05-29:
I attempted that, but as I noted, `usbipd` claimed the device was already attached, you can see it circled in the picture below.

I instead attempted to switch to using the hardware id of the "devices", but I still got a timeout in PlatformIO, which you see on the left of the screen.

Note that the middle Powershell window is autoconnecting the original device, and the right Powershell window is autoconnecting the flash device.

[image]