dorssel/usbipd-win #606: usbip: error: Attach Request for <busid> failed

jmbruges, 2023-05-26:
Hi,

I have been following all threads and issues related to this and have not been able to find where the issue originates in my laptop.

Laptop OS aspects:

> Edition	Windows 10 Enterprise
> Version	22H2
> Installed on	‎2023-‎04-‎24
> OS build	19045.2965
> Experience	Windows Feature Experience Pack 1000.19041.1000.0

WSL distro:
> Ubuntu
> 5.15.90.1-microsoft-standard-WSL2

usbipd version:
> 3.0.0+361.Branch.master.Sha.89d4b327c99a4eecc9b583e1843a428aae3c9270

usbip version
> usbip (usbip-utils 2.0)

All firewall rules are made TCP 3240 free. During testing firewall was disable.

After following installation linked at [microsoft - Connect USB devices](https://learn.microsoft.com/en-us/windows/wsl/connect-usb) and trying to attach a device to wsl the following error appears:

From Powershell:

> PS C:\Users\[PC NAME]> wsl attach --busid 4-4
> usbip: error: Attach Request for 4-4 failed - Unknown Op Common Status
>
> usbipd: error: Failed to attach device with busid '4-4'

On WSL:

> user@hostname : sudo usbip attach -r [PC NAME].local -b 4-4
usbip: error: Attach Request for 4-4 failed - Unknown Op Common Status 

When reading [issue80](https://github.com/dorssel/usbipd-win/issues/80) I try to log the usbipd server but I cannot find connection to it. Basically there is no log, which I can think there is no communication channel between wsl usbip command and usbipd from the host pc.

> PS C:\Users\[PC NAME]> usbipd server Logging:LogLevel:Default=Trace
> dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
> dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = ''
> dbug: Usbipd.Server[1000]
      3.0.0+361.Branch.master.Sha.89d4b327c99a4eecc9b583e1843a428aae3c9270
> dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveInterval = 500 ms
> dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveTime = 10000 ms
> info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
> info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
> info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\U-JAV-0003
> dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
> info: Microsoft.Hosting.Lifetime[0]
      Application is shutting down...
> dbug: Microsoft.Extensions.Hosting.Internal.Host[3]
      Hosting stopping
> dbug: Microsoft.Extensions.Hosting.Internal.Host[4]
      Hosting stopped

I have tried several distros and reinstallation of the programs. I cannot find the source of the problem yet.

Thanks in advance for any help.

dorssel, 2023-05-26:
What is the output of

```
ls /usr/lib/linux-tools/*/usbip
```

jmbruges, 2023-05-26:
Here is the output:

> /usr/lib/linux-tools/5.15.0-72-generic/usbip

dorssel, 2023-05-26:
Your problem is that `usbip` (Linux side) really isn't able to connect to the host. This means it fails in a very early stage. The error is bogus, see https://www.spinics.net/lists/linux-usb/msg230262.html.

Did you do anything with the WSL switch on the host (e.g., rename it)? Can you ping the host from the Linux side?

jmbruges, 2023-05-26:
I can reach the host from WSL (ubuntu) machine. I have not rename the host so far because it is a corporate machine.
However I rename the host name in the ubuntu WSL instance, just to check if there is some weird stuff happening between the virtual link. (I guess you wonder that after I used [PC NAME] instead of $(hostname).local)

However, the only unreachable instance in my pc from the WSL is the "Ethernet adapter vEthernet (Default Switch)". All others are ping-able from the WSL.

dorssel, 2023-05-26:
> user@hostname : sudo usbip attach -r [PC NAME].local -b 4-4

Can you try this with a (pingable) IP address, instead of the hostname?

jmbruges, 2023-05-26:
I did and it worked.

dorssel, 2023-05-26:
Are you running as a user with admin rights (UAC), or as a standard user (and you need a different account for admin) on the host?

jmbruges, 2023-05-26:
> I did and it worked.

I wanted to add to this answer that it worked to ping the host from WSL. 
If you meant to attach the usb using IP address instead, it continues to fail:

> sudo usbip attach -r 172.23.160.1 -b 4-4
usbip: error: Attach Request for 4-4 failed - Request Completed Successfully

jmbruges, 2023-05-26:
> Are you running as a user with admin rights (UAC), or as a standard user (and you need a different account for admin) on the host?

Yes, I am using admin rights to all instances. (wsl, powershell, and if needed VScode)

dorssel, 2023-05-26:
Then it must still be firewall related. `usbipd` isn't even logging an incoming connection. And obviously (ping) the network is reachable. The (bogus) errors from `usbip` (Linux side) is a clear indication that it fails *before* the network connect.

Can you try a connection using some tool (telnet, netcat, socat, nc, ...) to the host port 3240? With `usbipd` logging on the console?

jmbruges, 2023-05-26:
Before the test result just wanna say that all originated because firewall rules (not the ones installed when installing the usbipd program, but the global group policies) did not allowed TCP 3240. We removed and allowed all programs and policies to not block that connection and the message "firewall was blocking TCP 3240" disapeared. 
Since still did not connect to the WSL I disabled firewall completely from my pc, but now we are here. At this moment of test I have enabled firewall rules again (I needed to perform my job safely) but if you would like I can disable an perform a test. 

From PowerShell:

> ping 172.23.173.194

> Pinging 172.23.173.194 with 32 bytes of data:
> Reply from 172.23.173.194: bytes=32 time<1ms TTL=64
> Reply from 172.23.173.194: bytes=32 time<1ms TTL=64
> Reply from 172.23.173.194: bytes=32 time<1ms TTL=64

But when Telnet to the port:
 > Test-NetConnection -ComputerName 172.23.173.194 -InformationLevel "Detailed" -Port 3240
WARNING: TCP connect to (172.23.173.194 : 3240) failed
> 
> ComputerName            : 172.23.173.194
> RemoteAddress           : 172.23.173.194
> RemotePort              : 3240
> NameResolutionResults   : 172.23.173.194
> MatchingIPsecRules      :
> NetworkIsolationContext : Internet
> InterfaceAlias          : vEthernet (WSL)
> SourceAddress           : 172.23.160.1
> NetRoute (NextHop)      : 0.0.0.0
> PingSucceeded           : True
> PingReplyDetails (RTT)  : 0 ms
> TcpTestSucceeded        : False

Similar if 

> $ telnet 172.23.160.1 3240
> Trying 172.23.160.1...
> Connected to 172.23.160.1.
> Escape character is '^]'.
> Connection closed by foreign host.

dorssel, 2023-05-27:
The Windows host is the server (i.e. `usbipd-win` listens on 3240). So we need to check the connection *from* WSL *to* Windows. This appears to be working:

> $ telnet 172.23.160.1 3240
Trying 172.23.160.1...
Connected to 172.23.160.1.

Do you also see the connection on the `usbipd-win` server console?

jmbruges, 2023-05-29:
Unfortunately it closed after start (but I cannot record any action).

> 
> Trying 172.23.160.1...
> Connected to 172.23.160.1.
> Escape character is '^]'.
> Connection closed by foreign host.

:(

dorssel, 2023-05-29:
> Do you also see the connection on the usbipd-win server console?

Run with full logging while doing the telnet, like you did before:

```
usbipd server Logging:LogLevel:Default=Trace
```

dorssel, 2023-05-30:
Then it must be (some) firewall setting. telnet is reporting making a connection, but it doesn't even reach `usbipd-win`.

MykhailoNovotnyi, 2023-06-02:
Hi, I have the similar issue.
In my case it does not work because no access to the Windows network from WSL.
WSL network: `inet 172.19.86.12  netmask 255.255.240.0`
Windows network: `IPv4 Address. . . . . . . . . . . : 172.19.80.1`, `Subnet Mask . . . . . . . . . . . : 255.255.240.0`
Access to the `172.19.80.1` from WSL does not work but from windows to `172.19.86.12` works.
The `usbipd` internally detect wsl windows host address (`172.19.80.1` in my case) and uses it, and failed.
As a workaround I did next:
1. Create ssh tunnel from windows to wsl with port forwarding `ssh -R 3240:127.0.0.1:3240 <user>@localhost`
It will allow `usbip` connect to the 127.0.0.1:3240 in WSL
3. build `usbipd` with some patch (hardcode `distros.HostAddress` to 127.0.0.1)
In that case `usbipd` will run `usbip` with parameter `--remote=127.0.0.1` instead of '--remote=172.19.80.1'

Windows version
```
OS Name:                   Microsoft Windows 11 Pro
OS Version:                10.0.22621 N/A Build 22621
```
WSL version
```
WSL version: 1.2.5.0
Kernel version: 5.15.90.1
WSLg version: 1.0.51
MSRDC version: 1.2.3770
Direct3D version: 1.608.2-61064218
DXCore version: 10.0.25131.1002-220531-1700.rs-onecore-base2-hyp
Windows version: 10.0.22621.1702
```

jmbruges, 2023-06-08:
Hi,

I stopped today to look on the `service.msc` and usbipd configuration and I found the next dependency:
[image]

Is this dependency correct? I don't have VirtualBox installed but had been installed before and removed.

/Javier

dorssel, 2023-06-21:
The only odd thing is your TCP dump

> 4 13.372849230 172.23.173.194 172.23.160.1 TCP 74 53082 → 3240 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 SACK_PERM=1 TSval=1422778159 TSecr=0 WS=128
5 13.373167768 172.23.160.1 172.23.173.194 TCP 66 3240 → 53082 [SYN, ACK] Seq=0 Ack=1 Win=65535 Len=0 MSS=1460 WS=256 SACK_PERM=1
6 13.373222047 172.23.173.194 172.23.160.1 TCP 54 53082 → 3240 [ACK] Seq=1 Ack=1 Win=64256 Len=0
7 13.373267397 172.23.173.194 172.23.160.1 TCP 62 53082 → 3240 [PSH, ACK] Seq=1 Ack=1 Win=64256 Len=8
8 13.424591876 172.23.160.1 172.23.173.194 TCP 54 3240 → 53082 [ACK] Seq=1 Ack=9 Win=2102272 Len=0
9 14.508316592 172.23.160.1 172.23.173.194 TCP 54 3240 → 53082 [FIN, ACK] Seq=1 Ack=9 Win=2102272 Len=0
10 14.508829002 172.23.173.194 172.23.160.1 TCP 54 53082 → 3240 [FIN, ACK] Seq=9 Ack=2 Win=64256 Len=0
11 14.509240344 172.23.160.1 172.23.173.194 TCP 54 3240 → 53082 [ACK] Seq=2 Ack=10 Win=2102272 Len=0

- Line 4 is the connect (SYN)
- Line 5 is the accept (SYN,ACK)
- Line 7 is the request (Len = 8)
- Line 9 is hangup (FIN)

However, `usbipd-win` never actually sees the connect. There is no log line 'new connection from...', which is really the very first thing logged after a connection is received by `usbipd-win`. This means the accept only occurred at the OS level, in the TCP stack, where the listening socket lives. Then the client (Linux) sends a request, after which the host OS immediately hangs up the connection. Since nothing was ever read from the socket by the client (hangup before any reply was ever sent), the client produces an error. This error is "bogus", due to a bug in Linux: https://www.spinics.net/lists/linux-usb/msg230262.html.

TL;DR: the socket gets closed by the host OS, without ever informing `usbipd-win` that it was ever connected. So, yes, I still think something firewall/policy related.

jmbruges, 2023-06-29:
@dorssel I wanted to report that after upgrade OS from W10 to W11 and following the usbipd and wsl installation procedure, everything works as expected. The program attaches and detaches USB devices as expected.

closed 2023-06-29 by jmbruges