dorssel/usbipd-win #607: support FTDI devices : fail on Device ID 0403:6010

ThierryCam, 2023-05-26:
I'm trying to use this tool with a FTDI device (0403:6010). And it works almost fine, except when I try to use the FTDI library.

FT_Open() fail with FT_DEVICE_NOT_OPENED return code.

I see that this device is not on your tested list, but I wonder if there's a way to make it work.

I can attach and see it in the lsusb list : 
Bus 001 Device 002: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC

Note that I'm using Windows 10, so I had to build my own kernel as advised at this link: 
https://learn.microsoft.com/en-us/windows/wsl/connect-usb

dorssel, 2023-05-27:
Microsoft's documentation is a bit outdated. Currently, WSL 2 for Windows 10 *does* support USBIP out of the box. For this device you need to enable CONFIG_USB_SERIAL_FTDI_SIO in the kernel. This *is* enabled by default in the current WSL 2 kernel (did you enable it in your own compile?)

Anyway, I don't think you need to compile the kernel yourself for this device. It could also be access rights related. Note that e.g. `systemd` and `udev` are not started by default on WSL.

ThierryCam, 2023-05-30:
OK, thanks for the informations. Yes, I use WSL2 and I think I have ftdi_sio in my kernel. When I attach the USD device, I see a /dev/ttyUSB0.

But my problem is when I call FT_Open(). On the FTDI installation guide, it's specified that VCP driver (ftdi_sio) is incompatible with D2XX driver, and requested to remove the ftdi_sio module : 
> sudo rmmod ftdi_sio
Which is giving errors no my setup.

I also tried to add this udev.rules, but didn't work either: 
ACTION=="add",ATTRS{idVendor}=="0403",ATTRS{idProduct}=="6010",MODE="0666",RUN+="/sbin/rmmod ftdi_sio",OPTIONS="last_rule"

ThierryCam, 2023-05-30:
$ sudo rmmod -v ftdi_sio
rmmod: ERROR: ../libkmod/libkmod-module.c:1941 kmod_module_get_holders() could not open '/sys/module/ftdi_sio/holders': No such file or directory
rmmod: ERROR: Module unloading is not supported

dorssel, 2023-05-30:
Normally, the WSL kernel is monolithic (i.e., all drivers that are enabled are statically linked in, not loaded as modules).

There is an option to compile as modules, but that is rather difficult (e.g., the location where the kernel is looking for the modules is *not* that of the distro running, but from some "under water" file system). Maybe you just want to *remove* CONFIG_USB_SERIAL_FTDI_SIO and recompile the kernel.

Note that this is more a WSL issue, not much to do with `usbipd-win`. Maybe you should inquire at https://github.com/microsoft/WSL.

ThierryCam, 2023-05-30:
Agree, it's more WSL than usbidp-win. But maybe you knew a way to make it work, since I saw you made some tests with many devices.
And I don't want to globally remove FTDI_SIO, since I want to use is for another device. What I really try to do is really to disable FTDII_SIO for a specific device ID, not sure if it's possible.
But thanks for the link, I'll continue the investigation on the WSL side.

closed 2023-08-28 by dorssel